dontsovcmc/waterius #1: TODO:

dontsovcmc, 2018-01-19:
tiny
- 2 канала
- передача типа устройства
- передача версии прошивок tiny, esp
- передача напряжения батарейки
- передача сетевых настроек
esp
- dhcp
- 2й wi-fi
сервер
- тип устройства
- БД
- визуализация

dontsovcmc, 2018-04-11:
Attiny хорошо бы заменить на современные аналоги: PIC16(L)F1454/5/9

dontsovcmc, 2018-05-01:
- питания через EN
- срезы только при изменении. отправка по заполнению буфера. или раз в N дней.

dontsovcmc, 2018-05-03:
Реализация без сервера:

1. добавить параметры в Web интерфейс счетчика
2. при просыпании, узнавать текущее время от NTP сервера. И при необходимости слать сообщение с текущими показаниями:
https://github.com/arduino-libraries/NTPClient/blob/master/examples/Basic/Basic.ino

dontsovcmc, 2018-05-08:
1. добавить диод на reset
2. запустить общий код в Atmel Studio
3. проверить регистр MCUSR при включении. Там инфа о перезагрузке.
4. сделать EEPROM
6. Проверить резистор на EN
5. отладить в Atmel Studio

dontsovcmc, 2018-05-08:
byte ResetSrc = MCUSR; // save reset source
  MCUSR = 0x00; //cleared for next reset detection
  if (ResetSrc & _BV(PORF)) Serial.println("PORF");
  if (ResetSrc & _BV(EXTRF)) Serial.println("EXTRF");
  if (ResetSrc & _BV(BORF)) Serial.println("BORF");
  if (ResetSrc & _BV(WDRF)) Serial.println("WDRF");

dontsovcmc, 2018-05-13:
1. кнопка на SDA - GND.
2. переписать код
3.

dontsovcmc, 2018-05-21:
1. потребление 2мА при включении?!
2. DNS сервер

может все таки можно без сервера?

dontsovcmc, 2018-06-19:
У библиотеки Blynk есть возможность синхронизации времени.
1. добавить параметр: день рассылки
Если день рассылки не пустой:
    Если месяц отправки не равен текущему
         Отослать

https://github.com/blynkkk/blynk-library/blob/master/examples/Widgets/RTC/RTC.ino

dontsovcmc, 2018-09-17:
Согласно [обзору STM8](https://www.terraelectronica.ru/news/5482) нам подходит этот: [STM8L051F3](https://docs-europe.electrocomponents.com/webdocs/1285/0900766b8128502b.pdf) на 20 пин. Можно взять и на 8 пин, но лучше больше, чтобы подключить датчик протечки и дополнительные входы.
PlatformioIO еще не поддерживает STM8L =(

dontsovcmc, 2018-09-24:
ESP32:
https://habr.com/post/418573/#comment_18937991
ULP Pulse Counting Example
examples/system/ulp/ вот что пишут:
With the default configuration (20ms ULP wakeup period), average current consumption in deep sleep mode is 16uA.

dontsovcmc, 2018-10-17:
Вот пример использования Радиомодуля с Attiny85. Т.к. мы автономны, нам нужно подавать на него питание. Это можно сделать, как отдельным стабилизатором, так и транзистором.

https://www.instructables.com/id/Mini-weather-station-with-Attiny85/

Проект с подключением LCD экрана тоже там.

closed 2022-01-13 by dontsovcmc

dontsovcmc, 2022-01-13:
[image]

heX16, 2022-01-18:
[image]

"крутилки" которые с право внизу изменил - уменьшил у них "saturation", иначе они сильно бросаются в глаза и ухудшают восприятие картинки.

dontsovcmc, 2022-01-18:
> [image]
> 
> @dontsovcmc, вот так мне кажется лучше выглядит? - "крутилки" которые с право внизу изменил, уменьшил у них "saturation", иначе они сильно бросаются в глаза и ухудшают восприятие картинки.

Секс! Гордость за комьюнити! Спасибо! Хотите пул реквест на титул сделайте?)) будете в разработчиках ааха