dontsovcmc/waterius #3: Обоснование: Питание

dontsovcmc, 2018-04-05:
ESP питается от 2.5 до 3.6В. 
Attiny85 питается от 1.6 до 5.5В

Li-Po:
- требуется преобразователь
- требуется зарядка

Батарейки АА 2 шт:
+ цена
- емкость

Батарейки АА 3 шт:
+ цена
- требуется преобразователь

Согласно: https://geektimes.ru/company/madrobots/blog/243377/
Качественные и не дорогие батарейки имеют 2500 мА*В*ч при токе разряда 200мА.

== 2 батарейки ==
Согласно графику они разрядятся до 1.25В при 60% емкости. Значит в нашем распоряжении 1000 мА*В*ч. У нас две батарейки 2000 мА*В*ч и 3В, т.е. рассчитываем на ~850 мА*ч....

В году 8760 часов. 850 мА*ч израсходуем при токе 97мкА..
Если мы хотим работать 4 года, то ток должен быть 24мкА.

== 3 батарейки == 
потребление преобразователя 2-4мкА (MCP1700, HT7333)
лучше взять на 2.8В преобразователь, чтобы снизить напряжение входа.
MCP1700 есть в продаже. Вот статья про них http://www.microchip.ua/publicazii/02-08.pdf
https://eu.mouser.com/datasheet/2/268/20001826D-737536.pdf
Возьмем Vin на 0.6V больше Vout. 3.4В/3 = 1.15В
1.15В при 40% емкости.. Значит в нашем распоряжении будет 2500 мА*В*ч * 60% = 1500мА*В*ч. 3 батарейки = 4500 мА*В*ч! Это сильно больше 850 мА*ч, учитывая накладные расходы в 2мкА!

В году 8760 часов. 4500 мА*ч израсходуем при токе 500 мкА ...

dontsovcmc, 2018-04-12:
Еще подходящие регуляторы:
http://download.siliconexpert.com/pdfs/2010/3/17/18/51/48/247/tsc_/auto/ts9011_e07.pdf
TS9011PCY RM TS9011PCX RF  3.0В

Возможно этот http://ww1.microchip.com/downloads/en/devicedoc/21335c.pdf не пойму какое у него потребление при 25мкА

dontsovcmc, 2018-04-16:
При опросе входов через 16мс потребление вместо 70мкА (вместо 22 при опросе раз в секунду)

dontsovcmc, 2018-04-16:
В течении 3-х суток счетчик раз в 2 секунды проверял входы, раз в 30 минут выходил на связь.
175 раз. Из них 50 раз 134 байта, остальные разы 14 байт (проблемы i2c). 
за это время напряжение 2х АА снизилось с 3.184 до 3.181В.

dontsovcmc, 2018-04-23:
2.6 sec первое включение
```
  NOTICE    (ESP) : Booted
000:00:00:00:041  NOTICE    (WIF) : Config loaded: IP=192.168.1.80, Subnet=255.255.255.0, Gw=192.168.1.1, Remote IP=192.168.1.42, Remote Port=5001
000:00:00:00:047  NOTICE    (WIF) : Device Id=11499, password=2627
000:00:00:00:052  NOTICE    (WIF) : Starting Wifi
000:00:00:00:057  NOTICE    (WIF) : Wifi status: 6
.....
000:00:00:02:373  NOTICE    (WIF) : Wifi status: 6
000:00:00:02:423  NOTICE    (WIF) : Wifi connected, got IP address: 192.168.1.80
000:00:00:02:424  NOTICE    (ESP) : Wifi-begined
000:00:00:02:424  NOTICE    (ESP) : I2C-begined
000:00:00:02:428  ERROR     (I2C end) : 2
000:00:00:02:433  ERROR     (I2C) : requestFrom failed
......
000:00:00:02:471  ERROR     (I2C) : requestFrom failed
000:00:00:02:475  NOTICE    (Stat: bytesReady) : 65535
000:00:00:02:480  NOTICE    (Stat: voltage) : 65535
000:00:00:02:485  CRITICAL  (I2C) : Slave is returning too much data
000:00:00:02:491  NOTICE    (WIF) : Making TCP connection to 192.168.1.42, Port 5001
000:00:00:02:558  NOTICE    (WIF) : Sending 14 bytes of data
000:00:00:02:559  NOTICE    (WIF) : Data sent successfully
000:00:00:02:660  ERROR     (I2C end) : 2
000:00:00:02:661  NOTICE    (ESP) : Going to sleep
```

650 ms когда данных 4 байта (700 ms при отсутствии данных), 
```
000:00:00:00:188  NOTICE    (WIF) : Config loaded: IP=192.168.1.80, Subnet=255.255.255.0, Gw=192.168.1.1, Remote IP=192.168.1.42, Remote Port=5001
000:00:00:00:194  NOTICE    (WIF) : Device Id=11499, password=2627
000:00:00:00:200  NOTICE    (WIF) : Starting Wifi
000:00:00:00:206  NOTICE    (WIF) : Wifi status: 6
000:00:00:00:258  NOTICE    (WIF) : Wifi status: 6
000:00:00:00:309  NOTICE    (WIF) : Wifi status: 6
000:00:00:00:359  NOTICE    (WIF) : Wifi status: 6
000:00:00:00:409  NOTICE    (WIF) : Wifi status: 6
000:00:00:00:459  NOTICE    (WIF) : Wifi status: 6
000:00:00:00:510  NOTICE    (WIF) : Wifi status: 6
000:00:00:00:560  NOTICE    (WIF) : Wifi connected, got IP address: 192.168.1.80
000:00:00:00:560  NOTICE    (ESP) : Wifi-begined
000:00:00:00:560  NOTICE    (ESP) : I2C-begined
000:00:00:00:573  NOTICE    (Stat: bytesReady) : 4
000:00:00:00:573  NOTICE    (Stat: voltage) : 3146
000:00:00:00:574  NOTICE    (I2C) : Polling slave for 4 bytes
000:00:00:00:588  NOTICE    (I2C) : Data retreived
000:00:00:00:589  NOTICE    (WIF) : Making TCP connection to 192.168.1.42, Port 5001
000:00:00:00:650  NOTICE    (WIF) : Sending 18 bytes of data
000:00:00:00:651  NOTICE    (WIF) : Data sent successfully
000:00:00:00:654  NOTICE    (ESP) : Going to sleep
```

dontsovcmc, 2018-05-19:
Оптимизация wifi 
https://www.bakke.online/index.php/2017/05/21/reducing-wifi-power-consumption-on-esp8266-part-2/

dontsovcmc, 2018-05-24:
EN - не ключ для питания ESP8266, а обычный пин включающий интерфейсы.
EN low питание в deepsleep около 10мкА.

Нужен силовой ключ для включения питания ESP8266.
Пример: 
http://www.ti.com/lit/ds/symlink/tps22860.pdf
http://ww1.microchip.com/downloads/en/devicedoc/21335c.pdf

Посматриваемся, как сделана плата: http://www.ti.com/tool/TIDA-00484
http://www.ti.com/lit/df/tidrey5b/tidrey5b.pdf

dontsovcmc, 2018-06-14:
Статический IP для подключения к WiFi + сервер blynk...com + отправка данных = 3.8 сек
```
000:00:00:00:096  NOTICE    (WIF) : Starting Wifi
000:00:00:00:100  NOTICE    (WIF) : Wifi status: 6
000:00:00:00:305  NOTICE    (WIF) : Wifi status: 6
000:00:00:00:505  NOTICE    (WIF) : Wifi status: 6
000:00:00:00:705  NOTICE    (WIF) : Wifi status: 6
000:00:00:00:905  NOTICE    (WIF) : Wifi status: 6
000:00:00:01:106  NOTICE    (WIF) : Wifi status: 6
000:00:00:01:306  NOTICE    (WIF) : Wifi status: 6
000:00:00:01:506  NOTICE    (WIF) : Wifi status: 6
000:00:00:01:707  NOTICE    (WIF) : Wifi status: 6
000:00:00:01:907  NOTICE    (WIF) : Wifi status: 6
000:00:00:02:107  NOTICE    (WIF) : Wifi status: 6
000:00:00:02:308  NOTICE    (WIF) : Wifi status: 6
000:00:00:02:508  NOTICE    (WIF) : Wifi status: 6
000:00:00:02:708  NOTICE    (WIF) : Wifi status: 6
000:00:00:02:908  NOTICE    (WIF) : Wifi status: 6
000:00:00:03:109  NOTICE    (WIF) : Wifi status: 6
000:00:00:03:309  NOTICE    (WIF) : connected
000:00:00:03:562  NOTICE    (BLK) : run
000:00:00:03:763  ERROR     (BLK) : disconnected
```
Вроде ничего не менял в скетче, хз почему стал подключаться целых 3.5 секунды

dontsovcmc, 2018-07-07:
DHCP (wifimanager release)
```
rl␀l��|␀�l�|␂␌␌␌�␌l�␌b|��␂�␒�r�bl␌b��nn�lnn���␌b␜p�lrlrlp�n�␐␂␌␌�␌l␌��␌␌␌b␌n�|␂l�␌␌�b��nn�␀l��l`␂�␒␒nn␌l`␂␎␂nr���n␌␌b␌�␎l␎r��n␌␌b␌�␎l��l␌���p��l`␂��n�␂000:00:00:00:049  NOTICE    (ESP) : Booted
2 000:00:00:00:052  NOTICE    (I2C) : mode=2
3 0 D3 B 0 0 0 0 0 0 0 0 0 0 000:00:00:00:063  NOTICE    (I2C) : data received
000:00:00:00:064  NOTICE    (WIF) : Config loaded: IP=192.168.1.178, Subnet=255.255.255.0, Gw=192.168.1.1, hostname=blynk-cloud.com
000:00:00:00:068  NOTICE    (WIF) : key=email=
000:00:00:00:077  NOTICE    (WIF) : value0_start=0.00, impules0_start=0, factor=10
000:00:00:00:084  NOTICE    (WIF) : value1_start=0.00, impules1_start=0
000:00:00:00:091  NOTICE    (ESP) : new impulses=0 0
000:00:00:00:095  NOTICE    (ESP) : new values=0.00 0.00
000:00:00:00:101  NOTICE    (WIF) : Starting Wifi
000:00:00:00:105  NOTICE    (WIF) : Wifi status: 6
000:00:00:00:309  NOTICE    (WIF) : Wifi status: 6
000:00:00:00:510  NOTICE    (WIF) : Wifi status: 6
000:00:00:00:710  NOTICE    (WIF) : Wifi status: 6
000:00:00:00:910  NOTICE    (WIF) : Wifi status: 6
000:00:00:01:111  NOTICE    (WIF) : Wifi status: 6
000:00:00:01:311  NOTICE    (WIF) : Wifi status: 6
000:00:00:01:511  NOTICE    (WIF) : Wifi status: 6
000:00:00:01:712  NOTICE    (WIF) : Wifi status: 6
000:00:00:01:912  NOTICE    (WIF) : Wifi status: 6
000:00:00:02:112  NOTICE    (WIF) : Wifi status: 6
000:00:00:02:313  NOTICE    (WIF) : Wifi status: 6
000:00:00:02:513  NOTICE    (WIF) : Wifi status: 6
000:00:00:02:713  NOTICE    (WIF) : Wifi status: 6
000:00:00:02:913  NOTICE    (WIF) : Wifi status: 6
000:00:00:03:113  NOTICE    (WIF) : Wifi status: 6
000:00:00:03:314  NOTICE    (WIF) : Wifi status: 6
000:00:00:03:514  NOTICE    (WIF) : Wifi status: 6
000:00:00:03:715  NOTICE    (WIF) : Wifi status: 6
000:00:00:03:915  NOTICE    (WIF) : Wifi status: 6
000:00:00:04:115  NOTICE    (WIF) : connected
000:00:00:04:284  NOTICE    (BLK) : run
000:00:00:04:619  NOTICE    (BLK) : virtualWrite OK
000:00:00:04:619  NOTICE    (BLK) : send email
000:00:00:04:686  NOTICE    (BLK) : email was send
000:00:00:04:686  NOTICE    (BLK) : Новые показания {DEVICE_NAME}
000:00:00:04:686  NOTICE    (BLK) : ГВС: 0.0 м3, ХВС: 0.0 м3
дельта:
гвс: +0.00, хвс: +0.00
питание:3.027 В
CMC:
вода добавить: 0.0 0.0
000:00:00:04:703  NOTICE    (BLK) : disconnected
000:00:00:04:707  NOTICE    (BLK) : send ok
000:00:00:04:711  NOTICE    (WIF) : Config stored: IP=192.168.1.178, Subnet=255.255.255.0, Gw=192.168.1.1, hostname=blynk-cloud.com
000:00:00:04:722  NOTICE    (WIF) : key=
000:00:00:04:728  NOTICE    (WIF) : value0_start=0.00, impules0_start=0, factor=10
000:00:00:04:736  NOTICE    (WIF) : value1_start=0.00, impules1_start=0
000:00:00:04:742  NOTICE    (ESP) : Going to sleep
```

dontsovcmc, 2018-12-03:
Китайцы представили свою энергоэффективную плату для ESP32. На ней стабилизатор 3.3В 1мкА до 0.6А: ETA3425

dontsovcmc, 2019-01-21:
Сравнительное тестирование литий-тионилхлоридных батареек
https://www.terraelectronica.ru/news/5706

dontsovcmc, 2019-04-05:
Будят GSM модуль транзистором irlml2502pbf 
https://mysku.ru/blog/aliexpress/61934.html
https://www.infineon.com/dgdl/irlml2502pbf-1.pdf

terminuslat, 2019-12-16:
Я только только  срукожопил  девайс, пока на макетке. Настроить не могу пока, но для питания сразу выбрал LiFePO4. Плюсы: 1) Напряжение прям в точку, 3.65 при только с зарядки до 2.2 безвредного разряда - соответственно не нужен стабилизатор. 2) Нет прыжков напряжения, свойственным алкалинкам и (меньше) другим  Li-xxx. 3) Длительный срок службы (хранения) при слаботочной нагрузке - до 15 лет. 4) Это всё же АККУМУЛЯТОР! 5) Удовлетворительно средний саморазряд  до 5 % в месяц. 6) Термоустойчив, включая отсутствие ярко выраженного саморазряда как на морозе, так ми в жару.   Минусы: 1) Цена 2) низкая удельная емкость (типоразмер АА - 600 мА/ч., стоит в розницу на рынке 300 руб)

dontsovcmc, 2019-12-16:
@terminuslat Крутота! Только ватериус очень мало потребляет и аккумулятор скорее разрядится, чем сядет от нагрузки. Лучше всего ER14505 батарейку 3.6В 2.5А. От нее 7-10 лет должен пропахать, т.к. саморазряда нет. https://market.yandex.ru/search?cvredirect=2&text=ER14505&local-offers-first=0

terminuslat, 2019-12-16:
ну, я для себя, покамест, хочу. А поскольку слегка подвинут на зелёной энергетике, то вполне могу пельтьешку прикрутить к горячей трубе. Иэта музыка пообещает стать вечной )
Кстати,  у названной Вами батарейки приличная депассивация. Стало быть напряжение здорово может просесть при включении вайфая. То есть на излёте заряда легко опуститься ниже допустимого. Это сократит полезную ёмкость. Да, впрочем, даже в даташите быстрым назван разряд 30 миллиампер - при этом заявлена ёмкость уже 1300  мА/ч. В итоге по соотношению цена/мА/ч выигрывают среднеценовые алкалинки со стабилизатором... ИМХО...
Ладно, извиняюсь, что отвлекаю, пойду искать таки "прошивку с TEST_WATERIUS", не пойму где она лежит )), плата пока не работает ))

dontsovcmc, 2019-12-16:
@terminuslat хм.. может поэтому ставят кондеры побольше?

terminuslat, 2019-12-16:
спс )

dontsovcmc, 2020-03-23:
https://www.terraelectronica.ru/news/6478 по теме

Kabron287, 2021-02-18:
Собрал для погодного датчика такую схему питания.
Работает безупречно. Ток в отключке 10-20uA. 
В качестве главных мозгов WEMOS D1 mini Pro.
Хороша тем что для USB моста используется CP2104, которая при отсутствие USB спит.
Питается она от WEMOS Battery Shield с встроенным зарядником и повышалкой до 5В.
Есть ньюанс, который не сразу выявился - необходима индуктивность, ибо при включении транзистора,  время нарастания питания очень мало и возникающая просадка ресетит тиньку.  Q8, Q9, Q5 - идеальный диод для развязки от питания тиньки при прошивке(она у меня  - Digispark).

Battery Shield нуждается в минимальной переделке: поскольку батарея, теперь подключается не напрямую, плюс батарейного контакта соединен с A0_ADC, а через RST выведен VBUS, чтобы при зарядке тинька знала что нужно включить ключ. 
[image]

dontsovcmc, 2021-05-28:
LMR16006
https://www.revk.uk/2021/05/battery-powered.html

dontsovcmc, 2022-01-13:
@Kabron287 спасибо, что поделились!
сложности с включение esp-01 транзистором - у меня GPIO00, GPIO02 подтянуты к питанию т.к. линия i2c, а у GPIO02 внутренний резистор есть на плюс. Кажется их надо изолировать (но вопрос с прошивкой attiny возникает) или другие пины для i2c использовать. Так и поставили на паузу задачу...

dontsovcmc, 2023-09-15:
> LDO XC6206
Потребление 200mA (3.0V type). Т.е. надо ставить большой конденсатор, т.к. во время передачи.

dontsovcmc, 2024-03-31:
[CJ9113T6A](https://www.compel.ru/infosheet/JSCJ/CJ9113T6A)  - !
LM3670MF-3.3 https://rocelec.widen.net/view/pdf/ehgdv03kau/NATLS19550-1.pdf

MCP1640/B/C/D https://www.farnell.com/datasheets/1669480.pdf
MCP1623  https://docs.rs-online.com/e094/0900766b8132484e.pdf
BD70522GUL  https://fscdn.rohm.com/en/products/databook/datasheet/ic/power/switching_regulator/bd70522gul-e.pdf
[SP6651AEU](https://www.digikey.com/en/models/2472130)

tps736

dontsovcmc, 2024-12-11:
Хорошая статья про энергоэффективность плат на ESP32
https://diyi0t.com/reduce-the-esp32-power-consumption/