dontsovcmc/waterius #10: Заводские аналоги

dontsovcmc, 2018-06-08:
- [Контроллер SAURES R1](https://www.saures.ru/katalog/ustroystva/kontroller-r1/)

aimsson, 2018-07-30:
Вроде есть еще "умные" стрижи - https://strij.tech/products/umnyie-schetchiki-vodyi

dontsovcmc, 2018-07-30:
Спасибо, добавил!

bashlakov, 2018-07-31:
Ещё есть Нептун ProW+ WiFi
https://neptun-mcs.ru/catalog/controllers/modul_upravleniya_neptun_prowwifi/

dontsovcmc, 2018-09-04:
Пасиб!

evg26, 2019-01-06:
http://www.mastergrad.com/forums/t233797-srochno-distancionnoe-snyatie-pokazaniy-s-treh-schetkov-vody-s-imp-vyhodom/?page=2

dontsovcmc, 2019-04-04:
Статья от создателя Пульсара: http://www.rmcip.ru/files/rus/pdf/articles/2011/ef51679f479c2cb9171920011777aa3f.pdf

GodinRaider, 2019-11-15:
Есть еще аналог Сауреса - https://valtec.ru/catalog/pribory_ucheta/komplektuyuschie_dlya_priborov_ucheta/wi_fi_ustrojstvo_sbora_i_peredachi_dannyh.html
Что-то типа выполненной по лицензии штуки. Там 8 каналов, как в старых Сауресах, а стоит дешевле. Они официальные, я уточнял у поддержки Саурес.

dontsovcmc, 2019-11-15:
> Есть еще аналог Сауреса - https://valtec.ru/catalog/pribory_ucheta/komplektuyuschie_dlya_priborov_ucheta/wi_fi_ustrojstvo_sbora_i_peredachi_dannyh.html
> Что-то типа выполненной по лицензии штуки. Там 8 каналов, как в старых Сауресах, а стоит дешевле. Они официальные, я уточнял у поддержки Саурес.

Спасибо! Да это и есть Саурес с другой наклейкой на корпусе. Инструкция сауресовская.

OloloevReal, 2019-11-20:
А такая штука уже попадалась? https://icbcom.ru/ru/tovar/modul-nb-iot-dlya-schetchikov-vody-itelma/
Цепляется сверху на счетчик, данные по NB-IOT использует самый обычный SIM7020E
Внутренности из документации:
[image]

dontsovcmc, 2019-11-20:
Вау! Очень круто! Как раз именно на днях увидел такую же поделку от net868 и у меня закралось подозрение, что они не разбирают счетчики... Держите внутренности Saures NB-IoT! Тоже самое, что на вашей картинке, но технологичнее! STM32 + BC68 + память на 8Мб + еще какая-то микросхема квадратная.

[image]
[image]
[image]

OloloevReal, 2019-11-21:
Квадратная микросхема это скорее всего распаянная сим-карта, слота-то нет
Интересно, это получается что он работает только с определенным оператором

dontsovcmc, 2019-11-22:
@OloloevReal а точно же. Да, у них МТС. Любопытно, что еще несколько месяцев назад цена была 6000, а сейчас 3700. При этом 6 лет абонентской платы тоже заложены, раньше это стоило 50*12*6 = 3600р.

Alexey-K77, 2020-06-19:
ещё у болида есть АРМ "РЕСУРС". Но это уровень не квартиры, а многоквартирного дома. сами устройства для считывания показаний есть как проводные , так и беспроводные. Беспроводные работают на батарее и  построены на микроконтроллере от TI со встроенной радиочастью на 868МГц и блоком ускорения AES.

OloloevReal, 2020-11-07:
Немного офтоп, ютьюб подсунул интересный ролик от устройства КУЗЯ про цели устройства https://youtu.be/AMEHBZWBYNQ

dontsovcmc, 2020-11-07:
@OloloevReal о, Кузя ожил, я думал они забили на проект. Спасибо за ссылку!

OloloevReal, 2021-02-07:
Аналогом это язык не повернется назвать, копия
https://oiot.ru/catalog/kits/kit-counter-water-wi-fi-110mm/

[image]

honechko, 2021-08-10:
Добавьте в табличку:
Название: Счетчик импульсов https://arduino.ua/prod4180-schetchik-impylsov
Связь: 1-wire 
Питание: от 1-wire / CR2032, до 5лет (зависит от датчиков)
Счетчиков, шт: 2, namur, датчики
Доп. опции: импульсное питание датчиков
Доп. информация: много настроек через web
Цена, руб: $7
Описание: https://habr.com/ru/post/568314/

7918514, 2022-04-02:
Вот такие есть ещё аналоги из Москвы
https://promodem.ru/produkty/nb-iot-gprs-loggery-dlya-pokvartirnogo-ucheta/

dontsovcmc, 2025-04-24:
> Аналогом это язык не повернется назвать, копия https://oiot.ru/catalog/kits/kit-counter-water-wi-fi-110mm/
> 
> [image]

спи*дили прошивку