dontsovcmc/waterius #13: English please???

nardev, 2018-06-29:
Hello,

it looks like i have same meter. I would love to see this project translated and if possible, little bit about how the measuring is being done.

can we expect that soon?

dontsovcmc, 2018-06-29:
Thanks for you interest!

Quick info:

This is impulses counter for water meters or same devices that transmitts current values through WiFi to Blynk.cc mobile application or you own TCP server.
It consists from Attiny85 and ESP8266-01.

Current when counting 15-25uA. When transmitting 80mA for ~5sec.

Press button to setup Wi-Fi settings.
ESP starts access point with ip 192.168.4.1
Write your wifi title and password, static ip (i don't use dhcp for fast connect), you router ip and etc.

nardev, 2018-06-29:
Thnx,

is it related in any way with "M-Bus cyble"?

dontsovcmc, 2018-06-29:
It connects to water meters by "dry contact" or discrete output. Every 10 litres contacts closed = 1 impuls.

nardev, 2018-06-29:
I understand, i just read some about "M-Bus" looks like similar purpose, but completely differently defined protocol. This is the one i have: https://www1.itron.com/local/Poland%20Product%20Portolio/GA-CYBLEMBUSSTD-01-EN-10-14.pdf

dontsovcmc, 2018-06-30:
You want to read meter data from M-bus protocol and transmit it through Wi-Fi? 
You should find M-bus library for ESP8266. ESP8266 can be in sleep mode (10uA) and wake up every 30 min (AA batteries live ~3 month), or turned off by external chip (you can find external timers for this aims).

nardev, 2018-06-30:
That about sleep mode and ESP is all fine, but i have no idea about m-bus. So far i only found two, old and not so well documented projects about this. Unfortunately i can't find some simplified docs about m-bus.

At first sight i thought this is m-bus project so i run to ask few questions.

Anyway, thank you very much.

p.s. here is what concerns me: [here](https://github.com/rscada/libmbus/issues/139)

closed 2018-07-17 by dontsovcmc