dontsovcmc/waterius #22: Показания 42949840

dontsovcmc, 2018-07-31:
Впервые пришли некорректные показания 42949840. Надо минимум сделать проверку.

в V4:
9.0,1532985720000,0 Monday, 30 July 2018 г., 21:22:00
0.0,1532987580000,0  Monday, 30 July 2018 г., 21:53:00
0.0,1532989440000,0 Monday, 30 July 2018 г., 22:24:00
4.294967295E9,1532991840000,0

в V3:
0.0,1532987580000,0
0.0,1532989440000,0
4.294967295E9,1532991840000,0

dontsovcmc, 2018-08-06:
Лог без attiny включенной ESP показывает ,что в памяти:
```
000:00:00:00:085  NOTICE    (WIF) : value0_start=168.73, impules0_start=40, factor=10
000:00:00:00:092  NOTICE    (WIF) : value1_start=208.92, impules1_start=136
000:00:00:00:099  NOTICE    (ESP) : new impulses=0 0
000:00:00:00:104  NOTICE    (ESP) : new values=42949840.00 42949880.00
```
в веб интерфейсе тоже 
42949840.00 
42949880.00

dontsovcmc, 2018-09-22:
```
2018-09-22 03:27:47.463385: 000:00:00:00:104  NOTICE    (CFG) : channel0_start=0.05, impules0_start=5, factor=10
2018-09-22 03:27:47.469860: 000:00:00:00:112  NOTICE    (CFG) : channel1_start=0.00, impules1_start=0
2018-09-22 03:27:47.474531: 000:00:00:00:118  NOTICE    (ESP) : new impulses=0 0
2018-09-22 03:27:47.480163: 000:00:00:00:123  NOTICE    (ESP) : new values=42949672.00 0.00
```

OloloevReal, 2019-05-13:
Как вариант, такие значения могут появляться в случае вычитания из меньшего значения большего

`(data.impulses0 - sett.impulses0_start)`

```
void calculate_values(Settings &sett, SlaveData &data, CalculatedData *cdata)
{

    LOG_NOTICE("ESP", "new impulses=" << data.impulses0 << " " << data.impulses1);

    if (sett.liters_per_impuls > 0) {
        cdata->channel0 = sett.channel0_start + (data.impulses0 - sett.impulses0_start) / 1000.0 * sett.liters_per_impuls;
        cdata->channel1 = sett.channel1_start + (data.impulses1 - sett.impulses1_start) / 1000.0 * sett.liters_per_impuls;
        LOG_NOTICE("ESP", "new values=" << cdata->channel0 << " " << cdata->channel1);
        cdata->delta0 = (data.impulses0 - sett.impulses0_previous)*sett.liters_per_impuls;
        cdata->delta1 = (data.impulses1 - sett.impulses1_previous)*sett.liters_per_impuls;
        LOG_NOTICE("ESP", "delta values=" << cdata->delta0 << " " << cdata->delta1);
    }
}
```

Возможен такой случай, что Attiny была прошита, а ESP осталась со старыми значениями и произойдет вычитание хранившегося в ESP значения sett.impulses0/1_start из 0, что и приведет к таким похожим числам 42949880.00 (uint32_t - uint32_t)

dontsovcmc, 2019-09-16:
Причина этого, когда inpulses0_start - которые были записаны в момент входа в настройку больше, чем inpulses пришедшие с attiny.

```
000:00:00:00:154  NOTICE    (CFG) : channel0_start=0.00, impulses0_start=5, factor=10
000:00:00:00:162  NOTICE    (CFG) : channel1_start=0.00, impulses1_start=8
000:00:00:00:168  NOTICE    (ESP) : new impulses=1 3
000:00:00:00:173  NOTICE    (ESP) : new value0=42949672.00 value1=42949672.00

еще позамыкал:
000:00:00:00:153  NOTICE    (CFG) : channel0_start=0.00, impulses0_start=5, factor=10
000:00:00:00:160  NOTICE    (CFG) : channel1_start=0.00, impulses1_start=8
000:00:00:00:167  NOTICE    (ESP) : new impulses=4 9
000:00:00:00:172  NOTICE    (ESP) : new value0=42949672.00 value1=0.01

еще позамыкал:
000:00:00:00:152  NOTICE    (CFG) : channel0_start=0.00, impulses0_start=5, factor=10
000:00:00:00:160  NOTICE    (CFG) : channel1_start=0.00, impulses1_start=8
000:00:00:00:166  NOTICE    (ESP) : new impulses=6 9
000:00:00:00:171  NOTICE    (ESP) : new value0=0.01 value1=0.01
```

cdata.channel0 = sett.channel0_start + (data.impulses0 - sett.impulses0_start) / 1000.0 * sett.liters_per_impuls;
cdata.channel0 = 0.0 + (1 - 5) / 1000.0 * 10; //получается 42949670 если сначала поделить на 1000, а потом умножить или 42949672 если поделить на 100. Видимо компилятор оптимизирует.

4294967295 - 0xffffffff
(4294967295 - 3) / 1000.0 * 10

Остается вопрос, откуда в середине работы может появится, как у меня в заголовке.

dontsovcmc, 2019-10-26:
Грешу на i2c. Добавление crc должно решить проблему.

dontsovcmc, 2020-05-16:
Проблема не проявлялась больше. Действительно, только при сбросе attiny, это может проявиться. CRC добавил в i2c с версии 12 attiny и версии 0.9.13 ESP, поэтому закрою задачу.

closed 2020-05-16 by dontsovcmc