dontsovcmc/waterius #30: Вопросы по использованию

simak-alexandr, 2018-08-07:
Помогите разобраться, пожалуйста.

1. Что именно нарисовано на графиках? 
Последние снятые показания счетчиков не очень совпадают с графиками.

2. Не приходит e-mail, хотя при настройке устройства я его указывал.

3. Надо ли кнопку "play" в приложении держать непрерывно включенной?

Мануалы по приложению не курил, просто поставил по инструкции.

Спасибо!

dontsovcmc, 2018-08-07:
Добавил в конце: https://github.com/dontsovcmc/waterius/blob/master/Setup.md

dontsovcmc, 2018-08-09:
Получилось?

simak-alexandr, 2018-08-13:
@dontsovcmc, почти; только после того, как в настройках компоненты графика выставил Time range picker в High Resolution - дни перестали "слипаться", и показания на графике теперь совпадают с виртуальными источниками V3, V4. 
В приложении я добавил эти источники на форму для наглядности (мне так проще отслеживать суточное потребление).

closed 2018-08-13 by simak-alexandr