dontsovcmc/waterius #37: Поддержка OTA

dontsovcmc, 2018-10-19:
Сервер есть. Помогите разобраться =)

dontsovcmc, 2018-11-05:
https://github.com/esp8266/Arduino/blob/13f374666dfb0225167640bf695e4a943dc55b51/libraries/ESP8266httpUpdate/examples/httpUpdateSecure/httpUpdateSecure.ino

dontsovcmc, 2018-11-15:
Вроде понятно, как сделать. А нужно ли?

dontsovcmc, 2019-02-25:
Скомпилировал пример, работает:
https://gist.github.com/dontsovcmc/a5ac114fca763217709318e7ce6ee84a
Пример сервера
https://gist.github.com/dontsovcmc/a60009697cf78cd36f6dd52122cf1bad

Можно размещать обновления на гитхабе:
github.com/repository/X/firmware.bin, X - номер следующей версии и железка будет обращаться к серверу.

OloloevReal, 2019-02-25:
С HTTPS, с Blink и JSON столкнулся с тем, что на OTA апдейты не хватало места, пришлось заменить флешки ESP8266 4Мбит на 32Мбит (25Q32FVSIG)

dontsovcmc, 2019-02-26:
Упс... н - наивность? =) я тут хочу https+ota+json в ESP-01 запихнуть... 
Там BearSSL сделали, кстати.

dontsovcmc, 2019-02-26:
```
ESP-01

[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 319488
[httpUpdate]  - current Sketch Size: 444656
```

(((

dontsovcmc, 2019-02-26:
Надо бы оптимизировать память.. Под скетч оказывается всего 761840 байт. 
Значит прошивку надо уместить в 380920 байт

Сейчас: 
JSON 14k
Blynk 10k 
WiFiManager 60k
логгирование 15k

+ OTA
+ BearSSL

OloloevReal, 2019-02-26:
Оптимизация вещь похвальная и нужная, но быстрее перепаять флешку - 70 рублей и 5 минут делов:)

dontsovcmc, 2019-02-26:
> Оптимизация вещь похвальная и нужная, но быстрее перепаять флешку - 70 рублей и 5 минут делов:)

Ага =)

dontsovcmc, 2019-02-28:
В attiny85 надо добавить команду "подожди, я обновлюсь" :/

OloloevReal, 2019-02-28:
Тут еще такой момент, после обновления ESP перезагрузится и спросит причину загрузки - т.е. добавить еще один режим #define UPGRADE_MODE №

OloloevReal, 2019-02-28:
Да и вообще с ОТА много проблем как выясняется, иногда, обычно при существенном изменении кода и размера прошивки, после обновления кораптятся настройки. Не плохо бы их сохранять куда-нибудь (spiffs ?) перед обновлением и восстанавливать после обновления.

dontsovcmc, 2019-03-01:
ESP ничего не спрашивает. Причину перезагрузки ESP можно добавить в отсылаемые данные для диагностики.

Чтобы настройки не изменялись, надо знать распределение памяти =(. Сейчас настройки в eeprom храняться. Но его мало для нескольких сертификатов =(.

dontsovcmc, 2019-03-26:
Вот, говорят, хороший чип W25Q64FV

popsodav, 2019-05-14:
Я перепаивал ESP-01 на 32мбит flash вот такой:
https://www.aliexpress.com/item/New-and-original-W25Q32BVSSIG-W25Q32BVSIG-W25Q32-25Q32/2042981414.html?spm=a2g0s.9042311.0.0.27424c4dogSpZx

Единственная проблема - чип чуть больше оригинального, приходится поджимать лапы.

Наверно, стоит ещё подумать об апдейте прошивки в тиньке?
Нашёл вот такую либу для этих целей:
https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266AVRISP

dontsovcmc, 2019-05-14:
@popsodav о, спасибо! мне кажется усилия по написанию прошивки под ESP32 эквиваленты, придумыванию как esp01 прошивать attiny будет. библиотека верная.

dontsovcmc, 2026-08-25:
Добавили для версии 2, т.к. версия ватериуса обычная с 1мб памяти - туда не входит 2 прошивки из-за SSL =(

closed 2026-08-25 by dontsovcmc