dontsovcmc/waterius #38: Вотериус.PhotoEdition

dontsovcmc, 2018-10-19:
Для счетчиков без выходов: ставим камеру 640x480, скажем ov7670 или аналог + подсветку. Отправляем фото раз в Х дней на сервер, где она распознается. Можно и на ESP распознавать, но тогда точно нужно OTA поддержать.

dontsovcmc, 2018-10-31:
https://habr.com/post/330936/

dontsovcmc, 2018-11-03:
http://www.gnu.org/software/ocrad/ocrad.html (http://www-e.uni-magdeburg.de/jschulen/)
https://people.eecs.berkeley.edu/~fateman/kathey/ocrchie.html
https://github.com/yaacov/hebocr
https://github.com/sanke93/BitmapOCR
https://github.com/specialityProjectEnsimag/OCR
https://github.com/mihneasim/schnell-ocr
https://github.com/TetsuakiBaba/ofxOcrad
https://github.com/PiotrKala/OCR
https://github.com/4pm-nomnom/OCR

dontsovcmc, 2018-11-04:
http://www.cnx-software.ru/2018/10/04/esp32-cam-%D1%8D%D1%82%D0%BE-%D0%BF%D0%BB%D0%B0%D1%82%D0%B0-%D0%B4%D0%BB%D1%8F-%D1%80%D0%B0%D0%B7%D1%80%D0%B0%D0%B1%D0%BE%D1%82%D0%BA%D0%B8-esp32-%D1%81-%D0%BA%D0%B0%D0%BC%D0%B5%D1%80%D0%BE%D0%B9/

https://www.banggood.com/M5Stack-Official-ESP32-Camera-Module-Development-Board-OV2640-Camera-Type-C-Grove-Port-p-1333598.html?cur_warehouse=CN

olegkyka, 2019-11-04:
https://www.youtube.com/watch?v=D6JG2DZeWeY

dontsovcmc, 2019-11-04:
> https://www.youtube.com/watch?v=D6JG2DZeWeY

Вау! Это требует подробного изучения. Я думаю по схожему принципу работает Элехант. Вопрос, как бы добиться автономности большой. И не будет ли внешняя среда как-то мешать подсчету. А так, очень здорово.

a-marvvv, 2021-07-14:
@dontsovcmc 
готовый проект на esp32-cam, про автономность еще стоит подумать, но очень достоин внимания
https://www.youtube.com/watch?v=iUgxwbfkIqU