dontsovcmc/waterius #48: Не тухнет светодиод

OloloevReal, 2018-11-17:
Добрый день!

В ряде случаев, не нашел закономерность, после отсылке данных индикаторный светодиод LED1 не тухнет до конца - остается слабое свечение. Если дотронутся до выхода TX ESP то светодиод тухнет.

Подобное поведение наблюдается на обоих сделанных устройствах, на каком-то чаще на каком-то реже

**Светодиод**: GNL-3014PGC - "Светодиод зеленый 25° d=3мм 8000мКд 525нМ (Green)",  
**ESP8266**: ESP01S с отпаянным резистором с вывода EN, светодиод Wi-Fi не удалял

[image]

dontsovcmc, 2018-11-17:
Добрый вечер!
У меня тоже не гаснет, но потребление было низким и я забил =). Светодиод яркий и несколько мкА уже видны. 
 Хм, надо добавить в код отключение Serial!
 спасибо!

OloloevReal, 2018-11-18:
Отключением Serial у меня не получилось добиться 100% результата, возможно делал что-то не правильно, плохо знаком с микроконтроллерами.

`Serial.end()`

Вышел из положения поставив шунтирующий LED1 резистор ~ 50кОм

dontsovcmc, 2018-11-18:
я вчерв посмотрел код Serial.end(), он переводит пин в Input. Но там куча условий, не проверял. А что тестер показывает? Мне кажется резистор не поможет. Это обычный ток утечки пина. 
Вы обычный светодиод поставите и не увидите свечения.
А попробуйте после Serial.end() ещё 
setMode(TX, OUTPUT), 
digitalWrite(TX, 0), 
setMode(TX, INPUT)

я думаю будет тоже самое. 

на ESP стоит синий светодиод, который тоже еле заметно горит в спящем режиме. но потребление 1мкА

OloloevReal, 2018-11-19:
Попробовал сделать вот так:

```
    Serial.end();
    pinMode(1, OUTPUT);
    digitalWrite(1, 0);
    pinMode(1, INPUT);
```

Не всегда, но в ряде случаев тусклое свечение светодиода остается после передачи показаний.

Схема с шунтирующим резистором решает проблему, никаких проблем после его установки не заметил. Как вариант, еще попробовать поставить светодиод синего цвета, говорят что они большего потребления и могут не загораться от микротоков, но у меня их нет в наличии что бы попробовать.

dontsovcmc, 2018-11-19:
А что микроамперметр показывает? с резистором и без

OloloevReal, 2018-11-20:
Микроамперметра у меня нет, померил тем что есть:

1) Под нагрузкой, во время отправки данных - **85 мА**
2) В режиме спокойствия ток - **0.02-0.03 мА**
3) С установленным шунтирующим LED1 резистором (50кОм) - **0.02-0.03 мА**

dontsovcmc, 2018-11-20:
Потрясающе! 
У меня ESP8266-01:
при горящем светодиоде мультиметр прыгает 10.7-14.4 мкА. 
если замкнуть TX через резистор 50к на GND, то светодиод гаснет и потребление 6.9-10.5 мкА.
на ESP8266-01S ситуация аналогичная.
Я пропустил целых 3мкА потребления! Хотя я мог мерять на другом светодиоде и возможно там такого потребления не было...
Спасибо! Надо добавить резистор на схему - во всех мануалах пишут, что все выводы должны быть подтянуты на + или -. TX остался обделён и вот результат.
Что скажешь, @IvKovalenko ?

vaz399, 2018-11-26:
Добрый день,
подскажите на ESP8266-01 красный светодиод должен гореть всегда?

dontsovcmc, 2018-11-26:
> Добрый день,
> подскажите на ESP8266-01 красный светодиод должен гореть всегда?

https://github.com/dontsovcmc/waterius/blob/master/Making.md "Примечания"

IvKovalenko, 2018-11-26:
Я не понял, что ты мне скинул.

dontsovcmc, 2018-11-26:
> […](#)

А это не я, а github, т.к. ты отмечен в треде, то он тебе шлёт все комментарии. Я ответил про светодиод 2-му пользователю.

dontsovcmc, 2018-12-13:
Развел большой тред по проблеме. Пока нет решения. Резистор 47к гасит светодиод, но потребление при этом остается прежним... 
[https://esp8266.ru/forum/threads/utechka-3mka-pri-deep-sleep-i-en-podtjanutom-k-zemle.3848/](https://esp8266.ru/forum/threads/utechka-3mka-pri-deep-sleep-i-en-podtjanutom-k-zemle.3848/)

heX16, 2020-12-25:
Светодиоды продолжают слабо светится. (покупал гаджет пару недель назад)

dontsovcmc, 2020-12-25:
> Светодиоды продолжают слабо светится. (покупал гаджет пару недель назад)

Это нормально. Там сверхяркий диод и ЕСП почему то даёт утечку 3мкА.

dontsovcmc, 2021-02-26:
@OloloevReal Прилепил 2 конденсатора - свечение светодиода прекратилось.
[image]

dontsovcmc, 2021-03-28:
Чем дальше в лес, тем толще партизаны.
https://esp8266.ru/forum/threads/vysokoe-potreblenie-esp-pri-en-low.5568

[9b-esp8266-low_power_solutions__en.pdf](https://github.com/dontsovcmc/waterius/files/6218192/9b-esp8266-low_power_solutions__en.pdf)

Пока не понятно..

dontsovcmc, 2021-03-29:
Если отключить логирование, то светодиод моргнет только загрузочным сообщением (74400) и погаснет на время работы программы. Однако после deepsleep он включится тусклым светом! 
https://esp8266.ru/forum/threads/vysokoe-potreblenie-esp-pri-en-low.5568

neitri, 2022-10-06:
Вспомнил что то про светодиод. Возможно что проблема с паразитным питанием. т.к. i2c интерфейс запитан постоянно. можно попробовать согласовать уровни i2c на полевом транзисторе. Что даст возможность отключать шину со стороны ESP. Конденсаторы по RST и  CH_DP возможно и не нужны, но прикинул емкость по описанию #153 
[image]

dontsovcmc, 2022-10-06:
> Вспомнил что то про светодиод. Возможно что проблема с паразитным питанием. т.к. i2c интерфейс запитан постоянно. можно попробовать согласовать уровни i2c на полевом транзисторе. Что даст [...]

А можно ли это проверить просто разомкнув i2c после информационного обмена? Кажется что нет, там внутри может быть логика зависимая от уровней напряжения пинов.

dontsovcmc, 2023-01-23:
Кажется решение проблемы - подключить светодиод к плюсу!

"ENGIN33RRR
Вот вам и проблема. Этот светодиод подключен между gpio и питанием

естественно если включить еще один светодиод- будет слабо светится тот у которого напряжение падения меньше

Как решение- подключать его так-же к плюсу

либо выдрать и второй светодиод с есп"

[image]