dontsovcmc/waterius #51: Подключение датчика протечки

OloloevReal, 2019-02-22:
Добрый день!

Возможно предусмотреть на новой ревизии платы место под установку компонентов для подключения датчика протечки к пину RESET?

Например вот так:
[image]

- Добавить пин GND для подключения непосредственно датчика к RESET/GND
- Добавить ограничивающий резистор R12 в цепь датчика 
- Добавить джампер или DIP свитч для отключения pullup резистора R5 от RESET Pin. При прошивки Fuse RSTDISBL он становится лишним

OloloevReal, 2019-02-22:
Пробовал совместно с R5 отключать и C3, но плата ведет себя не стабильно - перезагружается при касании RESET. В МК понимаю чуть больше чем ничего, но думал что при прошивки RSTDISBL в нем необходимость так же отпадет

dontsovcmc, 2019-02-22:
RESET не получится задействовать, т.к. он "не ресетом" может быть до половины уровня питания. Т.е. слишком "скользкое" решение будет. Гораздо проще и лучше заменить attiny85 на attiny84! https://github.com/dontsovcmc/waterius/issues/41#issuecomment-439402464
Надо только плату перерисовать. Вы сможете с этим помочь?
Я сейчас заказал 2ю версию платы, чтобы она по длине как АА был - меньше резать корпус. Вот на ее основе можно развести дорожки под attiny84, добавив разъем для протечки.

OloloevReal, 2019-02-22:
У меня с датчиком на RESET работает, пока без замечаний.
Хотел попробовать сделать с attiny84, никакой мороки с HVProg, но не нашел их в свободной продаже

dontsovcmc, 2019-02-22:
Ого! Крутяк! А какое сопротивление при затоплении водой датчика? Как часто проверяете его? Это же чтение аналогового входа, значит и компаратор надо включать при пробуждении? 8 раз в секунду проверять смысла нет... раз в минуту бы хорошо.

Смотрите, то, что плата перезагружается от касания - это более чем нормально и штатно. Т.к. она работает в deep sleep и потребляет <10мкА, то любое внешнее воздействие, а особенно человеком - электрошокер для нее. Но конденсатор отключать плохо, он нужен как раз для стабильности. Резистор от reset можно отключить в принципе. Он рекомендательный.

А код покажете? Может сделать отдельную ветку под датчик протечки?

p.s. в blynk передается кол-во перезагрузок и причина последней перезагрузки.

OloloevReal, 2019-02-23:
Сопротивление получается что-то около ~50 кОм + защитный резистор

Компаратор как я понял из кода выключен почти все время что бы уменьшить потребление, поэтому сделал вот так:

```
	pinMode(INPUT2_PIN, INPUT_PULLUP);
	delayMicroseconds(5000); //Charging C2, 5ms
	uint8_t waterSensorState = digitalRead(INPUT2_PIN);
	pinMode(INPUT2_PIN, INPUT);
```

На самом деле не до конца понимаю почему происходит переключение уровней, по спецификации при питании от 3V низким уровнем считается напряжение ниже 0.9V, а 50кОм такого падения не дадут.

[image]

Опрос сенсора вставил в основной цикл, каждые ~7-8 секунд, плюс защита от ложных срабатываний

[image]

dontsovcmc, 2019-02-23:
Вот тут другая версия даташита... https://electronics.stackexchange.com/questions/195146/use-the-reset-pin-on-attiny85-as-input-with-voltage-divider в конце документа обычно указывают версию и изменения.

PULLUP резистор 47к, поэтому получается 0.5*3v = 1.5v на входе - 0. 
Во-первых, резистор 47к можно установить штатно - его нет на входах счетчика, т.к. счетчики могут остановится в момент замыкания входа и будет высокое потребление. Для датчика протечки это не актуально - протечка нештатное событие и длится конечный период. Значит, если я правильно понял, можно убрать `delayMicroseconds(5000);` 

Я не пойму, а почему 3к3 вы на GND подключили, а не на вывод датчика к Reset? 

Я давно читал даташит, напомните мне и читателям, почему от протечки не перезагружается attiny85?

OloloevReal, 2019-02-23:
> Я давно читал даташит, напомните мне и читателям, почему от протечки не перезагружается attiny85?

Прошит Fuse RSTDISBL

[image]

avrdude -p t85 -c linuxspi -b 16000 -B 4 -P /dev/spidev0.0 -U efuse:w:0xFF:m -U hfuse:w:**0x5F**:m -U lfuse:w:0x62:m

После этого, перепрошить attiny можно будет только сбросив фьюзы высоковольтным программатором. [https://habr.com/ru/post/249967/](url)

dontsovcmc, 2019-02-25:
Отлично! Но, все равно Attiny85 перезагружается если пальцем теребонькать пин, верно?

А как выглядит сам датчик протечки? Я видел несколько вариантов в энторнетах... Я предполагаю, что  самый простой способ - вытравить змейку на плате, но надо чем то покрыть от коррозии. Есть готовое что-то?

Если сделать чтение аналогового сигнала - это позволит разместить на датчике резистор для детектирования разрыва связи с ним! Если не делать, то можно на датчик припаять кнопку "ТЕСТ"+резистор50к и проверить при установки, что всё ОК.

OloloevReal, 2019-02-25:
> Отлично! Но, все равно Attiny85 перезагружается если пальцем теребонькать пин, верно?

Нет, не перезагружается, работает стабильно, конденсатор С2 сглаживает всплески от касаний. Но для его зарядки приходится вставлять паузу delayMicroseconds(5000); перед digitalRead(INPUT2_PIN);

Из готовых датчиков должен подойти WSP [https://gidrolock.ru/production/datchiki-protechki-vody-gidrolock/](url)  

Пробовал делать с аналоговым чтением, но так до конца и не разобрался, у меня после нескольких циклов чтения analogRead() получались одни и те же значения = 338, как будто ADC зависал или выключался. Как правильно разбудить attiny включить и настроить adc курить даташит надо

dontsovcmc, 2019-02-25:
> Но для его зарядки приходится вставлять паузу delayMicroseconds(5000); перед digitalRead(INPUT2_PIN);
Так я предлагаю впаять между RESET и Vcc резистор на 50к.

> WSP хех 500руб =). да, позолоченные контакты норм.
ого! У них заключение от [renata](https://gidrolock.ru/uploads/Lifetime%20calculation_Scanti_2450N_Gidrolock_130628.pdf) о работе их датчиков беспроводных. 2.5мкА - очень не дурно.

> Пробовал делать с аналоговым чтением, но так до конца и не разобрался

Так, нужен большой резистор параллельно датчику ~200к между RESET - GND

Алгоритм:
```
unsigned int check_leak_level(uint8_t pin) 
{
unsigned int level = 0;
adc_enable();
pinMode(INPUT_PULLUP, LEAK_PIN);
//нужно ждать зарядки конденсатора, если параллельно ему теперь резисто?
level = analogRead(LEAK_PIN);
adc_disable();
return level;
}

unsigned int noleak_value;

в setup() 
noleak_value = check_leak_level(LEAK_PIN) - SOME_VALUE; //запоминаем текущий уровень + чуток отступить 
```

в цикле опроса:
```
я бы добавил проверку 1 раз в минуту [сюда](https://github.com/dontsovcmc/waterius/blob/master/Attiny85/src/main.cpp#L169)

info.leak = check_leak_level(LEAK_PIN) < noleak_value;
if (info.leak) {
    break;
} 
```

OloloevReal, 2019-03-05:
Остановился пока на такой реализации:
[image]

При погружении в воду ток через датчик протекает 0.02mA, на входе RESET напряжение падает с 3.15 до 0.5В (в схеме стоит стабилизатор на 3.3В, что был в наличии)

Соответственно всякие задержки на зарядку C3 можно убрать, оставив только

```
        #define INPUT2_PIN  5	

        pinMode(INPUT2_PIN, INPUT);
	uint8_t waterSensorState = digitalRead(INPUT2_PIN);
```
Сенсор наколхозил из того что было в леруамерлен:
[image]

[image]

dontsovcmc, 2019-03-05:
Очень круто!! Мои поздравления! И конструкция датчика вдохновляет!
Хм. А входы разве не 1-2 ? ) 
Я закину фото в https://www.facebook.com/waterius?

OloloevReal, 2019-03-05:
> Я закину фото в https://www.facebook.com/waterius?

Конечно

OloloevReal, 2019-04-09:
В коллекцию фотографий, готовый комплект для установки в квартиру с двумя стояками

[image]

dontsovcmc, 2019-04-10:
Ваааау! Класс! 
Я сейчас НАМУР добавляю и упрощаю интерфейс настройки. Придумал показывать в вебморде, Есть ли связь со счетчиками - у кого контакты в пыли или плохо всовываются актуально.

dontsovcmc, 2019-04-10:
@OloloevReal  кстати , под датчик протечки можно Eventor использовать - Push уведомление.
Я планирую добавить вашу схему в новую версию платы. А, кстати, analogRead возможен для reseta? Я исследовал время опроса - analogRead лишь чуть чуть медленнее digitalRead.

Xa6ep, 2019-12-07:
> В коллекцию фотографий, готовый комплект для установки в квартиру с двумя стояками
> 
> [image]

А можно фото платы? Хочется понять как вы разводили это на плате. R12 наверное в корпус самого датчика можно засунуть...

OloloevReal, 2019-12-22:
Уф, пришлось опять лезть в сантехнишу что бы вспомнить что и как переделывал
Итого: R5 заменен на 210k, а R12 нет, использую землю GND1.

[image]
[image]

Белый герметик для фиксация кабеля.

PS. При таком использовании прошивка ATtiny будет не возможна без высоковольтного программатора, это важно.

dontsovcmc, 2019-12-22:
@OloloevReal а в прошивке опрос PB5 как обычного INPUT? ну и пробуждение по замыканию, верно?

OloloevReal, 2019-12-22:
Так точно

```
#define INPUT2_PIN  5

pinMode(INPUT2_PIN, INPUT);
uint8_t waterSensorState = digitalRead(INPUT2_PIN);
```

dontsovcmc, 2021-11-09:
Появилась идея - подключать датчик протечки параллельно счётчику воды (если он ГЕРКОН). 
Минусы:
 нельзя будет детектировать короткое замыкание, но вроде это совсем исключительная ситуация.
 не сделать поддержку НАМУР, потому что "протечка" бывает разного сопротивления.

closed 2026-08-25 by dontsovcmc