dontsovcmc/waterius #61: Blynk - incorrect delta values

OloloevReal, 2019-03-21:
Добрый день!

Обратил внимание на то, что в Blynk отправляется разница показаний за интервал измерения:

```
        unsigned int delta0 = (channel0 - sett.channel0_previous)*1000;  // litres
        unsigned int delta1 = (channel1 - sett.channel1_previous)*1000;
```

В результате такого приведения типов 2 импульса преобразуются не в 20, а в 19, 5 импульсов в 49 и т.д.

Возможно, стоит сделать вот так:

```
        unsigned int delta0 = (channel0 - sett.channel0_previous)*1000 + .5;  // litres
        unsigned int delta1 = (channel1 - sett.channel1_previous)*1000 + .5;
```

dontsovcmc, 2019-03-21:
Поясните подробнее =)
5 импульсов: дельта будет 0.05 * 1000  = 50. Откуда 49?

p.s. +.5 - если 10л/имп.

OloloevReal, 2019-03-21:
> 5 импульсов: дельта будет 0.05 * 1000 = 50. Откуда 49?

Преобразование float 50.00 в int дает результат 49, не всегда правда, в большинстве случаев.

dontsovcmc, 2019-03-21:
Ааа вы про вычислительные потери.
Так может переписать: unsigned int delta0 = channel0*1000 - sett.channel0_previous*1000  ?

OloloevReal, 2019-03-22:
Либо так:

```
        unsigned int delta0 = roundf((channel0 - sett.channel0_previous) * 100 * sett.liters_per_impuls);  // litres
        unsigned int delta1 = roundf((channel1 - sett.channel1_previous) * 100 * sett.liters_per_impuls);
```

dontsovcmc, 2019-03-22:
Некорректная формула. в channel0 хранятся кубометры. Мою используйте и напишите о результатах испытаний, пожалуйста.

OloloevReal, 2019-03-23:
`unsigned int delta0 = channel0*1000 - sett.channel0_previous*1000`
Так преобразование происходит корректно

> Некорректная формула. в channel0 хранятся кубометры. Мою используйте и напишите о результатах испытаний, пожалуйста.

Не понял, почему некорректная формула? в channel0/channel1 разве только кубометры? По коду смотрел вроде это показание счетчика со знаками после запятой, т.е. с литрами

dontsovcmc, 2019-03-23:
Потому что в channel0 хранятся кубометры. 
`*channel0 = sett.channel0_start + (data.impulses0 - sett.impules0_start) / 1000.0 * sett.liters_per_impuls;`

Тут ошибка не накапливается, т.к. каждый раз пересчёт.
Кстати, мы можем хранить предыдущее значение _импульсов_ и рассчитывать литры вообще без потерь. Напишите код?

dontsovcmc, 2019-03-23:
Написал сам =)

dontsovcmc, 2019-03-26:
Проверяйте. Теперь без делений. Только умножаем на кол-во литров/импульс.

closed 2019-03-26 by dontsovcmc