dontsovcmc/waterius #64: Отправка показаний ГДЕ-КУДА-КАК

dontsovcmc, 2019-03-29:
### Друзья! Напишите в какой сайт/приложение вы передаёте показания! Вы увеличите полезность и применимость Ватериуса!
(контакт: телеграм канал waterius)

г. Екатеринбург
https://frisbi24.ru/

г. Новосибирск
https://narod.nskes.ru/ электричество 
https://www.gorvodokanal.com/fdebts/login.php вода 

г. Санкт-Петербург
https://lk.kvartplata.info/LK/Home/Login

г. Лыткарино. УК Комфорт
http://comfortuk.ru

г. Москва
[x] https://www.mos.ru/services/catalog/popular/

Подмосковье
https://мособлеирц.рф

dontsovcmc, 2019-04-01:
ААА Уже всю работу кто-то проделал: http://schetchik-pokazanie.ru/

dontsovcmc, 2019-04-29:
Санкт-Петербург
https://lk.uksn.ru/index.php

Xa6ep, 2019-07-09:
Ростов-на-Дону
https://xn--c1afjenhpy1e1a.xn--p1ai/authorization
https://vodokanalrnd.ru/fastpay/send.php

DimsterTim, 2019-09-04:
Нужна интеграция с [HomeAssistant](https://www.home-assistant.io)

dontsovcmc, 2019-09-04:
> Нужна интеграция с [HomeAssistant](https://www.home-assistant.io)
Уже есть: https://github.com/dontsovcmc/waterius/issues/86
Будет круто, если поможете с инструкцией по настройке!

DimsterTim, 2019-09-04:
Как я понял, предложенная интеграция была расчитана на встроенный MQTT брокер. Но этот функционал в HomeAssistant посчитали "велосипедом" и несколько версий как не поддерживают, а далее планируют убрать. Так что вся интеграция сводится к уже имеющемуся функционалу отправки данных на mqtt сервер. А в настройках HA в файле yaml уже вставляется предложенный код для "датчиков" с подпиской на топики mqtt в формате:
```
sensor:
  # waterius на кухне
  - platform: mqtt
    name: "Kitchen Cold Water"
    state_topic: "waterius/kitchen/ch0"
    value_template: "{{ value }}"
    unit_of_measurement: "м3"
    icon: mdi:water
```
где sensor - тип "датчика" в терминах НА
platform: mqtt - формат данных для датчика
name: - отображаемое название датчика
state_topic: - тема для подписки на обновления данных mqtt
value_template: - формат парсинга полученного от mqtt значения, в данном случае просто берется "значение" как есть, но может быть много вариантов, JSON, вычисления и тд.
unit_of_measurement: "м3" - очевидно, что единицы измерения для показаний
icon: - иконка из набора НА.

Таким образом для интеграции Waterius в НА вы настраиваете отправку данных на mqtt по имеющейся [инструкции для mqtt](https://github.com/dontsovcmc/waterius/blob/master/Mqtt.md). В НА нужно будет добавить в файл конфигурации .yaml "датчик" настроенный на обновления значений от mqtt сервера в формате, разобранном выше. Затем, в НА датчик "Kitchen_Cold_Water" используется для отображения данных или в алгоритмах автоматизации, [построении графиков](https://www.home-assistant.io/lovelace/sensor/).

ddark008, 2019-09-29:
Ленобласть:
https://lk.epd47.ru/Account/LogOn

dontsovcmc, 2019-12-15:
Коллеги, для уточнения возможности добавления УК на сайт waterius.ru, напишите в contact@waterius.ru или телеграм @dontsovcmc данные для входа в личный кабинет.

dontsovcmc, 2019-12-15:
@DimsterTim т.е. HomeAssistant интегрируется без mqtt брокера и ничего кроме HomeAssistant не надо настраивать? Вы могли бы написать инструкцию, уверен она сэкономит многим время! Спасибо!

DimsterTim, 2019-12-16:
@dontsovcmc Home Assistant (HA) интегрируется через mqtt. Собственно инструкция уже написана, там настройка HA на mqtt. Ничего другого не нужно.

dontsovcmc, 2019-12-23:
@ddark008 нужны данные для входа на сайт.

kcinIvan, 2020-03-18:
Привет, а как передаются данные на mos.ru? Какой нибудь скрипт на python и request?
Дело в том что вывел показания счетчика электроэнергии в homeassistant (Pzem-004t -> esp8266 -> mqtt -> hass), вот бы их теперь передавать автоматически. На счётчики воды планирую  waterius поставить. 
Может есть возможность сделать интеграцию для hass которая через сайт waterius будет передавать показания? Или через hass-> mqtt -> устройство waterius - >...

dontsovcmc, 2020-03-18:
Привет, @kcinIvan ! Передаются от лица приложения на телефоне. Питон скрипт, глянь мои репозитории. Да, я могу добавить поддержку электрического счетчика на сайт Ватериуса, но для пары гиков это очень накладно делать =(. Тем более, что Pzem-004t не понятно с какой погрешностью считает..  Вам будет проще запускать самому Питон скрипт со своими данными для отправки показаний.

kolobus, 2020-10-02:
1. ЕИС ЖКХ (C-300) - очень популярная коммерческая система для УК. Есть ЛК в вебе и мобильное приложение. 
Можно ещё у них напрямую спросить, может есть отдельный API даже.

2. Отправка на электронную почту прямо УК. У многих УК есть адрес куда можно отправлять показания, но это должен быть именно второй адрес и без подтверждения.

dontsovcmc, 2020-10-02:
> 1. ЕИС ЖКХ (C-300) - очень популярная коммерческая система для УК. Есть ЛК в вебе и мобильное приложение.
>    Можно ещё у них напрямую спросить, может есть отдельный API даже.
> 2. Отправка на [...]

1. Когда я начал делать проект, то тоже подумал, покрыть максимум, а дальше как пойдет. Потом понял, что лучше другая модель - предложить клиентам самим выбирать, что добавить. Пока никто еис жкх не просил добавить.

2. Уже реализовали.

svesh, 2021-01-16:
> 1. Когда я начал делать проект, то тоже подумал, покрыть максимум, а дальше как пойдет. Потом понял, что лучше другая модель - предложить клиентам самим выбирать, что добавить. Пока никто еис жкх не [...]

Как раз появилась потребность в отправке в C-300. Отправлю запрос через кабинет.

dontsovcmc, 2021-03-27:
Тему закрываю. Нет потребности интегрировать все подряд. Сейчас добавляем по заявкам пользователей Ватериуса. Пока бесплатно. Дальше видно будет. Всем спасибо за участие =).

closed 2021-03-27 by dontsovcmc