dontsovcmc/waterius #65: СЧЁТЧИКИ ВОДЫ

dontsovcmc, 2019-03-31:
Давайте разбираться, какие счётчики воды подходят для Ватериуса и где выгоднее их купить!

ITELMA WFK 24.D080 ГЕРКОН
ITELMA WFW 24.D080 ГЕРКОН
ITELMA WFK 24.D011 ГЕРКОН
ITELMA WFW 24.D011 ГЕРКОН
[паспорт](https://www.i-bs.ru/files/tehnicheskie_opisaniya/TO_ITELMA-2015-12.pdf)
Не указывается Намур или Геркон, нужно уточнять в магазине, чтобы было написано Геркон.

Берегун.01.15.080
Берегун.02.15.080
Берегун.01.15.110
Берегун.02.15.110
[паспорт](http://www.tehnopostavka.ru/kipia/techdocs/beregun/passport_beregun_01.pdf)
Не указывается Намур или Геркон, нужно уточнять в магазине, чтобы было написано Геркон.

dontsovcmc, 2019-04-05:
Вот, что поддерживает Саурес:

ITELMA | WFK23, WFW23, WFK24, WFW24
VALTEC | VLF-15U-I
ZENNER | ETK, ETW, MTK, MTW
MINOMESS | FAZ
АЛЕКСЕЕВСКИЙ | СХИ, СГИ
ВОДОМЕРЪ | СХИ, СГИ
БЕРЕГУН | 01.15
ПУЛЬС | 15УИ
ПУЛЬСАР |  
ОХТА | ГЛ 15 И
ENBRA | EV-AMi, EV-AM1i
НОРМА | СВКМ-15УИ
METER | СВУ-15И
GERRIDA | СВК-15ГИ, СВК-15ГМИ

dontsovcmc, 2019-04-19:
Королевский список всех возможных счетчиков на рынке РФ с интерфейсами: https://www.tinko.ru/upload/ftp/ftpextrafiles/docs/C/8/C857E99DB9C11F794BC27FB837CE79B0.pdf

Trogl, 2023-11-19:
Itelma геркон или НАМУР  можно определить по модели
WFW 20.D080 - без импульсного выхода
WFW 24.D080 - Геркон
WFW 23.D080 - НАМУР
Так написано на коробке от счетчика

dontsovcmc, 2025-10-20:
Перенесли список на наш сайт: https://faq.waterius.ru/article/23088