dontsovcmc/waterius #66: Счётчики электроэнергии

dontsovcmc, 2019-04-02:
Ватериус не поддерживает счётчики электроэнергии. 
Но это реально доделать, поэтому соберем марки с RS485 портом и узнаем потребность от вас.

СЭБ-2А.07Д.212
СЭБ-2А.07Д.212.1 http://electroschetchik.ru/seb-2a.07d.php
МАЯК 101АТД.132Ш.2ИО2Б RS-485 http://electroschetchik.ru/mayak%20101atd.php

flystyle, 2019-04-04:
потребность есть
Меркурий 230 АRT http://electroschetchik.ru/m230art.php

dontsovcmc, 2019-04-04:
1. Мне не понятно, имеют ли владельцы доступ к RS485 счетчиков - они же запломбированы?
Меркурий, добавим спасибо.

flystyle, 2019-04-05:
Везде по разному, кто-то даёт доступ, кто-то нет, зависит от дремучести местного начальника электросетей. У меня счётчик с IrDA и подключён к ПК, но планирую переходить на RS485, заодно проверю, как сейчас дела обстоят с доступом к опломбированным интерфейсам.

spqer, 2019-04-17:
SDM630 - трехфазный
SDM220 - однофазный.
Потребность есть, заказчики готовы ставить второй счетчик после опломбированного. В Питере ленэнерго все информационные выходы на своих счетчиках опечатывает

dontsovcmc, 2019-04-17:
@spqer, а есть какие нибудь законодательные акты, что пользователь имеет право подключаться к информационным выходам? Это похоже на то, что Ленэнерго не хочет морочиться с разграничением доступа - нет сил, знаний, желания. Может есть какой то закон, который говорит, что "пользователь имеет право читать показания" со своего электросчетчика?
На подобии как "владелец жилья имеет право иметь ключи от крыши".

spqer, 2019-04-17:
@dontsovcmc Когда у себя дома делал ремонт, представитель Ленэнерго которая опечатывала счетчик сослалась на свои внутренние инструкции. Я не уверен в их законности, но место в щите под еще один автомат найдется. Спасибо китайцам за маленькие счетчики)

Обычно, когда говоришь заказчику что придется как-то договариваться с нашим государством, вариант не терять время поставив доп устройство всех устраивает.

Xa6ep, 2019-07-14:
А нельзя сделать с фоторезистором вариант, как у [LAVElek](https://habr.com/ru/post/428844/)?

dontsovcmc, 2019-07-16:
@spqer Мастер из Мосэнергосбыта сказал, что запрещается пользователям подключаться к интерфейсу, т.к. админские пароли на Меркурий 200 известны всем и пользователь может тарифы поменять, чтобы меньше платить..

@Xa6ep да, это рабочий вариант. В интернете есть отличный проект на ESP.

olegkyka, 2020-02-23:
> @Xa6ep да, это рабочий вариант. В интернете есть отличный проект на ESP.

Если не затруднит киньте ссылку почитать.

dontsovcmc, 2020-02-24:
> > @Xa6ep да, это рабочий вариант. В интернете есть отличный проект на ESP.
> 
> Если не затруднит киньте ссылку почитать.
https://esp8266.ru/forum/threads/distancionnoe-snjatie-pokazanij-ehlektroschetchika-merkurij.2266/

shum44, 2020-03-04:
посмотри как здесь сделали
https://sprut.ai/client/article/2184

spoterianski, 2020-03-04:
http://incotex-counter.blogspot.com/2012/07/blog-post.html

dontsovcmc, 2020-06-02:
Step UP преобразователи 
https://www.kirich.blog/stati/informaciya-dlya-nachinayuschih/136-para-step-up-konverterov-i-ih-nebolshoy-apgreyd-do-sepic.html

MT3608
-------
https://robotics.nethouse.ru/static/doc/0000/0000/0357/357325.ijcup4dklt.pdf с пином EN и 0.1мкА утечки. Кажется он тут: 
https://www.aliexpress.com/item/33009040632.html  20 руб
MT3608 3608 B6288

Shagrat2, 2023-04-13:
Могу помочь с iek61107 - Счётчики Энергомера
Доступ к пинам с верхней крышки счётчика (не пломбируют). RS485 только отдельные модели.

Делал для:
HomeAssistent: https://github.com/Shagrat2/hassio-addons/tree/master/energomera-iek61107
MajorDomo: https://github.com/Shagrat2/majordomo-energomera

dontsovcmc, 2023-09-15:
https://github.com/dontsovcmc/waterius/discussions/290

closed 2023-09-15 by dontsovcmc