dontsovcmc/waterius #71: Проверить, не включается ли точка доступа при передаче

dontsovcmc, 2019-04-06:
есть ощущение, что точка доступа включается и при включении передачи показаний. проверить.

OloloevReal, 2019-04-06:
Скорее всего включается, много раз наблюдал вот такую картину

[image]

Где значение в HEX после ESP_ уникальный ChipID ESP

Использую девелоперскую ветку [WiFiManager](https://github.com/tzapu/WiFiManager/tree/development)

OloloevReal, 2019-04-07:
Возможно, нужно добавить 

`WiFi.mode(WIFI_STA);`
перед
` WiFi.begin();`

[#833](https://github.com/tzapu/WiFiManager/issues/833#issuecomment-466580488)

dontsovcmc, 2019-04-07:
Хммм Спасибо!
Я такой космос сейчас делаю - опрос attiny во время работы вебсервера, чтобы сразу видеть, подключил ли провод или нет. И в следующей прошивке будет поддержка Намур. Ватериус сам разберется, что к нему подключили.

dontsovcmc, 2019-04-07:
> Возможно, нужно добавить
> 
> `WiFi.mode(WIFI_STA);`

Так у меня же добавлена: 
https://github.com/dontsovcmc/waterius/blob/master/ESP8266/src/main.cpp#L78

dontsovcmc, 2019-04-14:
Я сейчас hotfixes использую и у меня не появляется ESP_xxx  в списке сетей.

dontsovcmc, 2019-05-04:
@OloloevReal ошибка подтверждается? у меня поиском сетей во время включения Ватериуса ничего не находит...

OloloevReal, 2019-05-05:
@dontsovcmc у меня была ошибка с закомментированной строчкой WiFi.mode(WIFI_STA);, тянулась откуда-то из ранних версий + dev версия WifiManager (позволяет сделать темную тему оформления:) )
После того, как раскомментировал WiFi.mode(WIFI_STA); в списке сетей ничего подозрительного не видел.

dontsovcmc, 2019-05-05:
Ура! =)

closed 2019-05-05 by dontsovcmc