dontsovcmc/waterius #74: Долгая синхронизация времени

dontsovcmc, 2019-04-11:
Бывает быстро, бывает долго..

```
000:00:00:01:923  NOTICE    (NTP) : Waiting for NTP time sync: 
000:00:00:16:925  WARNING   (RQT) : SetClock fail ???
000:00:00:16:926  INFO      (SND) : Send HTTP code:     0
000:00:00:16:926  INFO      (SND) : Send result:        Error
000:00:00:16:929  NOTICE    (SND) : -- END --
000:00:00:16:987  NOTICE    (CFG) : Config stored OK
000:00:00:16:988  NOTICE    (ESP) : Going to sleep
```

OloloevReal, 2019-04-11:
Использование пула РФ похоже улучшает ситуацию

`configTime(3 * 3600, 0, "ru.pool.ntp.org", "time.nist.gov");`

А еще можно выкинуть вообще NTP и забирать время по HTTP например, прецизионная точность здесь не нужна

dontsovcmc, 2019-04-11:
У меня так и написано:
configTime(0, 0, "ru.pool.ntp.org", "time.nist.gov");

Он нужен для https.

OloloevReal, 2019-04-11:
Daytime вместо NTP можно использовать https://gist.github.com/OloloevReal/4a5f0137a2af5f09e46f41f6efdb990d

К сожалению получилось только tcp сделать, по udpсервер ничего не отвечает, хоть и должен https://tools.ietf.org/html/rfc867

dontsovcmc, 2019-04-11:
Воу! Крутяк. 
Он на http тоже отвечает. Но там троттлинг стоит, что нельзя часто спрашивать.
Мда.. надо свой прокси делать =(

OloloevReal, 2019-04-21:
C daytime так себе идея получилась, работает шустро, но как оказалось сервера NIST не всегда отвечают временем с датой, а просто молчат

Поставил перед запросом NTP проверку доступности по icmp, на случай если запрос синхронизации посылается до момента появления "нормального интернета"
Даже в этом случае бывает что синхронизация длиться по 30 секунд

```
000:00:00:04:231  NOTICE    (PING) : Success!
000:00:00:04:232  NOTICE    (SNTP) : Waiting for NTP time sync:
000:00:00:04:732  NOTICE    (SNTP) : .
000:00:00:05:233  NOTICE    (SNTP) : .
000:00:00:05:733  NOTICE    (SNTP) : .
000:00:00:06:234  NOTICE    (SNTP) : .
000:00:00:06:734  NOTICE    (SNTP) : .
000:00:00:07:234  NOTICE    (SNTP) : .
000:00:00:07:735  NOTICE    (SNTP) : .
000:00:00:08:235  NOTICE    (SNTP) : .
000:00:00:08:736  NOTICE    (SNTP) : .
000:00:00:09:236  NOTICE    (SNTP) : .
000:00:00:09:737  NOTICE    (SNTP) : .
000:00:00:10:237  NOTICE    (SNTP) : .
000:00:00:10:737  NOTICE    (SNTP) : .
000:00:00:11:238  NOTICE    (SNTP) : .
000:00:00:11:738  NOTICE    (SNTP) : .
000:00:00:12:239  NOTICE    (SNTP) : .
000:00:00:12:739  NOTICE    (SNTP) : .
000:00:00:13:239  NOTICE    (SNTP) : .
000:00:00:13:740  NOTICE    (SNTP) : .
000:00:00:14:240  NOTICE    (SNTP) : .
000:00:00:14:741  NOTICE    (SNTP) : .
000:00:00:15:241  NOTICE    (SNTP) : .
000:00:00:15:742  NOTICE    (SNTP) : .
000:00:00:16:242  NOTICE    (SNTP) : .
000:00:00:16:743  NOTICE    (SNTP) : .
000:00:00:17:243  NOTICE    (SNTP) : .
000:00:00:17:744  NOTICE    (SNTP) : .
000:00:00:18:244  NOTICE    (SNTP) : .
000:00:00:18:745  NOTICE    (SNTP) : .
000:00:00:19:245  NOTICE    (SNTP) : .
000:00:00:19:746  NOTICE    (SNTP) : .
000:00:00:20:246  NOTICE    (SNTP) : .
000:00:00:20:746  NOTICE    (SNTP) : .
000:00:00:21:247  NOTICE    (SNTP) : .
000:00:00:21:747  NOTICE    (SNTP) : .
000:00:00:22:248  NOTICE    (SNTP) : .
000:00:00:22:748  NOTICE    (SNTP) : .
000:00:00:23:249  NOTICE    (SNTP) : .
000:00:00:23:749  NOTICE    (SNTP) : .
000:00:00:24:250  NOTICE    (SNTP) : .
000:00:00:24:750  NOTICE    (SNTP) : .
000:00:00:25:251  NOTICE    (SNTP) : .
000:00:00:25:751  NOTICE    (SNTP) : .
000:00:00:26:252  NOTICE    (SNTP) : .
000:00:00:26:752  NOTICE    (SNTP) : .
000:00:00:27:253  NOTICE    (SNTP) : .
000:00:00:27:753  NOTICE    (SNTP) : .
000:00:00:28:254  NOTICE    (SNTP) : .
000:00:00:28:754  NOTICE    (SNTP) : .
000:00:00:29:254  NOTICE    (SNTP) : .
000:00:00:29:755  NOTICE    (SNTP) : .
000:00:00:30:255  NOTICE    (SNTP) : .
000:00:00:30:756  NOTICE    (SNTP) : .
000:00:00:31:256  NOTICE    (SNTP) : .
000:00:00:31:757  NOTICE    (SNTP) : .
000:00:00:32:257  NOTICE    (SNTP) : .
000:00:00:32:758  NOTICE    (SNTP) : .
000:00:00:33:258  NOTICE    (SNTP) : .
000:00:00:33:759  NOTICE    (SNTP) : .
000:00:00:34:259  NOTICE    (SNTP) : .
000:00:00:34:760  NOTICE    (SNTP) : .
000:00:00:34:760  NOTICE    (SNTP) :
000:00:00:34:760  NOTICE    (SNTP) : Current time:
000:00:00:34:761  NOTICE    (SNTP) : Mon Apr 22 00:37:25 2019

```

dontsovcmc, 2019-04-21:
Мне не понятно, зачем мы в либу передаем 3 адреса, а она "ждет у моря погоды".  Я думаю надо найти другую, которая действительно перебирает адреса и можно задать таймаут.

dontsovcmc, 2019-04-22:
@OloloevReal а можете мне в телеграм/эл. почте написать?

OloloevReal, 2019-04-24:
Хорошего стабильного результата удалось добиться просто продублировав попытку синхронизации через какое-то время, в данном случае ~5с.

```
bool setClock() 
{
	configTime(0, 0, "ru.pool.ntp.org", "time.nist.gov");

	LOG_NOTICE("NTP", "Waiting for NTP time sync: ");
	uint32_t start = millis();
	time_t now = time(nullptr);
	uint8_t cnt = 0;
	
	while (now < 8 * 3600 * 2  && millis() - start < ESP_CONNECT_TIMEOUT) {
		delay(500);
		now = time(nullptr);
		if (cnt < 10){
			cnt++;		
		}else{
			cnt = 0;
			LOG_NOTICE("NTP", "One else attempt to sync time");
			configTime(0, 0, "ru.pool.ntp.org", "time.nist.gov");
		}
	}
	
	if (millis() - start >= ESP_CONNECT_TIMEOUT) {
		return false;
	}

	struct tm timeinfo;
	gmtime_r(&now, &timeinfo);
	LOG_NOTICE("NTP", "Current time: " << asctime(&timeinfo));
	return true;
}
```

```
000:00:00:05:389  NOTICE    (NTP) : Waiting for NTP time sync:
000:00:00:10:890  NOTICE    (NTP) : One else attempt to sync time
000:00:00:11:392  NOTICE    (NTP) : Current time: Wed Apr 24 09:55:22 2019

```

dontsovcmc, 2019-05-04:
Я плюнул на исследование внутренностей sntp библиотеки и сделал 3 сервера с таймаутом 3 секунды.
См: https://github.com/dontsovcmc/waterius/commit/5ef9d5cac922805f787bac252055db4d41bba2d1

closed 2019-05-04 by dontsovcmc