dontsovcmc/waterius #75: USIWire VS Wire

OloloevReal, 2019-04-24:
Добрый день!

Иногда в работе waterius наблюдал такую картину, что attiny переставала отвечать на I2C команды от ESP. Почему это происходило и что является причиной честно сказать - не знаю, без логического анализатора на шине наверное это и не выяснить. Пробовал и понижать скорость и увеличивать задержки, изменять StretchLimit, стучать в бубен шамана все эти действия помогали, но не решали проблему, рано или поздно в логах все равно появлялись ошибки что та или иная команда отправлена в Attiny с ошибкой.

Код у меня сильно изменен, добавлены дополнительные команды для обмена Attiny <-> ESP, обмен более активный в обе стороны нежели master версия. Думаю что проблема поэтому и вылезла.

Так вот, обратил внимание что в Attiny используется для реализации I2C USIWire библиотека, ее я и заменил на стандартную Wire
В моем случае (надеюсь) это решило проблему, тестовое устройство вторые сутки отсылает данные с интервалом каждые 5 минут

Ничего не могу сказать плохого про USIWire, но если вдруг будут проблемы с I2C имеет смысл попробовать Wire.

Ишью можно сразу закрыть, просто в меморис.

dontsovcmc, 2019-04-24:
Usiwire перешел из проекта метеостанции с буфером. И нужен был потому что аттини сначала должна быть мастером, а потом слейвом. Wire мастер тогда отсутствовал.

dontsovcmc, 2019-04-28:
@OloloevReal Т.е. можно безболезненно на Wire перейти? 
1. проблемы с ошибками можно нивелировать добавлением crc в посылки =/.

OloloevReal, 2019-04-29:
Ни одной ошибки по I2C с момента перехода на Wire!

> Если не секрет, то что вы еще добавили в опрос.

Добавил следующее:
- Очистка конфигурации - если при подачи питания зажата кнопка отправки данных, произойдет сброс конфигурации Attiny и ESP на значения по умолчанию
- Изменение таймера отправки показаний из интерфейса ESP - после сохранения конфигурации ESP передает значение таймера Attiny
- Обновление firmware ESP - если ESP видит что доступно обновление посылает команду Attiny на увеличение времени ожидания ESP. После обновления ESP запрашивает у Attiny причину пробуждения и та отвечает Update что является триггером для отправки на сервер подтверждения окончания обновления.

dontsovcmc, 2019-04-29:
Очистка конфигурации - если при подачи питания зажата кнопка отправки данных, произойдет сброс конфигурации Attiny и ESP на значения по умолчанию
- Для продакшена отлично

Изменение таймера отправки показаний из интерфейса ESP - после сохранения конфигурации ESP передает значение таймера Attiny
- да, это сейчас единственный способ синхронизировать отправку

Обновление firmware ESP - если ESP видит что доступно обновление посылает команду Attiny на увеличение времени ожидания ESP. После обновления ESP запрашивает у Attiny причину пробуждения и та отвечает Update что является триггером для отправки на сервер подтверждения окончания обновления.
- Круто! А вы перепаяли SDRAM? А ОТА хорошо работает? У меня за 20 сек обновлялась. Но у меня мало понимания подводных камней, чтобы в master публиковать.

OloloevReal, 2019-04-29:
> Код надо менять под Wire?

Только инклуды поменять, убрать из platformio.ini зависимость

> Круто! А вы перепаяли SDRAM? А ОТА хорошо работает? У меня за 20 сек обновлялась. Но у меня мало понимания подводных камней, чтобы в master публиковать.

Да, поменял сразу как решил делать OTA, проще и быстрее поменять память чем оптимизировать код.

dontsovcmc, 2019-04-29:
1. Отлично. Конечно надо заменить на Wire.
2. Это 100%. С TLS так вообще не выоптимизируешь.

3. А зачем надо синхронизировать время просыпания? Это как-то нужно в коммерческом учете? Можно же взять и усреднить 25ч в 24ч... Обычно же месячный расход интересен, а погрешность мелкой получится - ~1ч на 30 дней.

OloloevReal, 2019-04-30:
> А зачем надо синхронизировать время просыпания? Это как-то нужно в коммерческом учете? Можно же взять и усреднить 25ч в 24ч... Обычно же месячный расход интересен, а погрешность мелкой получится - ~1ч на 30 дней.

А? Это про что?

dontsovcmc, 2019-04-30:
Я про это:
- Изменение таймера отправки показаний из интерфейса ESP - после сохранения конфигурации ESP передает значение таймера Attiny

OloloevReal, 2019-04-30:
Да, именно о периоде отправки показаний, по умолчанию в коде 24 часа, что для отладки и экспериментов слишком долго,а каждый раз шить контроллер морока еще та.

closed 2019-05-04 by dontsovcmc