dontsovcmc/waterius #76: Намур (Берегун)

OloloevReal, 2019-04-27:
Добрый день!

При подключении счетчика ГВС Берегун обнаружил пару проблем:
1) В паспорте указываются для цепи Намур сопротивления 5.6 кОм для открытого состояния и 3.9 кОм для закрытого. 

[image]

При подключении к вотериусу такого счетчика не попадаю в диапазоны

```
#define LIMIT_CLOSED       115   // < 115 - замыкание 
#define LIMIT_NAMUR_CLOSED 170 // < 170 - намур замкнут
#define LIMIT_NAMUR_OPEN   800   // < 800 - намур разомкнут
```
Сопротивление на входе 3.9 кОм дает значение analogRead 190, что не попадает в установленный лимит 170 для закрытого состояния Намур.

2) И вторая проблема, наиболее плохая, при нажатии на кнопку происходит изменение величины считываемой со входов. Например, если состояние входа NAMUR_CLOSED, в моем случае значение 191, и происходит нажатие кнопки, то значение на входе изменяется со 191 на 263

```
37881 [D]: 191
37881 [D]: 191
37881 [D]: 191
37881 [D]: 263  <- изменение, фактически NAMUR_CLOSED -> NAMUR_OPEN
37883 [I]: Data:
37910 [I]: 13
37933 [I]: 0
```
Это приводит к ложным увеличениям счетчика импульсов, т.е. при каждой принудительной отправке показаний кнопкой происходит +1 счетчика.

dontsovcmc, 2019-04-27:
Вау! Да, не протестировал "в поле".
Выставите LIMIT_NAMUR_CLOSED 210.

OloloevReal, 2019-04-27:
Да, та кнопка что для отправки/настройки

Если вот сюда вставить дебаг, то при нажатии кнопки будет видно что считываемое значение увеличится
```
    bool is_close()
    {
        //bool value = digRead();
        //state = value2state(value ? 1024 : 0);
        uint16_t value = aRead();
        LOG_DEBUG(value);
        state = value2state(value);
        return state == CounterState_e::CLOSE || state == CounterState_e::NAMUR_CLOSE;
    }
```

Временно вышел из положения таким образом:

```
inline void counting() {

    power_adc_enable(); //т.к. мы обесточили всё а нам нужен компаратор
    adc_enable();       //после подачи питания на adc

	if(button.digBit() == LOW){
		adc_disable();
    	        power_adc_disable();
		LOG_DEBUG(F("Skip!"));
		return;
	}
.....
```
Т.е. откидывая значение если кнопка нажата перед чтением данных со входов.

Мне кажется это как-то связанно с тем что контроллер находится power save, а нажатие кнопки просаживает напряжение

dontsovcmc, 2019-04-28:
Хм... Не понимаю. Это же разные пины и они не должны влиять друг на друга.

1. Странно, что лог работает, мне казалось LOG_BEGIN всегда после пробуждения нужно делать.
2. А как увеличивается значение value? Оно всегда плавает - это же компаратор от 0 до 1023.
```
uint16_t value = aRead();
LOG_DEBUG(value);
```

3. А можете протестировать вызов counting() в условии, вместо изменения ее содержимого:
```
		while ( wdt_count > 0 ) {
			noInterrupts();

			if (button_pressed()) { 
				interrupts();  // Пользователь нажал кнопку
				break;
			} else 	{
			        counting(); //Опрос входов
				interrupts();
				sleep_mode();  // Спим (WDTCR)
			}
		}
```

OloloevReal, 2019-04-29:
> А как увеличивается значение value? Оно всегда плавает - это же компаратор от 0 до 1023.

При постоянном значении нагрузки на входах значение плавает +- 1
А вот так оно меняется при нажатии на кнопку:
0:             109 -> 188
1.6 кОм:  143 -> 220
5.6 кОм:  221 -> 290
обрыв:    1020 -> 1023

> А можете протестировать вызов counting() в условии, вместо изменения ее содержимого:

С таким кодом работает корректно, ложных срабатываний не фиксируется
```
23803 [D]: 143
23803 [D]: 143
23803 [D]: 143
23803 [D]: 143
23803 [D]: 143
23803 [D]: 143
23803 [D]: 143
23803 [D]: 143
23803 [D]: 143
23803 [D]: 143
23803 [D]: 143
23803 [D]: 143
23803 [D]: 143
23808 [I]: Data:
23832 [I]: 4
23853 [I]: 0
24379 [D]: wake up for transmitting

```

dontsovcmc, 2019-05-04:
ААААА! Нашел ошибку!
Теперь counting() нужен Компаратор, а он выключается в функции readVcc!
И поэтому после каждой отсылки может быть дополнительный +10л..

dontsovcmc, 2019-05-04:
Я сейчас попробовал у себя - у меня вообще нет этой ошибки.. Придется вам код показать, а то не ясно, что вы переделали =(

dontsovcmc, 2019-05-04:
Проверьте, что включен компаратор в том месте. Потому что в readVcc  он выключается и я удивился, когда увидел другое значение сразу после вызова функции.

OloloevReal, 2019-05-04:
Проверяю на v0.8, стоковой версии, все отличия это активированный лог, уменьшенный таймер  #define WAIT_ESP_MSEC    5000UL и добавленный принтаут в функцию is_close()
Записал лог, в нем видно что при нажатии кнопки посылки показаний происходит скачек значения и счетчик импульсов увеличивает значение

ESP отключена, нагрузка на Input_0 подключена 1.6 кОм
[log.log](https://github.com/dontsovcmc/waterius/files/3144590/log.log)

OloloevReal, 2019-05-04:
Пробовал убирать из кода запрос напряжения	
`//info.voltage = readVcc();   // Текущее напряжение`
ни на что не повлияло, проблема сохраняется

dontsovcmc, 2019-05-04:
Упс. WAIT_ESP_MSEC надо вернуть. 
Я вставляю печать в is_close  и нет никакой разницы в показаниях. Т.е. кнопка нажата,нет. Нет разницы.
У вас она есть. пришлите кусок кода где вы нажимаете кнопку, пожалуйста.

OloloevReal, 2019-05-04:
WAIT_ESP_MSEC ставлю небольшим значением что бы проблему отлавливать быстрее - проблема проявляется когда Attiny в повер сейве и выходит из него. Если она уже проснулась, и например, ждет ESP, то жми, не жми кнопку результат стабильный, изменений не происходит.

Вот все изменения сделанные поверх master версии, все сток [diff ](https://gist.github.com/OloloevReal/df41f2b75453a0c0729c97dfdeea40bf)

dontsovcmc, 2019-05-04:
да, да. я также сделал. у меня нет разницы в aRead(). Слушайте, а на плате у вас нет замыкания какого-нибудь? Ну если код одинаковый, МК тоже, то только плата осталась...

OloloevReal, 2019-05-05:
Не похоже на замыкание, померил ток:
0.01 mA - power save
1.56 mA - кнопка нажата, attiny проснулась и ждет ESP

Очень странно, ладно, видимо нужно собирать еще одну тестовую плату, на новом МК

dontsovcmc, 2019-05-05:
Второй человек сегодня отписал об этой проблеме... Мне не сложно перенести counting() в условие, но это костыль, а нужно понять причину. 
Я смотрел в сторону AREF - опорное напряжение компаратора, но этот пин SDA, а кнопка замыкает SLK.

dontsovcmc, 2019-05-05:
о! я проверял без ESP всунутой. А ведь она может как-то влиять. У вас ошибка без ESP воспроизводится?

OloloevReal, 2019-05-05:
Я проверяю без ESP, ошибка воспроизводится без ESP

dontsovcmc, 2019-05-06:
> Я проверяю без ESP, ошибка воспроизводится без ESP

И ошибка будет только если замкнут НАМУР?

dontsovcmc, 2019-05-19:
Оставил пока так:
```
{
interrupts();  // Пользователь нажал кнопку
break;
} else {
    counting(); //Опрос входов
    interrupts();
    sleep_mode();  // Спим (WDTCR)
}
```

closed 2019-05-19 by dontsovcmc