dontsovcmc/waterius #78: Автоопределение кол-ва литров на 1 импульс

dontsovcmc, 2019-04-30:
Открою вновь, т.к. "душа требует юзабилити":

Задача: не объяснять пользователю, что вводить в графе "литров на 1 импульс".
Проблема: на счетчиках НЕ указано сколько литров на импульс. Даже в паспорте иногда умалчивают. Пруф: https://m.habr.com/ru/post/359429

Как можно определить кол-во импульсов на 1 литр:
ESP может замерять период прихода импульсов. Берем минимальный из них. Скажем, за 10 сек пришёл 1 импульс - значит 10л/имп. Если 3 сек - 1л/имп. 10л за 3 секунды - слишком для квартиры.

В интерфейсе будет:

Кол-во литров на 1 импульс:
[x] Авто
     [__1__] л //disabled

Или:
Кол-во литров на 1 импульс:
[  ] Авто
     [__10_] л  //enabled

dontsovcmc, 2019-05-06:
@OloloevReal @popsodav  мы тут "Теслу" изобретаем, как вам такая вариация взаимодействия с домохозяйками?
http://prntscr.com/nl96d3
Идея, чтобы ни одна живая душа не знала, что бывают "импульсы",

OloloevReal, 2019-05-13:
@dontsovcmc я убежден что чем меньше настроек доступно пользователю, чем меньше действий от него требуется тем лучше. В идеале должна быть только кнопка вкл/выкл
С идеальной системой, на мой взгляд, мне как-то довелось работать - клиентам устанавливались конечные устройства, и никаких настроек доступно пользователю не было, вообще, от слова совсем. На заводе прошивался URL куда устройство должно пробовать подключаться после появления питания, и откуда оно получало все остальные настройки из заранее сконфигурированного профиля. Устройство работало через ethernet поэтому никакой wifi выбирать необходимости не было.
Это я все к чему, может быть стоит разделить софт на две ветки - Waterius.Home и Waterius.Enterprise. В первом случае доступен только Blynk, во втором случае из настроек только выбор точки доступа, а дальше все настройки приходят из облака, туда же и отправляются данные. Облако в свою очередь может отправлять данные в Blynk.

dontsovcmc, 2019-05-13:
@OloloevReal да, безусловно меньше действий - лучше.
Пока решили на железке оставить. Вот проект прошивки для облака: http://prntscr.com/nnxryh

dontsovcmc, 2019-05-19:
Сделал автоопределение множителя (кол-во литров на импульс): 

1. Настраивать надо каждый раз при включении настройки
2. Множитель определяется по Входу №2 (холодная вода)
3. Множитель равен 10, если кол-во импульсов пришедших от открытия Окна "Настройка", до нажатия кнопки "Сохранить" меньше либо равно 3.
Множитель равен 1, если 4 и выше.
4. Множитель для горячей и холодной воды одинаковый.

В настройке предлагается спустить унитаз. Объём воды в нем обычно 4 литра. 
Таким образом, вы спускаете унитаз и Ватериус видит, если пришло больше 3-х импульсов, то у вас счетчик выдает 1 импульс на каждый литр. Если пришел 1 импульс, то 1 импульс на 10 литров.

closed 2019-05-19 by dontsovcmc