dontsovcmc/waterius #82: Если после настройки делать HTTPS - падение

dontsovcmc, 2019-05-13:
Хочу переделать, так, чтобы после успешной настройки показания сразу улетали на сервер.
Столкнулся с тем, что не хватает памяти. Падение в вызове new в недрах SSL.

Вопрос: как очистить память после работы WiFi точки доступа?
Вряд ли библиотеку BearSSL можно оптимизировать.

```
Включение без настройки
000:00:00:03:600  NOTICE    (MQT) : heap1=32672
000:00:00:03:604  NOTICE    (MQT) : stack1=2640
000:00:00:03:608  NOTICE    (SND) : -- START -- Send new data
000:00:00:03:615  NOTICE    (MQT) : heap2=32440
000:00:00:03:618  NOTICE    (MQT) : stack2=2160
ОК

000:00:00:03:599  NOTICE    (MQT) : heap1=32672
000:00:00:03:603  NOTICE    (MQT) : stack1=2640
000:00:00:03:608  NOTICE    (SND) : -- START -- Send new data
000:00:00:03:614  NOTICE    (MQT) : heap2=32440
000:00:00:03:617  NOTICE    (MQT) : stack2=2128
ОК

После настройки
000:00:00:34:722  NOTICE    (MQT) : heap1=29224
000:00:00:34:726  NOTICE    (MQT) : stack1=1760
000:00:00:34:730  NOTICE    (SND) : -- START -- Send new data
000:00:00:34:737  NOTICE    (MQT) : heap2=28992
000:00:00:34:740  NOTICE    (MQT) : stack2=1760
падение

Поменял JSON на DynamicJson 1024

000:00:00:03:599  NOTICE    (MQT) : heap1=32672
000:00:00:03:603  NOTICE    (MQT) : stack1=2640
000:00:00:03:608  NOTICE    (SND) : -- START -- Send new data
000:00:00:03:614  NOTICE    (MQT) : heap2=32440
000:00:00:03:617  NOTICE    (MQT) : stack2=2160
падение

000:00:00:48:134  NOTICE    (MQT) : heap1=29144
000:00:00:48:138  NOTICE    (MQT) : stack1=1856
000:00:00:48:143  NOTICE    (SND) : -- START -- Send new data
000:00:00:48:149  NOTICE    (MQT) : heap2=27880
000:00:00:48:152  NOTICE    (MQT) : stack2=1856
000:00:00:48:157  NOTICE    (RQT) : -- START -- Send JSON POST request
000:00:00:48:163  INFO      (RQT) : URL:	https://cloud.waterius.ru
000:00:00:48:169  INFO      (RQT) : Body:	{"delta0":0,"delta1":0,"good":1,"boot":9,"ch0":0.07,"ch1":0.03,"imp0":7,"imp1":3,"version":5,"voltage":2.948,"version_esp":"0.8.2","key":"","resets":1,"email":"@mail.ru"}
000:00:00:48:460  NOTICE    (NTP) : Waiting for NTP time sync: 
000:00:00:48:560  NOTICE    (NTP) : Current time: Mon May 13 02:14:44 2019

000:00:00:48:561  INFO      (RQT) : Begin client
000:00:00:48:562  NOTICE    (MQT) : heap3=21960
000:00:00:48:565  NOTICE    (MQT) : stack3=1856
падение

Выключил все сертификаты, кроме 1го

000:00:02:00:155  NOTICE    (MQT) : heap1=29016
000:00:02:00:159  NOTICE    (MQT) : stack1=1556
000:00:02:00:163  NOTICE    (SND) : -- START -- Send new data
000:00:02:00:170  NOTICE    (MQT) : heap2=27752
000:00:02:00:173  NOTICE    (MQT) : stack2=1556
000:00:02:00:177  NOTICE    (RQT) : -- START -- Send JSON POST request
000:00:02:00:184  INFO      (RQT) : URL:	https://cloud.waterius.ru
000:00:02:00:190  INFO      (RQT) : Body:	{"delta0":0,"delta1":0,"good":1,"boot":9,"ch0":0.07,"ch1":0.03,"imp0":7,"imp1":3,"version":5,"voltage":2.948,"version_esp":"0.8.2","key":"","resets":1,"email":"mail.ru"}
000:00:02:00:280  NOTICE    (NTP) : Waiting for NTP time sync: 
000:00:02:00:380  NOTICE    (NTP) : Current time: Mon May 13 08:14:56 2019

000:00:02:00:381  INFO      (RQT) : Begin client
000:00:02:00:382  NOTICE    (MQT) : heap3=26400
000:00:02:00:385  NOTICE    (MQT) : stack3=1556

падение
```

стек:
```
Decoding 56 results
0x4021ff50: operator new[](unsigned int) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3-nox/libsupc++/../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_opv.cc line 33
0x4020e01f: BearSSL::WiFiClientSecure::_connectSSL(char const*) at ?? line ?
0x402137a8: esp_yield at ?? line ?
0x40201e33: delay at ?? line ?
0x4020cdf9: WiFiClient::connect(IPAddress const&, unsigned short) at ?? line ?
0x4020e236: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at ?? line ?
0x40217d58: WiFiServer::_s_discard(void*, ClientContext*) at ?? line ?
0x402177c0: HTTPClient::connected() at ?? line ?
0x40210ec8: HTTPClient::connect() at ?? line ?
0x40212a27: String::changeBuffer(unsigned int) at ?? line ?
0x40212a00: String::~String() at ?? line ?
0x402116b0: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at ?? line ?
0x40212a27: String::changeBuffer(unsigned int) at ?? line ?
0x40212a73: String::reserve(unsigned int) at ?? line ?
0x40211764: HTTPClient::POST(unsigned char*, unsigned int) at ?? line ?
0x40211782: HTTPClient::POST(String) at ?? line ?
0x40204a4a: WateriusHttps::sendJsonPostRequest(String const&, char const*, char const*, String const&) at ?? line ?
0x40212a27: String::changeBuffer(unsigned int) at ?? line ?
0x40212abc: String::copy(char const*, unsigned int) at ?? line ?
0x40212af2: String::String(char const*) at ?? line ?
0x40203ec9: UserClass::sendNewData(Settings const&, SlaveData const&, CalculatedData const&) at ?? line ?
0x4021a120: _printf_i at /home/earle/src/esp-quick-toolchain/arena/newlib-install/xtensa-lx106-elf/newlib/libc/stdio/../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf_i.c line 244
0x4026b5e5: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4021ea17: _svfprintf_r at /home/earle/src/esp-quick-toolchain/arena/newlib-install/xtensa-lx106-elf/newlib/libc/stdio/../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 660
0x40212038: HardwareSerial::write(unsigned char const*, unsigned int) at ?? line ?
0x40212335: Print::write(char const*) at ?? line ?
0x40212038: HardwareSerial::write(unsigned char const*, unsigned int) at ?? line ?
0x40265a94: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40212335: Print::write(char const*) at ?? line ?
0x4021e606: __ssputs_r at /home/earle/src/esp-quick-toolchain/arena/newlib-install/xtensa-lx106-elf/newlib/libc/stdio/../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 233
0x4021a020: _printf_i at /home/earle/src/esp-quick-toolchain/arena/newlib-install/xtensa-lx106-elf/newlib/libc/stdio/../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf_i.c line 194 (discriminator 1)
0x4021e606: __ssputs_r at /home/earle/src/esp-quick-toolchain/arena/newlib-install/xtensa-lx106-elf/newlib/libc/stdio/../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 233
0x4021e540: __ssputs_r at /home/earle/src/esp-quick-toolchain/arena/newlib-install/xtensa-lx106-elf/newlib/libc/stdio/../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 180
0x4021a120: _printf_i at /home/earle/src/esp-quick-toolchain/arena/newlib-install/xtensa-lx106-elf/newlib/libc/stdio/../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf_i.c line 244
0x40217d58: WiFiServer::_s_discard(void*, ClientContext*) at ?? line ?
0x4026b5e5: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40206fe4: send_blynk(Settings const&, SlaveData const&, CalculatedData const&) at ?? line ?
0x40208c50: update_data(String&) at ?? line ?
0x4021ea17: _svfprintf_r at /home/earle/src/esp-quick-toolchain/arena/newlib-install/xtensa-lx106-elf/newlib/libc/stdio/../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 660
0x40264c8b: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x402649b4: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4021a96c: sprintf at /home/earle/src/esp-quick-toolchain/arena/newlib-install/xtensa-lx106-elf/newlib/libc/stdio/../../../../../../dl/newlib-xtensa/newlib/libc/stdio/sprintf.c line 646
0x4021a020: _printf_i at /home/earle/src/esp-quick-toolchain/arena/newlib-install/xtensa-lx106-elf/newlib/libc/stdio/../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf_i.c line 194 (discriminator 1)
0x4021e606: __ssputs_r at /home/earle/src/esp-quick-toolchain/arena/newlib-install/xtensa-lx106-elf/newlib/libc/stdio/../../../../../../dl/newlib-xtensa/newlib/libc/stdio/nano-vfprintf.c line 233
0x40212038: HardwareSerial::write(unsigned char const*, unsigned int) at ?? line ?
0x40212335: Print::write(char const*) at ?? line ?
0x40212348: Print::print(char const*) at ?? line ?
0x40212a27: String::changeBuffer(unsigned int) at ?? line ?
0x40217d98: WiFiServer::_s_discard(void*, ClientContext*) at ?? line ?
0x40207b4c: loop at ?? line ?
0x40205585: setup at ?? line ?
0x40217d58: WiFiServer::_s_discard(void*, ClientContext*) at ?? line ?
0x40213854: loop_wrapper() at core_esp8266_main.cpp line ?
0x40101511: cont_wrapper at ?? line ?
```

OloloevReal, 2019-05-13:
Можно перезагрузить ESP и затем выполнить отправку

dontsovcmc, 2019-05-13:
Да, но тогда надо слать команду "Restart" в Attiny. Она должна рестартануть в режиме Transmit... 
Я еще не пробовал выключать WiFiClient. Не знаете, как грамотно это проделать?

OloloevReal, 2019-05-14:
Решение в лоб,  установить в Attiny режим TRANSMIT_MODE после того как сказали ESP SETUP_MODE

```
		case 'M':  // Разбудили ESP для настройки или передачи данных?
			txBuffer[0] = setup_mode;
			if(setup_mode == SETUP_MODE){
				setup_mode = TRANSMIT_MODE;
			}
			break;

```

dontsovcmc, 2019-05-14:
@OloloevReal может нужно просто ребутнуть WiFi?

Вот вижу доку:
https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/soft-access-point-class.html#softapdisconnect

WiFi.softAPdisconnect(true); вызвать надо точно. это выключит веб сервер и отрубит клиентов. возможно уже эта мера очистит память.

может достаточно вызова WiFi.mode(WIFI_STA);  после настройки? Могли бы протестировать?

OloloevReal, 2019-05-15:
Попробовал, никакого толку, куча не уменьшается

```
            //Запускаем точку доступа с вебсервером
            LOG_NOTICE("ESP", "Heap1: " << ESP.getFreeHeap());
            setup_ap(sett, data, cdata);
            LOG_NOTICE("ESP", "Heap2: " << ESP.getFreeHeap());
            WiFi.softAPdisconnect(true);
            LOG_NOTICE("ESP", "Heap3: " << ESP.getFreeHeap());
```

```
000:00:00:00:180  NOTICE    (ESP) : Heap1: 33848
000:00:00:00:185  NOTICE    (ESP) : I2C-begined: mode SETUP
..........
.........
000:00:00:32:293  NOTICE    (AP) : impulses1=0
000:00:00:32:297  NOTICE    (CFG) : Config stored OK
*WM: freeing allocated params!
000:00:00:32:304  NOTICE    (ESP) : Heap2: 28272
000:00:00:32:308  NOTICE    (ESP) : Heap3: 28272
000:00:00:32:813  NOTICE    (WIF) : Starting
000:00:00:32:814  NOTICE    (WIF) : Status: 6
000:00:00:33:014  NOTICE    (WIF) : Connected, IP: 192.168.1.253
```

dontsovcmc, 2019-11-28:
После динамической загрузки SSID сетей, памяти стало больше. Уменьшилась фрагментация:
[MEM] free: 31080 | max: 25648 | frag:  18%

dontsovcmc, 2020-05-02:
Кажется я теряю 4кб памяти не вызывая EEPROM.end() после чтения настроек )

dontsovcmc, 2022-03-30:
Решил тем, что перезапускаю ESP после настройки.

closed 2022-03-30 by dontsovcmc