dontsovcmc/waterius #83: Blynk+MQTT + HTTPS - return -1

dontsovcmc, 2019-05-19:
Если включить Blynk, MQTT, то после них HTTPS возвращает -1...
```
 (ESP) : Booted
2 000:00:00:00:055  NOTICE    (I2C) : mode=2
9 1 D3 B 0 0 1 0 1 1 1 0 0 0 3 0 0 0 000:00:00:00:070  NOTICE    (I2C) : version: 9
000:00:00:00:070  NOTICE    (I2C) : service: 1
000:00:00:00:070  NOTICE    (I2C) : voltage: 3027
000:00:00:00:072  NOTICE    (I2C) : resets: 1
000:00:00:00:076  NOTICE    (I2C) : state0: 1
000:00:00:00:080  NOTICE    (I2C) : state1: 1
000:00:00:00:084  NOTICE    (I2C) : impulses0: 1
000:00:00:00:088  NOTICE    (I2C) : impulses1: 3
000:00:00:00:093  NOTICE    (CFG) : CRC ok
000:00:00:00:096  NOTICE    (CFG) : WATERIUS.RU
000:00:00:00:101  NOTICE    (CFG) : email=x@mail.ru
000:00:00:00:106  NOTICE    (CFG) : host=https://cloud.waterius.ru key=x
000:00:00:00:115  NOTICE    (CFG) : BLYNK.CC
000:00:00:00:119  NOTICE    (CFG) : host=blynk-cloud.com key=x
000:00:00:00:128  NOTICE    (CFG) : email=x@mail.ru
000:00:00:00:133  NOTICE    (CFG) : MQTT
000:00:00:00:137  NOTICE    (CFG) : host=broker.hivemq.com port=1883
000:00:00:00:143  NOTICE    (CFG) : login= pass=
000:00:00:00:147  NOTICE    (CFG) : topic=waterius/1234/
000:00:00:00:152  NOTICE    (CFG) : COUNTERS
000:00:00:00:156  NOTICE    (CFG) : channel0_start=0.00, impulses0_start=1, factor=10
000:00:00:00:164  NOTICE    (CFG) : channel1_start=0.00, impulses1_start=3
000:00:00:00:171  NOTICE    (ESP) : new impulses=1 3
000:00:00:00:175  NOTICE    (ESP) : new value0=0.00 value1=0.00
000:00:00:00:181  NOTICE    (ESP) : delta0=0 delta1=0
000:00:00:00:186  NOTICE    (WIF) : Starting
000:00:00:00:191  NOTICE    (WIF) : Status: 6
000:00:00:00:394  NOTICE    (WIF) : Status: 6
000:00:00:00:594  NOTICE    (WIF) : Status: 6
000:00:00:00:795  NOTICE    (WIF) : Status: 6
000:00:00:00:995  NOTICE    (WIF) : Status: 6
000:00:00:01:196  NOTICE    (WIF) : Status: 6
000:00:00:01:396  NOTICE    (WIF) : Status: 6
000:00:00:01:597  NOTICE    (WIF) : Status: 6
000:00:00:01:797  NOTICE    (WIF) : Status: 6
000:00:00:01:998  NOTICE    (WIF) : Status: 6
000:00:00:02:198  NOTICE    (WIF) : Status: 6
000:00:00:02:399  NOTICE    (WIF) : Status: 6
000:00:00:02:599  NOTICE    (WIF) : Status: 6
000:00:00:02:800  NOTICE    (WIF) : Status: 6
000:00:00:03:000  NOTICE    (WIF) : Status: 6
000:00:00:03:201  NOTICE    (WIF) : Status: 6
000:00:00:03:402  NOTICE    (WIF) : Status: 6
000:00:00:03:602  NOTICE    (WIF) : Connected, IP: 192.168.1.52
000:00:00:03:880  NOTICE    (BLK) : run
000:00:00:04:282  NOTICE    (BLK) : virtualWrite OK
000:00:00:04:283  NOTICE    (BLK) : send email: don-and-home@mail.ru
000:00:00:04:349  NOTICE    (BLK) : email was send
000:00:00:04:350  NOTICE    (BLK) : Новые показания {DEVICE_NAME}
000:00:00:04:350  NOTICE    (BLK) : Горячая: 0.0м3, Холодная: 0.0м3<br>За день:<br>Горячая: +0л, Холодная: +0л<br>Напряжение:3.027В
000:00:00:04:403  NOTICE    (BLK) : disconnected
000:00:00:04:403  NOTICE    (BLK) : send ok
000:00:00:04:886  NOTICE    (MQT) : send ok
000:00:00:04:886  NOTICE    (SND) : -- START -- Send new data
000:00:00:04:888  NOTICE    (RQT) : -- START -- Send JSON POST request
000:00:00:04:891  INFO      (RQT) : URL:	https://cloud.waterius.ru
000:00:00:04:897  INFO      (RQT) : Body:	{"delta0":0,"delta1":0,"good":1,"boot":9,"ch0":0,"ch1":0,"imp0":1,"imp1":3,"version":5,"voltage":3.027,"version_esp":"0.8.2","key":"x","resets":1,"email":"x@mail.ru"}
000:00:00:05:186  NOTICE    (NTP) : Waiting for NTP time sync: 
000:00:00:05:287  NOTICE    (NTP) : Current time: Sun May 19 21:59:33 2019

000:00:00:05:288  INFO      (RQT) : Begin client
000:00:00:42:098  INFO      (RQT) : Response code:	-1
000:00:01:05:328  INFO      (RQT) : Response body:	
000:00:01:05:329  INFO      (RQT) : Result:	Success
000:00:01:05:329  NOTICE    (RQT) : -- END --
000:00:01:05:330  INFO      (SND) : Send HTTP code:	-1
000:00:01:05:335  INFO      (SND) : Send result:	Error
000:00:01:05:340  NOTICE    (SND) : -- END --
000:00:01:05:344  NOTICE    (CFG) : Config stored OK
000:00:01:05:349  NOTICE    (ESP) : Going to sleep
```

dontsovcmc, 2019-05-20:
Как найти ошибку:
- вставить в качестве сервера блинк их IP
- вставить в качестве HTTP сервера IP
мы узнаем, есть ли проблема в mDNS модуле
- поменять местами отправку HTTP и Blynk

OloloevReal, 2019-05-27:
Вот так работает (Blynk + HTTPS):

`platform = https://github.com/platformio/platform-espressif8266.git#v2.1.1`

```
000:00:00:02:796  NOTICE    (WIF) : Status: 6
000:00:00:02:996  NOTICE    (WIF) : Status: 6
000:00:00:03:196  NOTICE    (WIF) : Connected, IP: 192.168.1.253
000:00:00:03:463  NOTICE    (BLK) : run
000:00:00:03:865  NOTICE    (BLK) : virtualWrite OK
000:00:00:03:913  NOTICE    (BLK) : disconnected
000:00:00:03:913  NOTICE    (BLK) : send ok
000:00:00:03:913  NOTICE    (SND) : -- START -- Send new data
000:00:00:03:917  NOTICE    (RQT) : -- START -- Send JSON POST request
000:00:00:03:922  INFO      (RQT) : URL:        https://cloud.waterius.ru
000:00:00:03:928  INFO      (RQT) : Body:       {"delta0":0,"delta1":0,"good":1,"boot":9,"ch0":0,"ch1":0,"imp0":0,"imp1":0,"version":5,"voltage":3.011,"version_esp":"0.8.2","key":"513B2AED3D21473D59F2341F694F4B7A","resets":7,"email":""}
000:00:00:04:213  NOTICE    (NTP) : Waiting for NTP time sync:
000:00:00:04:313  NOTICE    (NTP) : Current time: Mon May 27 21:08:18 2019

000:00:00:04:314  INFO      (RQT) : Begin client
000:00:00:05:145  INFO      (RQT) : Response code:      400
000:00:00:05:147  INFO      (RQT) : Response body:      {"email":["Это поле не может быть пустым."]}
000:00:00:05:150  INFO      (RQT) : Result:     Success
000:00:00:05:154  NOTICE    (RQT) : -- END --
000:00:00:05:158  INFO      (SND) : Send HTTP code:     400
000:00:00:05:163  INFO      (SND) : Send result:        Error
000:00:00:05:168  NOTICE    (SND) : -- END --
000:00:00:05:173  NOTICE    (CFG) : Config stored OK
000:00:00:05:177  NOTICE    (ESP) : Going to sleep

```

Кстати, с самой последней версией platform-espressif8266.git#v2.2.1 при заходе в Настройки происходит краш

dontsovcmc, 2019-05-27:
@OloloevReal а можете провести тест:
1. заполнить email у https. Возникнет ошибка?
2. Заполнить email у Блинка. Возникнет ошибка?

В моем логе это приводит к ошибке =(.
Да, видел, что падает, расстроился, т.к. там мой коммит с повторным подключением.
Значит в WiFiManager что-то не так.. мда, она так медленно развивается, что надо искать замену. Вы не смотрели?

OloloevReal, 2019-05-27:
@dontsovcmc заполнил, работает без ошибок

```
000:00:00:03:201  NOTICE    (WIF) : Connected, IP: 192.168.1.253
000:00:00:03:389  NOTICE    (BLK) : run
000:00:00:03:791  NOTICE    (BLK) : virtualWrite OK
000:00:00:03:791  NOTICE    (BLK) : send email: nikolay.sikorsky@gmail.com
000:00:00:03:858  NOTICE    (BLK) : email was send
000:00:00:03:858  NOTICE    (BLK) : Новые показания {DEVICE_NAME}
000:00:00:03:859  NOTICE    (BLK) : Горячая: 0.0м3, Холодная: 0.0м3<br>За день:<br>Горячая: +0л, Холодная: +0л<br>Напряжение:3.011В
000:00:00:03:955  NOTICE    (BLK) : disconnected
000:00:00:03:956  NOTICE    (BLK) : send ok
000:00:00:03:956  NOTICE    (SND) : -- START -- Send new data
000:00:00:03:960  NOTICE    (RQT) : -- START -- Send JSON POST request
000:00:00:03:964  INFO      (RQT) : URL:        https://cloud.waterius.ru
000:00:00:03:970  INFO      (RQT) : Body:       {"delta0":0,"delta1":0,"good":1,"boot":9,"ch0":0,"ch1":0,"imp0":0,"imp1":0,"version":5,"voltage":3.011,"version_esp":"0.8.2","key":"D26FD0AA69300FDDB3FD7069EBD28762","resets":1,"email":"nikolay.sikorsky@gmail.com"}
000:00:00:04:258  NOTICE    (NTP) : Waiting for NTP time sync:
000:00:00:04:358  NOTICE    (NTP) : Current time: Mon May 27 22:14:50 2019

000:00:00:04:359  INFO      (RQT) : Begin client
000:00:00:05:415  INFO      (RQT) : Response code:      200
000:00:00:05:415  INFO      (RQT) : Response body:
000:00:00:05:416  INFO      (RQT) : Result:     Success
000:00:00:05:418  NOTICE    (RQT) : -- END --
000:00:00:05:422  INFO      (SND) : Send HTTP code:     200
000:00:00:05:427  INFO      (SND) : Send result:        Success
000:00:00:05:432  NOTICE    (SND) : -- END --
000:00:00:05:437  NOTICE    (CFG) : Config stored OK
000:00:00:05:441  NOTICE    (ESP) : Going to sleep

```

Dev ветка WiFiManager гораздо стабильнее, хотя бы не падает 

```
platform = https://github.com/platformio/platform-espressif8266.git#v2.2.1
WifiManager = https://github.com/tzapu/WiFiManager.git#development
```

dontsovcmc, 2019-06-05:
`хотя бы не падает`
@OloloevReal это очень важно: в какой вы ветке тестируете, что не падает? И какая версия arduino фреймворка?

dontsovcmc, 2019-07-02:
Аналогично MQTT+HTTPS не работают =(

dontsovcmc, 2019-09-15:
С новым фреймворком все окей

closed 2019-09-15 by dontsovcmc