dontsovcmc/waterius #86: Google HomeAssistant тест+инструкция

dontsovcmc, 2019-06-26:
Появилась конфигурация Ватериуса для HomeAssistant
https://github.com/dontsovcmc/waterius/pull/85 за что респект и спасибо Антону @freenetwork!

Требуется: 
1. Проверить конфигурацию, внести корректировку
2. Написать инструкцию: что настроить в Ватериусе и как добавить его в HomeAssistant

DimsterTim, 2019-09-04:
Как я понял, предложенная интеграция была расчитана на встроенный MQTT брокер. Но этот функционал в HomeAssistant посчитали "велосипедом" и несколько версий как не поддерживают, а далее планируют убрать. Так что вся интеграция сводится к уже имеющемуся функционалу отправки данных на mqtt сервер. А в настройках HA в файле yaml уже вставляется предложенный код для "датчиков" с подпиской на топики mqtt в формате:
```
sensor:
  # waterius на кухне
  - platform: mqtt
    name: "Kitchen Cold Water"
    state_topic: "waterius/kitchen/ch0"
    value_template: "{{ value }}"
    unit_of_measurement: "м3"
    icon: mdi:water
```
где sensor - тип "датчика" в терминах НА
platform: mqtt - формат данных для датчика
name: - отображаемое название датчика
state_topic: - тема для подписки на обновления данных mqtt
value_template: - формат парсинга полученного от mqtt значения, в данном случае просто берется "значение" как есть, но может быть много вариантов, JSON, вычисления и тд.
unit_of_measurement: "м3" - очевидно, что единицы измерения для показаний
icon: - иконка из набора НА.

Всего вы можете создать 12 "датчиков" с соответствующими "state_topic:" mqtt:
```
waterius/6901727/ch0
waterius/6901727/ch1
waterius/6901727/delta0
waterius/6901727/delta1
waterius/6901727/voltage
waterius/6901727/resets
waterius/6901727/good
waterius/6901727/boot
waterius/6901727/imp0
waterius/6901727/imp1
waterius/6901727/version
waterius/6901727/version_esp
```

Таким образом для интеграции Waterius в НА вы настраиваете отправку данных на mqtt по имеющейся [инструкции для mqtt](https://github.com/dontsovcmc/waterius/blob/master/Mqtt.md). В НА нужно будет добавить в файл конфигурации .yaml "датчик" настроенный на обновления значений от mqtt сервера в формате, разобранном выше. Затем, в НА датчик "Kitchen_Cold_Water" используется для отображения данных или в алгоритмах автоматизации, [построении графиков](https://www.home-assistant.io/lovelace/sensor/).

dontsovcmc, 2019-09-04:
@DimsterTim файл yaml https://github.com/dontsovcmc/waterius/pull/85/commits/61b58f9db0261c302814e46e2e21563ffad5ec70 содержит ту же самую информацию, которую вы написали выше - описание сенсора. 
Единственное в нем 2 ватериуса.
Вы могли бы сделать конфигурацию для 1 Ватериуса?

DimsterTim, 2019-09-04:
Я описал общий принцип. В файле по ссылке на mqtt сервере данные от двух устройств: waterius/kitchen/ и waterius/bathroom/ я же использовал ссылку на настройки mqtt клиента из [вашего примера](https://github.com/dontsovcmc/waterius/blob/master/Mqtt.md): waterius/6901727/ 
вот и вся разница.

dontsovcmc, 2019-11-26:
Пользователя Domoticz обогнали вас =)) https://www.hackster.io/dontsovcmc/domoticz-4346d5

HomeAssistant предоставляется возможность, не писать инструкцию, а написать какие данные куда слать по MQTT и какие настройки нужны в устройстве. Возможно удастся добавить сразу.

freenetwork, 2019-11-30:
Ммм, надо гайд?
У меня есть package. Вставляеь файл, меняешь ID и полетели. Но mqtt должен быть настроен

dontsovcmc, 2019-11-30:
А для HomeAssistant не нужны всякие NodeRed? 
Я не пойму, нужно добавлять чтото в прошивку или достаточно иметь конфиг+поменять mqtt topic в ватериусе?

closed 2019-12-29 by dontsovcmc

AlexSZbrn, 2022-09-22:
Народ, создатели HA выкатили апдейт, и теперь ваш конфиг пока работает, но помечен как deprecated, в следующем апдейте может уже не заработать... надо переписывать. попробую сам, но мозгов и времени может не хватить, потому довожу до сведения сообщества. всем 73!

neitri, 2022-09-24:
Я сильно  не вникал, но там вроде две строки меняется. -platform: mqtt обзавёлся собственным доменом mqtt: