dontsovcmc/waterius #88: killfeature: Кнопка "я уехал" для оповещения о расходе

dontsovcmc, 2019-07-31:
Клиентом предложена идея: добавить кнопку "я уехал". Чтобы если был расход воды, то Ватериус оповестил бы об этом. 
В Blynk.cc есть оповещения на телефоне - уже решение. 
Для сайта waterius.ru можно добавить отправку СМС и звонок. Но этот функционал можно будет сделать при переходе на другую микроконтроллер... хотя...
На Attiny кнопку можно повесить на reset и смотреть при включении "причину" - если причина перезагрузки кнопка, то включить функцию отправкой статуса на сервер.

OloloevReal, 2019-07-31:
Почему бы не использовать ОЧЕНЬ длительно удержание кнопки для активации режима
короткое нажатие - отправка данных
длительное нажатие - режим настройки
очень длительное нажатие - активация режима отпуск

В любом случае придется лезть в сантех нишу для доступа к устройству, что очень лень. В блинке можно кнопку нажимать и передавать активацию режима при следующее передаче показаний, максимум получится задержка в сутки, ну и что

dontsovcmc, 2019-07-31:
О, передача через Блинк тоже вариант =)

dontsovcmc, 2019-12-29:
Возможная реализация через сервер:
- при передачи показаний получать в ответ настройки. Если включен "отпуск" ,то в attiny посылается команда "режим отпуск". И Attiny начинает будить ESP каждый импульс, пока режим отпуск не выключится.
Требуется доработка: attiny, ESP, сайта

dontsovcmc, 2020-07-02:
Есть проблема: счетчики воды НЕ крутятся при низком потреблении. Поэтому удастся детектировать только большой прорыв трубы.

closed 2020-07-02 by dontsovcmc