dontsovcmc/waterius #106: При кол-ве Wi-Fi больше 20 страница не грузится

dontsovcmc, 2019-11-25:
При кол-ве сетей Wi-Fi в списке больше 24 страница не умещается в 12000байт и не загружается полностью

dontsovcmc, 2019-11-25:
@tablatronix is it possible to add /ssids handler and return networks in /wifi page asyncronously?
I got memory limit with 24 ssid's and want "unlimited" decition.

1. /wifi page onload function call "/ssids"
2. esp call 
WiFi_scanNetworks(server->hasArg(F("refresh")),false); //wifiscan, force if arg refresh
and return html code
3. JS code change network 'div' html code to returned by handler.

tablatronix, 2019-11-25:
Is this the same page string running out of memory

dontsovcmc, 2019-11-25:
> Is this the same page string running out of memory

Yeap! I reserved(12000), page larger. That's why I want to load networks async. Is there is any problems with calling WiFi_scanNetworks in handler?

dontsovcmc, 2019-11-28:
It works =)

closed 2019-11-28 by dontsovcmc