dontsovcmc/waterius #107: Заменить разъемы на Wago 235

dontsovcmc, 2019-11-25:
"хватит издеваться над пользователями", говорит мой дизайнер =)

Эти разъемы гораздо понятнее людям. И дырка под провод больше:
[image]

OloloevReal, 2019-11-25:
Эх, клемники жир блеск красота, но где их взять в шаговой доступности. В широко известном магазине который упаковывает детальки в прозрачные пакетики их нет

dontsovcmc, 2019-11-25:
> Эх, клемники жир блеск красота, но где их взять в шаговой доступности. В широко известном магазине который упаковывает детальки в прозрачные пакетики их нет

Это да, может ИП как-то мне поможет.

dontsovcmc, 2019-12-29:
Заказал фантастически красивые разъёмчики - красненькие и синенькие! Обещают 50-90 дней доставки... ппц. Быстрее друзей попрошу привезти из Европы.
Придется поднимать цену за пафос и удобство втыкания проводов =)

dontsovcmc, 2022-01-19:
Эти разъемы лучше, чем текущие, но тоже не панацея. Пока закрою задачу до лучших времен.

closed 2022-01-19 by dontsovcmc