dontsovcmc/waterius #109: Индикация низкого заряда

OloloevReal, 2019-12-03:
В копилку идей - в режиме TRANSMIT_MODE (по таймеру или при нажатии кнопки) если напряжение питания ниже определенного значения (какого?) не включать WiFi, а сигнализировать о недостаточном питании, например, пятью короткими вспышками LED1

Это позволит продолжить собирать данные со счетчиков еще очень продолжительное время, но не передавать их. А при попытки ручной передачи сразу станет понятно в чем дело.

Т.к. индикатор LED1 подключен к ESP поморгать индикатором можно будет только разбудив ее.

dontsovcmc, 2019-12-03:
О! Вот зачем моргать надо, когда плохо! А я никак не мог понять, зачем гирлянды вешают на устройства. 
Есть нюанс: просадка идет во время DHCP т.е. за 0.1 секунду с включения не понять, что с питанием плохо =(. 

Я вставил в Ватериус батарейки ИКЕА, которым 5 лет, и включил отправку раз в 30 минут:
Графики "усредненные":
[image]

[image]

Вот 2 точки включения
[image]

Это график падения напряжения за время подключения к роутеру ~5сек

dontsovcmc, 2023-10-31:
У ESP нет аналогового входа =(. Вариант с ключом на транзисторе и читать свободный пин RX только.

closed 2023-10-31 by dontsovcmc