dontsovcmc/waterius #110: Корпус

dontsovcmc, 2019-12-04:
Для меня не стало сюрпризом, что некоторые управляющие компании хотят закрывать устройство от пользователей. Пломбировать и вот это вот всё. 
Давайте найдем удобный корпус для Ватериуса =).

Требования:
- размер не менее 65х65х30мм
- брызгозащитный ввод провода
- водонепроницаемая кнопка
- будет круто, если не потребуется отвертка для открытия корпуса

OloloevReal, 2019-12-05:
Если DIY то не плохо подходят распред коробки

[image]

yarolig, 2019-12-15:
Недавно собрал такой. 
Кнопка недоступна, остальные требования удовлетворяются:

[image]
[image]

dontsovcmc, 2019-12-15:
@yarolig класс! брызгозащитный - если лить воду, ничего не замкнет.

yarolig, 2019-12-15:
Ага. Чтобы всё продолжало работать, когда рядом внезапно полились фонтаны. Теоретически, дырку для проводов можно заделать горячим клеем или силиконовым герметиком.

dontsovcmc, 2019-12-15:
@yarolig эт да. Любопытно найти готовый корпус, чтобы и герметичный и без винтов всяких.

Кстати, таким может быть крепление на трубу:
[image]

dontsovcmc, 2022-07-26:
Компаунд для заливки
https://nomacon.ru/katalog-tovarov/teploprovodyashhie-elektroizolyacionnye-materialy-kptd/kompaundy-zalivochnye-teploprovodyashhie-elektroizolyacionnye-nomakon-kptd-1.html

dontsovcmc, 2023-04-30:
[image]