dontsovcmc/waterius #111: Схема не соответствует надписям на плате v0.2

sintech, 2019-12-18:
Номера резисторов на схеме не соответствуют шелкографии на плате версии 0.2.
https://raw.githubusercontent.com/dontsovcmc/waterius/master/Board/scheme.png
и
https://raw.githubusercontent.com/dontsovcmc/waterius/master/Board/waterius-factory-board2_top.jpg
https://raw.githubusercontent.com/dontsovcmc/waterius/master/Board/waterius-factory-board2_bottom.jpg
Например, на плате R11 а на схеме R8, на плате R9 а на схеме R10.

dontsovcmc, 2019-12-18:
@sintech да, на плате R10 лишний и что-то съехало =(. Вы могли бы поправить в соответствии с фото платы? И схему скриншотом в png обновить.

sintech, 2019-12-19:
Готово, #112

dontsovcmc, 2019-12-21:
Спасибо!

Вопрос! Я вижу вы со звуком работали. Есть такой девайс Amazon Dash Button. Она настраивается звуком со смартфона. Вам понятно, что тут написано про модуляцию: http://www.blog.jay-greco.com/wp/?p=116 ?
Если да, насколько трудно подобное закодить?

sintech, 2019-12-21:
Пожалуйста!
А как это можно использовать в проекте? Там описан способ передачи имени сети и пароля wifi Dash Button'у, но дальше он начинает работать напрямую с серверами Амазон.
Судя по описанию вроде бы не сложно: 4 частоты, каждая из которых соответствует 2 битам.
18130=00, 18620=01, 19910=10, 19600=11. Ну и сам протокол передачи.

dontsovcmc, 2019-12-21:
> Пожалуйста!
Добро пожаловать в ряды контрибьютеров =)
> А как это можно использовать в проекте? 
Тогда можно будет не поднимать Wi-Fi, а настраивать прямо в браузере! Режим настройки будет включать микрофон и ждать данных. В браузере заполняем анкету, нажимаем  кнопку "записать", которая включит звуковые данные. Ватериус их примет, моргнет и выйдет на связь - отвал башки функционал! =)
Паять, конечно, побольше, но юзабилити получше. Плюс я не видел такую библиотеку для Ардуино, а значит open source сообщество будет благодарно.

dontsovcmc, 2019-12-28:
Спасибо

closed 2019-12-28 by dontsovcmc