dontsovcmc/waterius #115: Test of Waterius

terminuslat, 2019-12-23:
Я сильно извиняюсь. Если я правильно понял, то для тестирования надо:
1. залить в Ардуину скетч " test_attiny85" (доделанный программатор :) ) С этим проблем не возникло. Остался вопрос для перехода в режим "тестера" кнопку на чём нажимать и в какой момент?
2. А вот для самого Ватериуса предназначен "attiny85_measure.cpp"? Скомпайлить его не получается из-за чехарды заголовочников *.h  . Не затруднит ли выложить *.HEX  или, что наглядней, рабочий комплект исходников?
Заранее признателен...

dontsovcmc, 2019-12-23:
0. Не надо извиняться! Нужно, чтобы все инструкции были понятны и подробны с FAQ. Буду рад помощи! Хоть в гугл доке, хоть пул ревестом, хоть статьями на форумах - всё в копилку.

1. во-первых, скетч " test_attiny85" для платы Ардуино ,которая будет тискать attiny.

DETECT_PIN Ардуино подключаете в разъем EN ESP8266 (т.е. к пину 1 Attiny85) - тинька будет будить Ардуино и Ардуино запросит с нее данные.

2. нет, это какой-то другой тест для отладки.. Идея теста test_attiny85, что раз в ~10 сек тинька пробуждается и "сдает" показания Ардуино (i2c master).

ВАЖНО: 
1. файлы i2c_master устарели. Надо взять из папки attiny
2. Проект attiny нужно перекомпилировать, т.к. вы будете сутки ждать выхода на связь тиньки. 
#define ONE_MINUTE 240  - меняйте на другое число
WAKE_EVERY_MIN и эту меняйте
ONE_MINUTE * WAKE_EVERY_MIN  = кол-во пробуждений для проверки входов. сейчас период 250мс , поэтому ONE_MINUTE = 240 раз  = (1 сек / 250 мс)  * 60 сек

terminuslat, 2019-12-23:
Размотаю ответ с конца.. 
3. «А что вы хотите сделать?» Я хочу повторить Waterius.  :)  Для себя любимого. Если получится, попытаюсь размножить, но об том пока рано. Для меня программирование МК – хобби. Правда паяю я уже только с лупой и туплю слегонца– мне 6- десяток. Потому вопросы могут быть неразумны порой. Я заморочился подобным девайсом несколько ранее и уже смакетил его на отладочной плате  с STM8L152.. Считает.. :)  Правда без WiFi. Но пришло осознание, что STM избыточен. Там пока отключишь лишнюю периферию – страницы две кода, не считая библиотек. Правда ног много, датчиками протечки и пр. хоть обвешайся, хоть свой LCD лепи : ) и с обновлением «на лету»  проблем нет..  Ну, правда,  спит он с RTC таймером и правда  с 4 мкА. И тут на фоне этой избыточности я и наткнулся на Waterius… :) 
«Как протестировать? С какой целью?» Дык я собрал девайс на макетке. Прозвонился по полной. Вроде ок. Прошился (вроде). Не работает. Вот по тому и хочу проверить отдельно плату с прошитой тинькой. Методично, но медленно, сначала убедЮсь, что сигналы идут, Потом, посредством внешней ардуины-уны, что идут правильно – посмотреть в мониторе порта  - бесценно. :). Потом займусь ESP. Кстати выложенную её прошивку шить начиная с 0x00000 ? (ESP8266Flasher об том спрашивает, и с загрузчиком или без?)
2. & 1. & «важно»    Завтра попробую выполнить инструкции. Начну с Arduino IDE, Может потискаю в  Atmel Studio, я оттуда пока только тиньку шью. 
0.	После того, как отлажусь, обязуюсь написать пошаговую инструкцию пройденного пути!. (Хотя бы чисто на поржать).   Если разберусь досконально, смогу попробовать перенести на STM.

dontsovcmc, 2019-12-23:
Вау! Очень круто! Ничего себе, я не осилил STM8, потому что в наше стремительное время ему не написали кучу библиотек, а рыться в даташитах некогда.
Чтобы проверить, прошили ли вы attiny достаточно нажать на кнопку и отпустить - она должна подать HIGH на ЕСП пин. А еще можно скомпилировать какое нибудь моргание светодиодом.. но это проверит работоспособность тиньки, компилятора. Вам еще там фьюзы 1мгц надо выставить.
Атмел студия хорошая, в ней собирался проект, но там сложно, смысла нет.
Я бы порекомендовал visual studio code + platformio plugin. Долго ставить, зато потом легко.
Ардуино ИДЕ тоже можно. Но вручную настривать и скачивать библиотеки.
Platformio сама все сделает. И не потому что мы ленивые, а потому что 21й век.
Терминал, да, наше всё!
Ыыыы не помню про ЕСП. Адрес точно с 0х0. Я ее esptool иногда шью.

Кстати код под STM8 если будете выкладывать, скиньте, я бы посмотрел, как оно там делается. Для проекта нужна связь с ЕСП. У меня она по i2c, думаю в СТМ8 это реализовано давно.

Независимая инструкция - это моя мечта!

terminuslat, 2019-12-24:
С конца как всегда удобней ответить :)
Независимая инструкция будет и будет предельно конкретной и понятной, это я умею писать (в отличии от кодить ))). Если разберусь сам :)
Про код под STM. Я писал выше, что пока задвинул свой проект, связью и не начинал заниматься, но аппаратных uart-ов, i2c- шек и прочих can-шин там есть для вас сколько надо.  Кстати, вы не правы: библиотек и примеров на оффСайте как шпал до Питера. Библиотеки Актуальны и конкретны под серию камней, в нашем случае L ( в ардуине , кстати, только серия S). Примеров с описаловом тоже есть почти на все случаи жизни, но они больше под снятые с производства девелопные платы. Но фантазия помогает. Тяжело там с прерываниями, тут понадобится уже и даташит и мозги. Крч, избыточен камень сей… Посему у меня нет готового годного проекта (!), только отладочный(!), чисто на посмотреть. Тем не менее, по вашей просьбе,  я перезалил один из своих проектов тут. https://github.com/terminuslat/sklad .  Язвительные коменты в main.c присутствуют.  Проект  писан,  собран и компайлится без ошибок в среде ST Visual Develop (STVD-STM8 https://www.st.com/en/development-tools/stvd-stm8.html)  с компилятором  Cosmic. Среда средне дружелюбная с жёстким С++. 

Светодиод для моргания у меня прямо на переходнике для прошития. То,  что я шью Тиньку успешно, я не сомневаюсь. По фьюзам тоже ок (читаю, пишу, и даже сбрасываю – собрал себе 12 вольт  ресеттер) .  Пляски с бубном вокруг программатора Атмел МК2 тоже  позади. Вопрос что я шью, из чего собран проект, я понимаю, что библиотеки могут устаревать, обновляться и тп.. Посему вижу только один способ  напрячься собрать один раз, а потом всё стороннее скопипастить себе, чтоб не дать проекту расползаться….
Про среду. Если ВЫ хотиТЕ расползания Ватериуса в народе, то надо иметь 100% локальный проект в основном пристанище продвинутых ламеров – Arduino IDE. Остальное – дело вкуса. :)
И так я не справился с ходу.  Прошивка *.hex  тоже не дала мне ничего.  Завтра пройдусь по данным мне советам… поглядим…
Про «Вау! Очень круто!» . Уметь играть в шахматы означает вариации от знания правил хода фигур до уровня гроссмейстера. Ссылку для ознакомления с лично моим быдлокодом я привёл выше. :)

dontsovcmc, 2019-12-24:
1. Не запушили: https://github.com/terminuslat – пусто. git push origin master
2. По кол-во пинов может и избыточен, но есть свои плюсы.
3. для Arduino IDE скорее нужна инструкция в картинках.
мда, у меня целых 5 напутственных строчек: https://github.com/dontsovcmc/waterius/blob/master/Firmware.md#%D0%BF%D1%80%D0%BE%D1%88%D0%B8%D0%B2%D0%BA%D0%B0-%D1%81-%D0%BF%D0%BE%D0%BC%D0%BE%D1%89%D1%8C%D1%8E-arduino-ide

terminuslat, 2019-12-24:
Странно, что не запушилось. Я как раз сегодня с утра перекомпильнул для проверки и перезалил. Я его вижу на своём sklade. Видать в конце мимо enter  промахнулся. ( Бывает . Вечером проверю....
Избыточен он не столько пинами, сколько всякими там ЦАП АЦП контроллерами lcd etc. А вот на нормуль прерывания места в камне не хватило. ...

terminuslat, 2019-12-24:
Вроде от'PABLIC'ил проект. Виден без sign_in'a

dontsovcmc, 2019-12-24:
Да, теперь ок. Я месяц убил на прерывания, а потом понял, что если счетчик остановится в момент замыкания, то тинька не засыпает (. Поэтому выкинул кучу кода и сделал периодический опрос. Заодно защита от дребезга

terminuslat, 2019-12-25:
Поскольку в водосчётчик традиционно ставят геркон, борьба с дребезгом несложна. Вообще-то у тиньки вроде есть тип прерывания по любому фронту. При этом она не останется включённой 
Но , конечно, основной потребитель - цепь самого счётчика. Посему решение у ватериуса правильное. Так сделано , наверное, у всех. Я собирался чуть по другому решить вопрос.

terminuslat, 2019-12-25:
Дело в том, что у stm8l есть забавное прерывание- по скачку напряжения питания. Вейкапит даже из режима полного вырубания камня так вот можно на сборках аналоговых ключей сделать переключение выходов счётчика от ног камня к цепи питания камня в спящем (halt) режиме . Счётчики последовательно естессно. Тем паче есть особенность у stm8l - ток сна при 2 вольтах типа как заявлено, 4 мкА, а при 3,3 уже 40, что было сюрпрайзом.....

dontsovcmc, 2019-12-25:
Я вам говорю, что тинька не засыпает при приходе прерывания и замкнутом герконе. 
Вы сложное решение какое то делаете.
 Не может быть разница в 10 раз у СТМ, чтото не то. Там всю периферию надо выключить.

terminuslat, 2019-12-25:
* Про бодрствование тиньки верю, но проверю.. )))
**Сложное у же не делаю. ) Я для себя установил критерий - если интегрально (сон-не сон) ток потребляемый сабжем сравним с током саморазряда батарейки, то нефиг и искать сложности.
***Всё то. У платы девелоперской https://www.st.com/en/evaluation-tools/stm8l-discovery.html есть своя  внутренняя  схема измерения тока камня. ну и регулируемый источник питания есть у меня ) периферия отключена и не меняется. я просто кручу напряжение и меряю ток. Зависимость однозначная. Это просто особенность схемотехники stm8l

terminuslat, 2019-12-28:
Как говаривал г- н Швондер, это просто какой-то… позор …
Ох , чёт не клеится собрать проект в ардуине. Снёс и переставил IDE, вычистив всё лишнее, чутка полегчало.
Итак, по порядку. Вопросы, ответы на которые облегчат мне жисть ( заранее спс).
1.	Проект test_attiny85, выложенный  по пути waterius/tests/test_attiny85/src/,  предназначен для заливки в некую Arduino UNO, которая призвана заменить на время отладки ESP-01. Это так?
1.1.	 Этот проект успешно компайлится в Arduino IDE, но по нему есть замечание автора  относительно файлов MasterI2C.cpp и MasterI2C.h.  Однако выше написано, что устарели файлы «i2c_master» (на самом деле master_i2c.h . Но, по дате выкладывания на  Github они более свежи.  И лежат они не в «Attiny», а в waterius/ESP8266/src/, что логично. Соответственно вопрос: Какие файлы использовать для проекта test_attiny85 ?
1.2.	Нужно  поменять внутренности в  MasterI2C.* или пробежаться по коду  поменять название на master_i2c.* ?
2.	  Проект waterius/Attiny85/  собрался, компайлится. Но пока не заливал. Нет резона ибо пока в тиньке готовый *.HEX, залитый в Atmel Studio. Осциллоскопом тоже лень лазить пока, так как п.1 под вопросом.
3.	Перед вопросом №4 есть одно замечание. Каким то волшебным образом ArduinoJson и PubSubClient  присутствовали в комплекте поставки IDE 1.8.10. Попытки корректного засовывания через zip  приводили к ругани, а простое подсовывание в нужную папку тоже  к ругани, но у же при компиляции. В моей личной инструкции будут пара пунктиков, как их проставить изнутри и в нужной версии. Ибо двоить нельзя.  
4.	С проектом по адресу waterius/ESP8266/ пока полный швах.  При залитом готовом  *.HEX  узреть  точку Waterius_0.8.2 не получается никак.  Понятно, что либо я криво шью, либо не работает плата с тинькой ( может я и недорукожопил некий контактик).  В силу чего очень прошу запродать мне пару комплектов для сборки (или готовые) .  С прошитыми esp  и тинькой, конечно.  Надо же что то на чём то проверить. ))
5.	С проектом по адресу waterius/ESP8266/  пока такой вопрос. Что означает фраза «Установить вручную из zip WiFiManager#waterius_release_090» ?  По моим скромным мозгам я знаю три папки, куда можно подсунуть «ручками»  библиотеку и она подхватится, или же скетч/подключить библиотеку/добавить zip... или же нужно ручками поменять нутро какого либо из уже существующих WiFi манагеров?
…..

dontsovcmc, 2019-12-28:
Это ни в коем случае не позор. Проект, не такой простой, как хотелось бы и есть зависимости от библиотек, которые вручную надо ставить.
Итак, по порядку:
1. Да
1.1. ух они могут не собраться под ардуино. 
Если будет ругаться на setClockStretchLimit, закомментируйте этот вызов.
1.2. да, взять файлы из esp8266 и переименовать include

2. Проверить работу тиньки элементарно: надо замкнуть пин EN с пином TX  к которому светодиод подключен. При коротком нажатии кнопки он должен загореться на время WAIT_ESP_MSEC. Если загорелся, уже можно судить об успехе.
3. Вот тут перечислены библиотеки и версии, требуемые для компиляции ЕСП:
https://github.com/dontsovcmc/waterius/blob/master/ESP8266/platformio.ini
Все их кроме wifimanager ставим из поиска библиотек в ИДЕ.
А wifimanager расположен у меня. И в отдельной ветке. Ее надо на гитхабе выбрать и скачать зип. Его скормить ИДЕ.
https://github.com/dontsovcmc/WiFiManager/tree/waterius_release_099 вот эта ветка.

4. ЕСП без тиньки не включиться. А если зажать EN без тиньки, то включится, не получит ответа от тиньки и заснет. Работу ЕСП просто проверить: подключить к ТХ usb-ttl.и псмотреть лог на 115200 8N1.

5. Ответил. В гите бывают ветки. Их можно переключать. Обычно ветка  master. Но т.к. я ркгулярно вношу изменения в wifimanager,  я вынужден создавать каждый раз новую ветку, чтобы не заботиться об обратной совместимости.

dontsovcmc, 2019-12-30:
@terminuslat давайте в чат перейдем https://gitter.im/waterius
чуть дополнил https://github.com/dontsovcmc/waterius/blob/master/Firmware.md#%D0%BF%D1%80%D0%BE%D1%88%D0%B8%D0%B2%D0%BA%D0%B0-%D1%81-%D0%BF%D0%BE%D0%BC%D0%BE%D1%89%D1%8C%D1%8E-arduino-ide

terminuslat, 2019-12-30:
Да, конечно в чат, там есть подходящий :)

dontsovcmc, 2020-02-08:
@terminuslat чат по сборке https://t.me/waterius_forum

dontsovcmc, 2020-04-11:
@terminuslat как успехи то?

terminuslat, 2020-04-14:
Я пока призабил. ( у меня заказик за денежку делается, как ни смешно тоже с водосчетчиками) А  так я походу накосячил в монтаже. раскапывать уже некогда было.  я поэтому и хотел купить пару комплектов . Или из под ардуины_ Ide не собирается норм. В общем муть пока. Заказ я свой к лету закончу, вернусь в тему.  Но если есть возможность купить комплекте для сборки, я готов.

dontsovcmc, 2020-07-02:
@terminuslat как успехи? комплекты скоро прекратим продавать — нет времени. 
Для Arduino IDE не хватает подробной инструкции =(. А там куча нюансов, т.к. используем не установленные сразу библиотеки.

closed 2020-11-30 by dontsovcmc