dontsovcmc/waterius #124: Настройка без вебсервера: широковещательными запросами

dontsovcmc, 2020-02-29:
Заботливый клиент рассказал про новую технологию настройки.

"технология такая вкратце
надо нажать кнопку
и на счетчике активируется wifi
который слушает широковещательные запросы
а на телефоне указывается SSID и пароль
и телефон в эту же сеть отправляет в пакете в поле длинна данные о сетке и пароле
счетчик это слушает"

[инструкция к счетчику с вайфай](https://www.meters.taipit.ru/upload/%D0%A2%D0%90%D0%A1%D0%92.411152.002%20%D0%982_%D0%98%D0%BD%D1%81%D1%82%D1%80%D1%83%D0%BA%D1%86%D0%B8%D1%8F%20%D0%BF%D0%BE%20%D0%BF%D0%BE%D0%B4%D0%BA%D0%BB%D1%8E%D1%87%D0%B5%D0%BD%D0%B8%D1%8E%20%D1%81%D1%87%D0%B5%D1%82%D1%87%D0%B8%D0%BA%D0%B0%20%D0%BA%20%D0%BE%D0%B1%D0%BB%D0%B0%D1%87%D0%BD%D0%BE%D0%BC%D1%83%20%D1%81%D0%B5%D1%80%D0%B2%D0%B8%D1%81%D1%83%20%C2%AB%D0%A2%D0%90%D0%99%D0%9F%D0%98%D0%A2%C2%BB%20(%D0%9C%D0%A21%D0%A54%20WF1)_[6133205]_06.08.19.pdf)

neitri, 2022-01-23:
Пробовал добавить в прошивку.
После перехода в режим настройки, в течении 10 секунд необходимо второй раз нажать кнопку отправки. После этого можно передать имя сети и пароль. Что интересно, после такой настройки можно подключиться как по адресу из домашней сети, так и по сети Waterius***.
В целом работает, но настройки через родной диалог это не отменяет. Единственное что можно настроить через компьютер. Но телефон все равно потребуется.
Актуально?

dontsovcmc, 2022-01-23:
Спасибо за эксперимент!
Я думаю, надо провести изучение потребительских свойств при такой надстройке: выглядит сильно проще. Но тогда мне на сайт надо будет добавить ввод начальных значений счётчиков. 

Если можно выложите в отдельную ветку smartconfig код. 
А настраиваете каким приложением для телефона? Я видел esp-touch

neitri, 2022-06-05:
Появилась одна мысль.
В виду того, что передать можно только имя сети и пароль можно попробовать так.
1. в кабинете предварительно регистрируем новое устройство. Вводим для него имя сети wifi и мак адрес. Северная часть сгенерирует тот же хэш что и ватериус.
2. В кабинете конфигурируем ватериус.
3. Настраивает через smartconfig. Ватериус генерирует хэш по имени сети и мак адреса такойже как в серверной части. И получает по хэшу параметры.
Как вариант использовать разные хэши для настройки и работы.