dontsovcmc/waterius #127: Установка частоты соединений и Конфигурация по MQTT

Fortunatto, 2020-03-18:
Enhancements:
1) Возможность увеличивать частоту соединений с сервером. Обоснование: у части пользователей в стояк уже подведено электричество - для управления кранами запирания воды. Части пользователей среднего и выше уровня продвинутости не составит труда купить один LIion аккумулятор и плату зарядки и подключить к нему Waterius. Это снимает ограничение по энергопортеблению и позволяет чаще публиковать данные на сервере/mqtt.
Ради чего это нужно(пример): в системе УД настроен режим "Дом пуст" в рамках которого собираются данные со счетчика воды, и да, в то время, когда краны закрыты. При этом, состояние "Waterius насчитал литры + закрытые краны" говорит о том, что а) запирающий кран не закрылся до конца (для автоматики -> провести повторный цикл блокировки, для меня(пользователя) -> запирающий кран неисправен и  нуждается в проверке/замене). Это не фантазия, это моя реальная ситуация, в которую я попал. Хорошо, что не попал на стоимость ремонта - соседи снизу оказались хорошими людьми и не приняли во внимание небольшие подтеки. И да, датчики протечки есть, но сама протечка шла по трубе, до датчика вода не добралась.
Для реализации этого, с учетом того, что публикация данных по mqtt уже сделана, осталось
2) научить Waterius считывать данные по количеству требуемых публикаций из специального топика mqtt.

Т.е. в принципе все это, наверное, небольшая доработка софтверной части продукта. И плюс к его универсальности.

dontsovcmc, 2020-06-18:
@Fortunatto "научить Waterius считывать данные по количеству требуемых публикаций из специального топика mqtt." а можете пояснить, что такое требуемые публикации?

По остальному:
1. Вывести настройку периода выхода на связь в интерфейс
Вопрос: а что лучше, увеличение периода выхода на связь или раз в сутки слать 24 значений? 
Архивные данные:
Минус: В MQTT же нет времени, поэтому для него может быть проблемой архивные данные.
Плюс: Ежечасные показания с работой от батарейки.

2. Включение режима "нет дома"
Вопрос:
1. как включать? Подписываться на mqtt топик и получить retain данные о требуемом режиме?
2. как сигнализировать о расходе? писать в mqtt топик? на сайт ватериуса можно и СМС шлюз подключить, в принципе.

wrewolf, 2022-04-15:
Я ща возможно некропостну, но с недавних пор у меня ватериус начал постить сообщения в mqtt раз в 10 минут практически

dontsovcmc, 2022-04-15:
> Я ща возможно некропостну, но с недавних пор у меня ватериус начал постить сообщения в mqtt раз в 10 минут практически

Воу! Напишите мне в телеграм, надо разобраться- так не должно быть!

wrewolf, 2022-04-15:
> > Я ща возможно некропостну, но с недавних пор у меня ватериус начал постить сообщения в mqtt раз в 10 минут практически
> 
> Воу! Напишите мне в телеграм, надо разобраться- так не должно быть!

Сделал это

Spirituss, 2022-10-09:
> Я ща возможно некропостну, но с недавних пор у меня ватериус начал постить сообщения в mqtt раз в 10 минут практически

У меня наоборот - поставил период обновления 5 минут, в ЛК раз в 5 минут обновление показаний происходит, в MQTT - нет. Период обновления в MQTT не понимаю, но обновления не происходит.

Spirituss, 2022-10-09:
Посмотрел - примерно раз в час mqtt обновляется.

dontsovcmc, 2023-04-29:
В последней версии появилась возмжность настроить период отправки через MQTT! 
Требуется доработать до: https://github.com/dontsovcmc/waterius/issues/273