dontsovcmc/waterius #128: Версия в герметичном боксе и с поддержкой датчиков протечки

badenbaden, 2020-03-25:
Добрый день!

Евгений! Повторил Ваше устройство, спасибо, что делитесь!))
Хочется иметь возможность подключить большее количество счетчиков (например у меня есть 2 стояка и проложен провод между ними) для использования 1 устройства для всех стояков.
Также хочется иметь датчики протечки по нескольким зонам.
Герметичный корпус тоже очень неплохо было бы иметь.

Вот мои мысли:
Т.к. attiny85 не имеет столько ног, то взять предлагаю attiny84 (тем более, что доработка кода там минимальна). Стоимости этих микросхем почти сопоставимы, разница в размерах не критична.
Герметичный корпус на базе распределительной коробки с задатками герметичности.
Обработка датчика протечки по 2 зонам.

Возможные проблемы:
Увеличение количества сохраняемых показаний счетчиков (если их будет больше 2х) повлечет за собой меньший ресурс eeprom.
Как поведет себя attiny при подключении счетчика длинным кабелем.
Возможно увеличится энергопотребление.

Изыскания:
В строительном магазине нашел очень красивую распределительную коробку от ABB. Класс защиты IP65 (т.е. полная защита от пыли и защита от водяных струй), думаю больше и не нужно, защита от морских волн и погружения на 1 метр в квартире рассматривать не стоит.
Кабельные вводы встроены в коробку методом двойной экструзии. Просто прокалываешь шилом или отверткой отверстие и вставляешь кабели. Ввод работает как сальник. Стоимость коробки - 88 рублей. Внутри есть возможность установить и закрепить плату. Вот ее фотки.

[image]
[image]
[image]

Вот мои наброски этой конструкцией в 3d.

[image]
[image]
[image]

Кнопку можно нажимать через подвижный ввод. Светодиод фиксируется в соседнем вводе. ESP между ними (не нарисовал на эскизе).

Собственно для чего я это написал. Заменить на тини84 я могу, плату развести тоже могу, а вот программер из меня хреновый)). Евгений, если есть желание помочь с адаптацией прошивки под тини84 с заявленными плюшками, буду очень рад. В свою очередь предоставлю отчет о проделанной работе, выложу фото получившегося.

dontsovcmc, 2020-03-25:
Добрый день! Да, вы все правильно пишите!
Хочется СТМ32 конечно, но пока желающих нет на него переписать прошивку :(.

Смотрите, я могу помочь с прошивкой, но мне сильно быстрее будет это делать имея плату на руках. Сможете тогда прислать плату обычным письмом и тиньку84? Я оплату скину.

По энергопотреблению: надо МОСФЕТом отключать ЕСП полностью - вот где главная проблема. Она только по даташитам ничего не потребляет, а на деле в зависимости от партии от 2 до 30мкА во сне. Там еще SDRAM же. Если сможете с этим помочь, будет замечательно.
А увеличение входов не сильно повлияет на него.
Можно сразу ЕСП12 поставить, кстати. Чтобы потом ОТА сделать.

badenbaden, 2020-03-25:
Конечно смогу прислать. Займусь на днях платкой. ЕСП12 только из-за 4 мб памяти?

dontsovcmc, 2020-03-25:
> Конечно смогу прислать. Займусь на днях платкой. ЕСП12 только из-за 4 мб памяти?

Плюсы ЕСП в 4мб памяти, потом кстати можно попробовать сделать обновление прошивки attiny84. Но думаю, лучше на СТМ32 перейти все таки.

badenbaden, 2020-03-26:
Набросал схему. Хотелось бы комментариев - вдруг что не так. 
У контроллера есть 1 свободная нога, может кнопку перенести на свободный пин?
Ну и схему первичной прошивки тини и есп надо проверить.

[image]

В протеусе накидал симуляцию ключа для включения есп. Полно дешевых подходящих транзисторов, но собственный ток утечки около 3 мкА. Это конечно меньше 30, но больше 2х. 
А вот с NVTR4503NT1G или NVTR4501NT1G, опять же по данным протеуса, утекает от 0 до 0,01 мкА!!! Надо протестить такой вариант в реальной схеме. 

Не включено:
[image]

Включено:
[image]

dontsovcmc, 2020-03-26:
1. ESP точно надо через полевик выключать. Там только через светодиод (ногу TX) утекает 3мкА =(.
вот тут я еще силовые ключи выписал: https://github.com/dontsovcmc/waterius/issues/3#issuecomment-380848998
по транзисторам помочь не могу - нужен электронщик. может что на форумах пишут? должны же быть энергоэффективные схемы с отключением ЕСП.
 
2. Alarm и wakeleak по своей сути одинаковые входы - везде надо будет будить устройство и передавать состояние входов.
R17 убрать, а контакты 2-4-6 замкнуть. R16 - сделать 300ом.  Возможно вместо него следует дроссель поставить - для длинных проводов наводки убирать.
В целом защита от перезагрузки есть в прошивке.

3. R19, R18 тоже убираются. Для протечки не нужно дополнительных. Достаточно Аналогового входа. Внутри Тиньки есть 20к подтягивающие резисторы. Мы их включаем и сравниваем уровень с покоем. 

4. Вообще говоря, возможно нельзя ESP включать так, как вы предлагаете, лишая GND. Надо со спецами говорить. МК может быть плохо, от того, что питание подано, а земли нет.

5. MCP1700T-3002. На 3В.

badenbaden, 2020-03-26:
6. Значит с кнопкой решили. Ставлю на свободный пин.
5. Точно! Чего же мне взбрело 3.3 поставить. И в действующую плату я на 3.3 поставил))).
4. Можно коммутировать Р-канальным + питания. Но я особой разницы не вижу в данном случае. Как МК может поплохеть, если + заведен на все пины, земли (-) нет, значит и цепи нет. Она только через SDA\SCL соединена с тинькой, да и те подтянуты к питанию. Когда вставляем батарейки, мы же не думаем - как вставить, сначала плюсовую или минусовую клемму. 
2 и 3. Waterleak1 и Waterleak2 это 2 разных входа датчиков протечки, чтобы по зонам определять место самой протечки (на клемме не изменил название, там оба Waterleak1). ALARM - это выход, для информирования или запуска сценариев на внешних устройствах. Также к этому выходу бы прикрутить пищалку или светодиод. R17 думаю нужно оставить, чтобы они не влияли друг на друга. R18, R19 подтягивают входы к питанию, без них что будет на входе с проводом 2-3 метра? И при протечке должно меняться сопротивление между контактами, а если нет подтяжки (в нашем случае это ведь резисторный делитель), то контроллер не поймет разницу между сопротивлением датчика 1 килоом и 1мегаомом - все будет казаться землей.

badenbaden, 2020-03-27:
"Причесал" схему.  Добавил порты для прошивки. Резисторы R18 R19 пока оставил, в крайнем случае их номинал меняется на раз-два, либо ставится перемычка.

[image]

dontsovcmc, 2020-03-27:
Я думаю стоит ставить резистор прям в датчик протечки. Тогда можно будет обрыв линии детектировать, что важно.
Да, R18,19 не мешают. В идеале там пусто будет.
А R30 зачем?

dontsovcmc, 2020-03-27:
Вспомнил еще один плюс от такого Ватериуса: возможность прикрепить внешнюю антенну. Правда это все колхозить только... https://www.youtube.com/watch?v=fQIwG2ooE9k (

OloloevReal, 2020-03-27:
Добрый день! Это не здоровый способ питания esp, может быть утечка в таком случае.
Ставьте отдельный регулятор с EN входом, либо ESP как нагрузку ставьте в коллектор транзиторного ключа.
Как-то так:
[image]

Или на полевых, разницы нет.

dontsovcmc, 2020-03-27:
@OloloevReal Да, я склоняюсь к вашему мнению. 
О, вот есть пример, когда по кнопке подается питание и сохраняется до отключения:  https://www.makermoekoe.net/post/a-dashbutton-like-thing-called-octoclick

В плате Adafruit используется DMG3415 .. Вот бы европейский аналог найти. Мне кажется это то, что нужно.
[image]

badenbaden, 2020-03-27:
> А R30 зачем?
Защита от выгорания порта при коротком замыкании.

dontsovcmc, 2020-03-27:
@badenbaden КЗ на землю не страшно. Посмотрите, здесь про устройство портов: http://easyelectronics.ru/avr-uchebnyj-kurs-ustrojstvo-i-rabota-portov-vvoda-vyvoda.html
Для проверки входа мы:
1. включаем подтягивающий резистор
2. проверяем уровень компаратором
3. выключаем подтягивающий резистор

badenbaden, 2020-03-27:
> Вспомнил еще один плюс от такого Ватериуса: возможность прикрепить внешнюю антенну. Правда это все колхозить только... https://www.youtube.com/watch?v=fQIwG2ooE9k (

> 
Зачем колхозить, когда есть вот такой модуль - ESP07

[image]

[image]

Выводы полностью совместимы.

badenbaden, 2020-03-27:
> @badenbaden КЗ на землю не страшно. Посмотрите, здесь про устройство портов: http://easyelectronics.ru/avr-uchebnyj-kurs-ustrojstvo-i-rabota-portov-vvoda-vyvoda.html
Для проверки входа мы:

включаем подтягивающий резистор
проверяем уровень компаратором
выключаем подтягивающий резистор

> 
Я не догоняю. Как подтягивающий резистор защитит от КЗ? ALARM это выход. Если порт выключен, то все хорошо. А если я перепутал и замкнул землю и ALARM. При возникновении тревоги контроллер подаст на этот выход питание. Порт максимум можно нагрузить на 40 мА (по крайней мере в атмега328 так). При КЗ порт выгорит моментально. А если стоит резистор, то КЗ нет и при данном номинале нет и превышения по току. 
Есть такой популярный контроллер МегаД, так там каждый порт так защищен.

dontsovcmc, 2020-03-27:
@badenbaden ааа я не понял, что Alarm - это выход. Вопрос снят!
О, ESP-07 - это чудесно, спасибо!

badenbaden, 2020-03-27:
Единственное, есп07 точно имеет 512кб памяти, возможно есть на 1мб, а вот по второй картинке нужно узнавать. Это заморочки, если иметь возможность расширение функционала с большим объемом памяти.
По вопросу включения транзистором ЕСП - вопрос можно решить несколькими путями.
1. Оставить существующую схему питания от стандартного ватериуса и использовать модули нормального качества. Тогда и не будет получаться жор от 2 до 30 мкА, в зависимости от экземпляра модуля.
2. Можно стаб с выходом EN
3. Транзистором N-канальным (как я предлагаю) или Р-канальным (коммутация + питания)
4. Возможно еще что-то, до чего мы не додумались)))

Вопрос только в том, что утечка будет в любом случае, а вот где выгодней для нашего случая покажет только натурные испытания.

@OloloevReal - Приветствую! А где будет утечка? Земли на есп нет (при выключенном транзисторе, все входы подключены к +, линии данных SDA и SCL также подтянуты к +. Через что утекать будет. Схема ключа абсолютно рабочая, вот только не тестированная на питании контроллера.

badenbaden, 2020-03-27:
Еще вопрос, а есп впадает в спячку или тини просто вырубает ее выводом EN? Если впадает в спячку, то в какой режим. Это я про оригинальный Ватериус.
И вообще, что мы можем выиграть, вырубая всю есп?

OloloevReal, 2020-03-27:
@badenbaden 

> Приветствую! А где будет утечка? Земли на есп нет (при выключенном транзисторе, все входы подключены к +, линии данных SDA и SCL также подтянуты к +. Через что утекать будет. Схема ключа абсолютно рабочая, вот только не тестированная на питании контроллера.

Там же еще какие-то RX и TX есть, а есть еще пыль и влажность и еще куча необъяснимых явлений. По сути, если в документации не указанно что микроконтроллер так можно запитать, - значит нельзя. Давным давно пробовал через транзисторный ключ включать ESP, у меня не получилось, ESP зависала в непонятном состоянии. Разбираться почему это происходило не стал, может быть транзистор полностью не открывался, не знаю.
А чем вы собираетесь открыть мосфет, для него обычно 5 вольт нужно на затвор

dontsovcmc, 2020-03-27:
1. Единственное, есп07 точно имеет 512кб памяти
- память только для ОТА нужна. 

2. Пути:
а. Оставить существующую схему питания от стандартного ватериуса и использовать модули нормального качества.
- В топку. Я тоже думал, что "ща мануал почитаю", перестанут есть. А когда через пин выключенной ЕСП течет 3мкА и никто на форуме не отвечает почему... Я не нашел объяснений, почему одни жрут мало, другие много. Это с одинаковой микрухой памяти происходит. Гадание не наш метод.

б. стаб с выходом EN
- Да, самый понятный вариант. Но электронщики могут шапками закидать. Плюс 20-30руб не играют роли. Однако, может быть проблема купить.
Можно даже Attiny без стаба подключать, а стаб только на ЕСП. Но тогда нужно делитель напряжения для i2c делать и вроде все. Надо уточнить, не сильно ли убегает частота attiny от напряжения. 

в. Транзистором N-канальным (как я предлагаю) или Р-канальным (коммутация + питания).
- Способ для настоящих ковбоев. Так в принципе и надо. Осталось сделать /найти схему.
А можете задать вопрос на форуме esp8266.ru ? 

3. А где будет утечка? 
- Я лишь добавлю, что attiny прекрасно работает если остаётся питание на пинах, а снимается с пина +5В. Т.е. внутри может быть что угодно.

4. Еще вопрос, а есп впадает в спячку или тини просто вырубает ее выводом EN? Если впадает в спячку, то в какой режим. Это я про оригинальный Ватериус. И вообще, что мы можем выиграть, вырубая всю есп?
- я вызываю deepsleep и вырубаю EN. deepsleep типа 20мкА, а EN это 0мкА. Как бы не так (. Мы выигрываем независимость от качества пайки и компонентов китайцев. Сейчас ЕСП жрет от 50 до 70% энергии вхолостую и это боооль.

5. А чем вы собираетесь открыть мосфет, для него обычно 5 вольт нужно на затвор
- я скинул примеры мосфетов которые открывают. насколько я помню, существуют от 2.5В, возможно это именно они.

badenbaden, 2020-03-27:
> @badenbaden
>
>
> Там же еще какие-то RX и TX есть, а есть еще пыль и влажность и еще куча необъяснимых явлений. По сути, если в документации не указанно что микроконтроллер так можно запитать, - значит [...]

Пыли там точно не будет. Влажности по идее тоже. К ТХ подключен к светодиод (через который у ватериуса утекает 3 мкА, со слов @dontsovcmc)? но земли у него опять же нет.
Симуляцию в протеусе я выкладывал [тут] (https://github.com/dontsovcmc/waterius/issues/128#issuecomment-604642790)

dontsovcmc, 2020-03-27:
@badenbaden я думаю надо в интернете найти решение задачи, оно должно быть. ESP надежнее отключать целиком, чтобы не связываться с китайскими заморочками ). И лучше, чтобы отключался плюс... )

OloloevReal, 2020-03-29:
@badenbaden 

> У выбранного транзистора - Пороговое Напряжение Vgs 1.75В

У выбранного это какого? NVTR4503?

[image]

Посчитайте симуляцию на эквивалентной нагрузке - ESP потребляет 200-300 мА. Боюсь что может быть существенное падение напряжения, а если питать от 3В то это уже ниже допустимого порога для ESP

badenbaden, 2020-03-29:
@OloloevReal 
Вы абсолютно правы. Будет падение напряжение. Были мысли, что мои симуляции надо бы провести с адекватной нагрузкой, а не просто со светодиодом.
Но я поступил так. Переделал схему включения ESP на P-канальный мосфет. Из той же линейки, что и NVTR4503 (N канальный) есть транзистор NVTR4502 (P канальный). Параметры у них почти одинаковые и он нам также не подходит (хотя при напряжении от 5 вольт он должен работать нормально).
Симуляции хорошо, но испытания на настоящих детальках лучше. Купил в чип-дипе транзистор [Fairchild FDN306P](https://static.chipdip.ru/lib/002/DOC001002372.pdf). Хотел еще [Alpha & Omega AO3401A](https://static.chipdip.ru/lib/244/DOC001244348.pdf), но они их потеряли)).
В общем к испытаниям.
Тестовый стенд - питание 3,3в. (нет 3 вольтового стаба под руками, хотя мог бы и в дипе купить), транзистор FDN306P, Gate (Затвор) подключен к + через 2 последовательно включенных резистора 3,3к, Source (Исток) подключен к +, к Drain (Сток) подключена нагрузка в виде 4 параллельно включеных резисторов 39 ом (дополнительно, для индикации нагрузки, параллельно резисторам включен светодиод с резистором 39 ом), в разрыв плюсового провода от источника питания включен амперметр. Управление транзистором осуществляется путем притяжки Gate к земле через резистор 3,3к.

Фото стенда:
[image]

Потребление схемы при высоком уровне Gate (Затвор), Тестер в режиме измерения мкА:
[image]

Потребление схемы при низком уровне Gate (Затвор), Тестер в режиме измерения А:
[image]

Напряжение на источнике питания (без нагрузки):
[image]

Напряжение на источнике питания (с нагрузкой 0,33А):
[image]

Напряжение на нагрузке (параллельно включенные резисторы):
[image]

Падение напряжение на тестере в режиме Амперметра:
[image]

Если посчитать сумму 2.86 и 0.15 , то получается более 3 вольт. Но напряжение питания 3.3 в.
Также, если нагрузка будет не такой большой, а в подавляющем большинстве случаев так и будет (ЕСП кажется потребляет импульсно), то напряжение просядет еще меньше.
Ну и погрешность моих простейших приборов никто не отменял. Хотел проверить на реальной есп, но дома не оказалось.
Как то так.

badenbaden, 2020-03-29:
Обновленная схема выглядит пока так.

[image]

dontsovcmc, 2020-03-29:
@badenbaden вы в Москве? напишите мне в телеграм

dontsovcmc, 2020-03-29:
@badenbaden ошибка в порту прошивки attiny:  waterleak2 (PA5) нужен пин, а не waterleak1. 
https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/extras/ATtiny_x4.md

badenbaden, 2020-03-29:
> @badenbaden вы в Москве? напишите мне в телеграм

> 
В личку или группу? Да, я в Москве.

badenbaden, 2020-03-29:
> ошибка в порту прошивки attiny: waterleak2 (PA5) нужен пин, а не waterleak1.

> 
Вот это не понял. waterleak1 и waterleak2 это 2 независимых входа для 2х датчиков протечки. Оба порта ADC. Что не так?

dontsovcmc, 2020-03-29:
@badenbaden в личку.
вы вытащили 6 пинов для прошивки тиньки. но в них нет PA5. вместо него PA7.

badenbaden, 2020-03-29:
Теперь понял)) Это я поменял их местами, для лучшей разводки, а в порту забыл.

badenbaden, 2020-03-31:
Продолжение экспериментов.
Т.к. нет в наличии [MCP1700T-3002E](https://static.chipdip.ru/lib/264/DOC005264291.pdf), то решил запитать через DC-DC. Настроил выход на 2,96 вольт (специально чуть ниже требуемых), согласно моих тестеров. В качестве нагрузки подключил ESP8266 модель 12. Прошил прошивку ESPEasy. 

Общий вид (квадратный вольтметр занижает показания на 2-3 сотки, отображает напряжение после dc-dc):
[image]

ЕСП с резисторами, соплями и проводами:
[image]

Стенд по крупнее:
[image]

ЕСП надо бы нагрузить. На сколько смог нагрузил ее разными отправками данных по HTTP и MQTT, синхронизацией времени, опросом датчиков и т.д. В общем как умел)) Ниже кусок лога есп через web интерфейс:
[image]

Тут главная ESPEasy. Видно, что нагрузка 100%, это конечно нагрузка наверняка процессора, но тоже важно.
[image]

При всех моих манипуляциях напряжение питания не изменялось, проверить потребление в любом состоянии я не смог. Мой тестер с падением напряжения на проводах не дал мне этого сделать - показывал 196-198 мА потребления, но есп не стартовала. Вообще заметил, что все эти китайские провода и макетки очень сильно вносят погрешность на таких низких напряжениях. На физической плате из-за пайки и малого размера между элементами такого не будет наблюдаться.

Результаты:
При питании от 3 вольт и использовании транзистора FDN306P в качестве ключа ЕСП работает стабильно. Думаю что питание через MCP1700 не внесет никаких корректив.

badenbaden, 2020-04-01:
Несколько рендеров разведенной платы.

[image]

[image]

[image]

dontsovcmc, 2020-04-01:
@badenbaden вкуснота то какая! =) 
p.s. не вижу габариты: а кнопка и светодиоды напротив дырок под вводы на корпусе?

OloloevReal, 2020-04-01:
@badenbaden на заметку, вдруг пригодится мой опыт
Вполне удачно получилось разместить индикацию и кнопку управления в резиновых заглушках распредкоробки

[image]

badenbaden, 2020-04-01:
габариты платы 73х73.

Кнопка так изначально и планировалась - без прорезания отверстий.
Светодиоды смотрят в верх, через кабельные вводы. И как я и говорил - они сами у этой коробки работают как сальник. Но для верности герметиком можно промазать.
Вот скрины в корпусе.

[image]

[image]

Тут "вскрыл" заднюю часть коробки:
[image]

OloloevReal, 2020-04-01:
@badenbaden а в чем смысл тогда использовать Г образную кнопку?

badenbaden, 2020-04-01:
> а в чем смысл тогда использовать Г образную кнопку?

> 
Как в чем? Плата расположена в другой плоскости относительно кабельного ввода. Потому то и "Г" образная. Можно поставить стандартную, но чтоб ее нажать нужно будет вскрывать крышку.

OloloevReal, 2020-04-01:
@badenbaden 

`Кнопка так изначально и планировалась - без прорезания отверстий.`

А как без отверстия на нее нажимать тогда?

badenbaden, 2020-04-01:
Длина штыря кнопки подобрана так, что он упирается в центр кабельного ввода. Коробка сделана методом двойной экструзии - корпус жесткий, а кабельные вводы из мягкого материала, похожего на резину. Нажатие снаружи вызывает продавливание кабельного ввода, который и нажимает на кнопку. Отверстие не нужно, все герметично.
Вот [видео](https://www.youtube.com/watch?v=Hu_ML8vEduI) об этой коробке.

badenbaden, 2020-04-01:
Можно уже заказать платы и компоненты. Только вот дойдут ли? Везде все перекрывают блин.

dontsovcmc, 2020-04-01:
@badenbaden а сколько потребляет поддержание HIGH на полевом транзисторе? Он ведь в таком виде должен спать, верно?

badenbaden, 2020-04-01:
@dontsovcmc Вот [тут](https://github.com/dontsovcmc/waterius/issues/128#issuecomment-605686931) фото и описание моего первого теста. Мой тестер показывает 00.0 в режиме измерения до 200 мкА. Получается что менее 1 десятой микроампера. 

[image]

badenbaden, 2020-04-05:
Вот к чему мы пришли - [https://github.com/badenbaden/Waterius-Attiny84-ESP12F](https://github.com/badenbaden/Waterius-Attiny84-ESP12F)

На текущий момент платы заказаны. По прибытию начнем тестирование.

SergeiSOficial, 2020-04-06:
MCP1700 выдет максимум 250мА. Для ESP32 рекомендуют источник питания на 500мА. 
Рекомендую сделать фильтры по питанию между +3V и +3V-ESP. [BLM31](https://ru.mouser.com/Search/Refine?Keyword=BLM31) и конденсаторы.
 Т.к. во время включения или работы ESP мы получем очень много шумов.

badenbaden, 2020-04-06:
Не совсем понял, что значит между +3V и +3V-ESP?
В оригинальной версии этот стаб прекрасно тянет есп. И у нас не есп32 а есп8266.

SergeiSOficial, 2020-04-06:
пример:
[image]
Еще часто разделяют аналоговую часть и цифровую. Часто разделают полигоны земли. [Пример](http://radiostorage.net/1026-podsistema-pitaniya-v-mikrokontrollere.html)
[Пример](https://studfile.net/preview/7611260/page:9/)

SergeiSOficial, 2020-04-06:
В пике ESP8266 тоже потребляет в пике до 500мА

badenbaden, 2020-04-06:
@SergeiSOficial Вы все правильно пишите. Только это прототип. потребуются дополнительные элементы-добавим, будут помехи - разделим. Да и не для военной промышленности мы разрабатываем)). А по поводу потребления в 500 мА - что-то более сотни устройств с есп и таким стабом работают очень и очень давно.
Все схемы и разводка в открытом доступе. Конструктивные предложения приветствуются.

dontsovcmc, 2020-04-07:
@SergeiSOficial огромное спасибо за комментарий!
Ого, а чем эти фильтры отличаются от smd дросселей-катушей которые можно часто встретить на питании?

Такой фильтр можно еще на все входы от счетчиков поставить?

MCP1700 держит пики превышающие 250мА. Если не хватит, другой стаб поставим. У тебя есть на примете?
Я вот тут выписываю https://github.com/dontsovcmc/waterius/issues/3#issuecomment-380848998

SergeiSOficial, 2020-04-07:
Эти фильтры настроены таким образом, чтобы гастить ВЧ помехи и не мешать пстоянному току. Там есть разные частотные характеристики. Можно ставить на входы от счетчиков.
Относительно дросселей они похожи. Разница в частотных харакетристиках и в величине индуктивности. Дроссели по питанию часто применяются для выпрямления и стабилизации токов, но для ВЧ помех они бывают слабы.

dontsovcmc, 2020-07-02:
Тема переросла в отдельную модификацию Ватериуса, закрываю. Все обсуждения ее в другом репозитории.
https://github.com/badenbaden/Waterius-Attiny84-ESP12F

closed 2020-07-02 by dontsovcmc

Kabron287, 2021-01-15:
Я не понимаю, что вы так мучаетесь с ключами питания. Есть годами проверенная схема питания в транзисторном тестере Маркуса. 
Работает безупречно.

dontsovcmc, 2021-01-15:
> Я не понимаю, что вы так мучаетесь с ключами питания. Есть годами и тысячами реализаций проверенная схема питания в транзисторном тестере Маркуса.
> Работает безупречно.

А можете привести ссылку? Спасибо!

Kabron287, 2021-01-15:
Классика в общем:
https://vrtp.ru/index.php?s=d7e4167b0d2c188d8f231b518dcebeee&showtopic=25020&st=0&#entry598223

Giperson400, 2021-02-20:
Здравствуйте! Подскажите пожалуйста сей девайс работает как полагается, но кроме датчиков протечки? Появилось желание реализовать сбор данных с 4х счетчиков и поэтому интересуюсь.

badenbaden, 2021-02-20:
Приветствую! Да, все 4 канала работают и передаются данные, но на сайте ватериуса есть поддержка только 2х каналов. Датчики протечки тоже работают, но не так как хотелось бы.

Giperson400, 2021-02-20:
badenbaden, а можно в случае вопросов написать на вашу электронную почту ?

badenbaden, 2021-02-20:
Можно

Giperson400, 2021-02-20:
А какой адрес электронной почты? В профиле у вас видимо скрыт.

[4 more replies]