dontsovcmc/waterius #132: Использование совместно с газовыми счетчиками - перебор или недобор

Fortunatto, 2020-05-06:
Попробовал использовать Waterius с газовым счетчиком (замыкание геркона).
В зависимости от положения геркона в 2х из четырех позиций получил "недосчет" и в 2х - "пересчет". Причём значительно и там и там. Счётчик BK-G6, три разряда ",000" после запятой, магнит закреплен под нулём третьего разряда. 
Задумался почему так происходит, поэкспериментировал со снятием показаний - накопление расхождения не линейно. Предполагаю, что 
а) в ситуации когда геркон стоит далеко от магнита, срабатывает только "медленное" прохождение магнита (т.е. когда работает только одна-две конфорки на плите). Если же потребление значительное - "проскок" не успевает зафиксироваться.
б) в ситуации когда геркон близко к магниту, и расход мал (т.е. магнит проходит в точке срабатывания медленно), возможно, происходят множественные срабатывания, что приводит к завышению показаний.
Если ситуацию (а) программно никак не исправить, то ситуацию (б) можно попробовать скорректировать принудительным заданием минимального интервала срабатывания (например, не чаще 1 раза в секунду. Это нужно настраивать для разных счётчиков индивидуально).
Можно ли "прикрутить" такие настройки?

dontsovcmc, 2020-05-06:
Добрый день! Спасибо за исследование!
У меня вопрос: а есть в описании BK-G6 "вес импульса"? Если он 1 тысячная кубометра, то надо, чтобы в Ватериусе стоял 1л/имп. А по умолчанию в нем 10л/имп. Таким образом может быть ошибка в 10 раз.
1л/имп можно сделать в режиме настройки, 4 раза замкнув вход холодной воды.

А можете включить светодиод через геркон? Можно даже в ватериусе вытащить ESP и подключить геркон к GND и TX пинам - будет включаться светодиод.
И включить медленно газ - будет ли однократно включаться светодиод?

Сейчас алгоритм подсчета импульсов уже очень щадящий: импульс - это замыкание на 250мс И размыкание на 750мс. Импульс будет после размыкания считаться.

Fortunatto, 2020-05-06:
Вес импульса - 0,01м3. Ошибка в моем случае примерно в 1/3 от реальных (в ту или другую сторону).
Со светодиодом поэкспериментирую - как вернусь за город.

dontsovcmc, 2020-09-20:
@Fortunatto новости есть?

Fortunatto, 2020-09-21:
Евгений, привет! Прости, не добрался еще - удалёнка и карантин многие планы нарушили. В будущие выходные проверю степень расхождения.

dontsovcmc, 2021-03-22:
@Fortunatto кто-то подключил уже к газовому счетчику. У вас как успехи?

closed 2021-03-22 by dontsovcmc