dontsovcmc/waterius #133: Заменить синхронизацию времени

dontsovcmc, 2020-05-16:
Не обновить версию фреймворка, т.к. они сломали configTime
https://github.com/esp8266/Arduino/issues/7141#issuecomment-629638092

dontsovcmc, 2020-05-16:
Может на эту либу перейти https://github.com/gmag11/NtpClient

g4m3r-87, 2021-03-02:
Доброго времени суток!
`#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <time.h>

const char *ssid     = "MTSRouter_042190";
const char *password = "23121987";

void setup() 
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  WiFi.begin(ssid, password);

**// конфигтайм выше чем проверка статуса вафай так работает почти всегда**
configTime(0, 0, "pool.ntp.org");

  while ( WiFi.status() != WL_CONNECTED ) {
      delay ( 500 );
      Serial.print ( "." );
  }

**// конфигтайм ниже чем проверка статуса вафай так в 90% случаев не работает**
//configTime(0, 0, "pool.ntp.org");
  

  Serial.println("Waiting for NTP time sync");
  unsigned long start_ = millis();

  time_t now = time(nullptr);

  while (now < 50000 && millis() - start_ < 3000UL) 
  {
    delay(100);
    now = time(nullptr);
  }

  if (millis() - start_ >= 3000UL)
  {
    Serial.print("NTP error");
  } 
  else
  {
    time_t now = time(nullptr);
    struct tm timeinfo;
    gmtime_r(&now, &timeinfo);
    Serial.print("Current time: ");
    Serial.print(asctime(&timeinfo));
  } 
}

void loop() 
{

}`

heX16, 2021-03-03:
> Доброго времени суток!
> ...

Что это?

neitri, 2022-01-22:
Проверил сейчас на версии 3.2.0 синхронизация времени работает.
Часть лога:
000:00:00:03:980  INFO      (WIF ) : mode N
000:00:00:03:984  INFO      (WIF ) : Connected, IP: 192.168.0.112
000:00:00:03:990  DEBUG     (WIF ) : RSSI: -68
000:00:00:03:994  INFO      (BLK ) : SKIP
000:00:00:03:998  INFO      (MQT ) : SKIP
000:00:00:04:002  INFO      (SND ) : -- START -- Send new data
000:00:00:04:008  INFO      (SND ) : JSON size: 371
000:00:00:04:012  INFO      (RQT ) : -- START -- Send JSON POST request
000:00:00:04:018  INFO      (RQT ) : URL:       https://cloud.waterius.ru
000:00:00:04:024  INFO      (RQT ) : Body:      {"delta0":0,"delta1":0,"good":1,"boot":1,"ch0":321,"ch1":654,"imp0":3,"imp1":5,"version":22,"voltage":3.476,"version_esp":"0.10.5","key":"AAA7FE7D1ABAD041F6197751B43A9446","resets":2,"email":"***","voltage_low":false,"voltage_diff":11,"f0":1,"f1":1,"rssi":-68,"waketime":10083,"setuptime":496505,"adc0":0,"adc1":0,"period_min":1440,"serial0":"","serial1":""}
000:00:00:04:272  INFO      (NTP ) : Waiting for NTP time sync:
000:00:00:04:373  INFO      (NTP ) : Current time: Sat Jan 22 12:26:36 2022

000:00:00:04:374  INFO      (RQT ) : Begin client
000:00:00:05:677  INFO      (RQT ) : Response code:     200
000:00:00:05:678  INFO      (RQT ) : Response body:
000:00:00:05:678  INFO      (RQT ) : -- END --
000:00:00:05:680  INFO      (SND ) : Send HTTP code:    200
000:00:00:05:685  INFO      (SND ) : -- END --
000:00:00:05:764  INFO      (CFG ) : Config stored OK
000:00:00:05:765  INFO      (ESP ) : Going to sleep

dontsovcmc, 2022-02-21:
Да, работает. Закрываем задачу.

closed 2022-02-21 by dontsovcmc