dontsovcmc/waterius #139: Чтение напряжение питания с батареек аналоговым входом ESP

OloloevReal, 2020-06-26:
Добрый день!

Для ESP32 замер напряжения с элементов питания (4xAA или 1x18650) делал следующим образом:

[image]

Где:
VBAT  - выход элементов питания
GPIO-13 - Выход ESP разрешающий замер
GPIO-27 - Аналоговый вход ESP

Т.к. напряжение снимается с делителя 50/50, то АЦП считывает значение эквивалентное половине VBAT

[image]

Цель всей этой схемотехники избежать постоянной утечки на землю, и делать замер только по запросу.
Необходимы конечно свободные IO, с ESP32 с этим все в порядке, а вот ESP8266 дефицит
Может пригодится

dontsovcmc, 2020-06-26:
Мне Иван рисовал похожую схему:
[image]

Дмитрий ее реализовал на большой плате:
[image]

На текущий ватериус не ясно, куда ее запихнуть =( А вот на ESP-12F влезает.

OloloevReal, 2020-06-27:
Так-то тоже самое, только на полевиках. С ESP01, если правильно помню, только один ADC и тот не распаян, вариант только подпаиваться к чипу.  И попасть в диапазон от 0 до 1.0 V придется

closed 2021-03-22 by dontsovcmc