dontsovcmc/waterius #141: Разделить топики в MQTT

HYBRID-BEING, 2020-08-05:
Для того, чтобы было удобно подписываться на нужную информацию, предлагаю отделить идентификаторы от ch, delta и imp:
waterius/6901727/ch/0
waterius/6901727/ch/1
waterius/6901727/delta/0
waterius/6901727/delta/1
waterius/6901727/imp/0
waterius/6901727/imp/1
Таким образом можно будет сделать подписку, например, только на каналы нескольких ватериусов, без лишней сторонней информации:
waterius/+/ch/+

dontsovcmc, 2020-08-10:
Соглашусь, но кажется это надо было делать изначально.. Сейчас или ломать совместимость или дублировать топики. А надо ли оно, если речь о 3-х топиках?
Плюс я не понимаю практической пользы - всем надо отображать данные ,а значит все равно все топики писать вручную.
Где вам понадобилось waterius/+/ch/+ ?

HYBRID-BEING, 2020-08-14:
Ну, пожалуй, действительно, ломать совместимость из-за мизерного каприза смысла нет.

closed 2020-08-14 by HYBRID-BEING