dontsovcmc/waterius #142: Поддержка m-bus

Stariy, 2020-08-21:
Неплохо бы прикрутить поддержку протокола m-bus, для снятия показаний со счётчиков тепла.

khk0088, 2022-05-23:
did 

Hi Stariy! 
did you found any thing related to M-Bus in Russian?

khk0088, 2022-11-17:
> I know only one C open source library for M-Bus. What's your goal?

Hi Evgeny.

I did not succeed in Russian. There is not so much information in Russian.
Most of the information I found was only in English weather in German.

My goal was to integrate protocol to our device and make a sensor to be
able to read from the Mbus port of the different meters.
Now I have succeeded  with this task. If you need some help let me know.
If you want we can continue our conversation in Russian too.

Kind regards
Khusan.

dontsovcmc, 2025-06-08:
> ocol to our device and make a sensor to be able to read from the Mbus po

Hi, Khusan!
Have you achieved your goal? Sorry for delay =(. My friend is making m-bus device now. I can connect you with him.

khk0088, 2025-06-09:
Hi Evgeny, yes I did have achieved my goals but it would be good to have
talked with you friend about his MBus development projects.
By the way I do speak Russian too in case your friend or you prefer to
speak Russian. I am open to using Russian instead of English.