dontsovcmc/waterius #144: Добавить настройку периода отправки

dontsovcmc, 2020-08-31:
В вебстранице пусть будет настройка "период передачи данных" в секундах или минутах.
ЕСП передаст в attiny значение после настройки.

obrianw, 2020-11-26:
Нужный параметр, особенно когда есть Умный дом. 👍

dontsovcmc, 2020-12-03:
> Нужный параметр, особенно когда есть Умный дом. 👍
> Понятно, что при увеличении передачи данных (будим ESP8266, wifi, согласование) потребуется сменить на более мощные аккумуляторы или ставить БП.

У меня проработал около 2-х лет при передачи каждые 30 сек.

obrianw, 2020-12-04:
> > Нужный параметр, особенно когда есть Умный дом. +1
> > Понятно, что при увеличении передачи данных (будим ESP8266, wifi, согласование) потребуется сменить на более мощные аккумуляторы или ставить БП.
> 
> У меня проработал около 2-х лет при передачи каждые 30 сек.

Не ожиданно! Это же меняет всё (!) Так часто это конечно сильно, но для проверки начальных показаний самое-то!

И значит не зря, что затеял собрать себе таких три штуки, и даже по Вашему шаблону из гайда таки заказал с десяток плат (3-ей версии :) ) на PCBWay (использовав скидку на первый заказ) оплатив только доставку и комиссию за оплату (1$ комиссии с $5 это конечно сильно, но что-ж возьмешь с банков, ещё те крахоборы) и опять же по Вашему списку так же заказал недостающие компоненты, коих нет под рукой... 
Огромное спасибо за такой подробный и открытый проект!

dontsovcmc, 2021-01-27:
Заголовок поля: "Как часто обновлять данные о показаниях, мин"

dontsovcmc, 2021-02-25:
Можно задать период выхода на связь в минутах. По умолчанию 1 сутки (1440 мин). Период хранится в памяти ESP. При каждом пробуждении она передает в attiny это значение.

closed 2021-02-25 by dontsovcmc