dontsovcmc/waterius #145: Не компилируется прошивка

NOV62, 2020-10-04:
Добрый день! Пытаюсь собрать прошивку для ESP8266 в Arduino IDE, не выходит, компилятор ругается на  Дополнение к WifiManager.

In file included from sketch\setup_ap.h:8:0,
                 from sketch\setup_ap.cpp:2:
D:\����� �����\Arduino\libraries\WiFiManager/WiFiManager.h: In constructor 'LongParameter::LongParameter(const char*, const char*, long int, uint8_t)':
D:\����� �����\Arduino\libraries\WiFiManager/WiFiManager.h:65:10: error: 'void WiFiManagerParameter::init(const char*, const char*, const char*, int, const char*)' is private
     void init(const char *id, const char *placeholder, const char *defaultValue, int length, const char *custom);
          ^
In file included from sketch\setup_ap.cpp:2:0:
setup_ap.h:25:9: error: within this context
         init(id, placeholder, String(value).c_str(), length, " type=\"number\"", WFM_LABEL_BEFORE);
         ^
setup_ap.h:25:82: error: 'WFM_LABEL_BEFORE' was not declared in this scope
         init(id, placeholder, String(value).c_str(), length, " type=\"number\"", WFM_LABEL_BEFORE);
                                                                                  ^
In file included from sketch\setup_ap.h:8:0,
                 from sketch\setup_ap.cpp:2:
D:\����� �����\Arduino\libraries\WiFiManager/WiFiManager.h: In constructor 'FloatParameter::FloatParameter(const char*, const char*, float, uint8_t)':
D:\����� �����\Arduino\libraries\WiFiManager/WiFiManager.h:65:10: error: 'void WiFiManagerParameter::init(const char*, const char*, const char*, int, const char*)' is private
     void init(const char *id, const char *placeholder, const char *defaultValue, int length, const char *custom);
          ^
In file included from sketch\setup_ap.cpp:2:0:
setup_ap.h:39:9: error: within this context
         init(id, placeholder, String(value,3).c_str(), length, " type=\"number\" step=\"0.001\" placeholder=\"0,000\"", WFM_LABEL_BEFORE);
         ^
setup_ap.h:39:121: error: 'WFM_LABEL_BEFORE' was not declared in this scope
         init(id, placeholder, String(value,3).c_str(), length, " type=\"number\" step=\"0.001\" placeholder=\"0,000\"", WFM_LABEL_BEFORE);
                                                                                                                         ^
In file included from sketch\setup_ap.h:8:0,
                 from sketch\setup_ap.cpp:2:
D:\����� �����\Arduino\libraries\WiFiManager/WiFiManager.h: In constructor 'IPAddressParameter::IPAddressParameter(const char*, const char*, IPAddress)':
D:\����� �����\Arduino\libraries\WiFiManager/WiFiManager.h:65:10: error: 'void WiFiManagerParameter::init(const char*, const char*, const char*, int, const char*)' is private
     void init(const char *id, const char *placeholder, const char *defaultValue, int length, const char *custom);
          ^
In file included from sketch\setup_ap.cpp:2:0:
setup_ap.h:58:9: error: within this context
         init(id, placeholder, val.c_str(), 16, "minlength=\"7\" maxlength=\"15\" size=\"15\" pattern=\"^((\\d{1,2}|1\\d\\d|2[0-4]\\d|25[0-5])\\.){3}(\\d{1,2}|1\\d\\d|2[0-4]\\d|25[0-5])$\"", WFM_LABEL_BEFORE);
         ^
setup_ap.h:58:191: error: 'WFM_LABEL_BEFORE' was not declared in this scope
         init(id, placeholder, val.c_str(), 16, "minlength=\"7\" maxlength=\"15\" size=\"15\" pattern=\"^((\\d{1,2}|1\\d\\d|2[0-4]\\d|25[0-5])\\.){3}(\\d{1,2}|1\\d\\d|2[0-4]\\d|25[0-5])$\"", WFM_LABEL_BEFORE);
                                                                                                                                                                                               ^
In file included from sketch\setup_ap.h:8:0,
                 from sketch\setup_ap.cpp:2:
D:\����� �����\Arduino\libraries\WiFiManager/WiFiManager.h: In function 'void handleStates()':
D:\����� �����\Arduino\libraries\WiFiManager/WiFiManager.h:123:39: error: 'std::unique_ptr<esp8266webserver::ESP8266WebServerTemplate<WiFiServer> > WiFiManager::server' is private
     std::unique_ptr<ESP8266WebServer> server;
                                       ^
setup_ap.cpp:100:6: error: within this context
   wm.server->send(200, F("text/plain"), message);
      ^
sketch\setup_ap.cpp: In function 'void handleNetworks()':
setup_ap.cpp:107:6: error: 'class WiFiManager' has no member named 'WiFi_scanNetworks'
   wm.WiFi_scanNetworks(wm.server->hasArg(F("refresh")),false); //wifiscan, force if arg refresh
      ^
In file included from sketch\setup_ap.h:8:0,
                 from sketch\setup_ap.cpp:2:
D:\����� �����\Arduino\libraries\WiFiManager/WiFiManager.h:123:39: error: 'std::unique_ptr<esp8266webserver::ESP8266WebServerTemplate<WiFiServer> > WiFiManager::server' is private
     std::unique_ptr<ESP8266WebServer> server;
                                       ^
setup_ap.cpp:107:27: error: within this context
   wm.WiFi_scanNetworks(wm.server->hasArg(F("refresh")),false); //wifiscan, force if arg refresh
                           ^
setup_ap.cpp:108:6: error: 'class WiFiManager' has no member named 'getScanItemOut'
   wm.getScanItemOut(message);  
      ^
In file included from sketch\setup_ap.h:8:0,
                 from sketch\setup_ap.cpp:2:
D:\����� �����\Arduino\libraries\WiFiManager/WiFiManager.h:123:39: error: 'std::unique_ptr<esp8266webserver::ESP8266WebServerTemplate<WiFiServer> > WiFiManager::server' is private
     std::unique_ptr<ESP8266WebServer> server;
                                       ^
setup_ap.cpp:109:6: error: within this context
   wm.server->send(200, F("text/plain"), message);
      ^
In file included from sketch\setup_ap.h:8:0,
                 from sketch\setup_ap.cpp:2:
D:\����� �����\Arduino\libraries\WiFiManager/WiFiManager.h: In function 'void bindServerCallback()':
D:\����� �����\Arduino\libraries\WiFiManager/WiFiManager.h:123:39: error: 'std::unique_ptr<esp8266webserver::ESP8266WebServerTemplate<WiFiServer> > WiFiManager::server' is private
     std::unique_ptr<ESP8266WebServer> server;
                                       ^
setup_ap.cpp:113:6: error: within this context
   wm.server->on(F("/states"), handleStates);
      ^
In file included from sketch\setup_ap.h:8:0,
                 from sketch\setup_ap.cpp:2:
D:\����� �����\Arduino\libraries\WiFiManager/WiFiManager.h:123:39: error: 'std::unique_ptr<esp8266webserver::ESP8266WebServerTemplate<WiFiServer> > WiFiManager::server' is private
     std::unique_ptr<ESP8266WebServer> server;
                                       ^
setup_ap.cpp:114:6: error: within this context
   wm.server->on(F("/networks"), handleNetworks);
      ^
sketch\setup_ap.cpp: In function 'void setup_ap(Settings&, const SlaveData&, const CalculatedData&)':
setup_ap.cpp:120:8: error: 'class WiFiManager' has no member named 'debugPlatformInfo'
     wm.debugPlatformInfo();
        ^
setup_ap.cpp:121:8: error: 'class WiFiManager' has no member named 'setWebServerCallback'
     wm.setWebServerCallback(bindServerCallback);
        ^
Что надо подправить? 
Спасибо

dontsovcmc, 2020-10-04:
Все верно. Дело в том, что нужна моя ветка Wifimanager:   https://github.com/dontsovcmc/WiFiManager.git#waterius_release_100
ее в архиве и подкладывать в Arduino IDE

NOV62, 2020-10-04:
Сделал. Не помогло.

И Platformio не хочет компилировать, ругается на саму себя. Хотя другие проекты в той же папке собираются без проблем. Скрин прилагаю.
[image]

Прошу прощения за назойливость. Я начинающий, хочу разобраться, как у Вас реализовано открытие окна доп. настроек WiFiManager. Как я понял, при отметке чекбокса вызывается функция advSett(), которая должна быть в скрипте, но её нет! И в исходниках advSett нигде не находится. 
Непонятно...

NOV62, 2020-10-04:
В функции handleNetworks() вкладки setap_ap.cpp вызываются wm.WiFi_scanNetworks и wm.getScanItemOut, которых тоже нет в библиотеке, по Вашей ссылке тоже.

OloloevReal, 2020-10-04:
advSett() как раз в форке  WifiManager и реализован, в файле strings_en.h смотрите

dontsovcmc, 2020-10-04:
> Сделал. Не помогло.
> 
> И Platformio не хочет компилировать, ругается на саму себя. Хотя другие проекты в той же папке собираются без проблем. Скрин прилагаю.

в Platformio ничего не надо менять , там все должно собраться сразу... 
а какая версия у вас? 

да, чекбокс на JS сделан

NOV62, 2020-10-04:
Спасибо, теперь понятно. Просто я не оттуда скачивал.
Platformio 1.49.3, не собирается, скрин выше. Остальные проекты компилируются нормально, в том числе импортированные из Ардуино, специально попробовал ещё раз. Ничего не понимаю...

NOV62, 2020-10-04:
Вот строчка, на которой стопится:

UnicodeDecodeError: 'charmap' codec can't decode byte 0x98 in position 365: character maps to <undefined>
The terminal process "C:\Users\O1962\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

dontsovcmc, 2020-10-04:
А попробуйте удалить все русские символы из platformio.ini

NOV62, 2020-10-05:
Ещё раз Спасибо за помощь! Разобрался. Из-за бардака в ноуте запутался в версиях WM.
Трудно осваивать что-то новое в 58 лет, до последнего времени программированию учился в институте в 1985г и с тех пор не имел отношения к этой теме вообще. А сейчас затянуло, очень интересно, но мозги ворочаются с трудом.
У Вас замечательный проект, хочу к весне на его базе сделать автономные датчики влажности почвы на дачу для управления системой полива и переделать веб-морды уже работающих устройств - датчики температуры, RF-MQTT-HTTP мост с контроллером охранной сигнализации, контроллер управления UPS  для насосной станции, чтобы при отсутствии сети он включался только на время работы насоса (пока недоделано).
p.s. А platformio так и дурит - теперь, после удаления кирилицы и установки платформы ESP 2.4.0 вместо 2.6.2 компиляция начинается, молотит минуты полторы, потом просто вылетает с сообщением:

collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp01_1m\firmware.elf] Error 1.

Для меня это слишком сложно...

NOV62, 2020-10-05:
Всё, разобрался. Лишние папки надо выкидывать из рабочих директорий. Я ардуиновскую версию там оставил, из-за неё и глючило. Ещё раз спасибо всем, кто откликнулся. И простите за глупые вопросы.

closed 2020-10-05 by NOV62

dontsovcmc, 2020-10-05:
@NOV62 Отлично! Заходите в чат https://t.me/waterius_forum
Arduino ide и platformio разные продукты. Абсолютное большинство юзеров не знают, что у них под капотом. Поэтому ваши ошибки не являются простыми.
У Platformio есть свой менеджер библиотек, тянуть arduino ide не нужно.
Посмотрите в сторону ESPHome и подобных конструкторов, вдруг там уже есть решение из коробки.
Веб морда это тяжелая тема. Аналогов Wifimanager не знаю, а он не очень хорошо сделан.