dontsovcmc/waterius #146: Электричество

paveleremin, 2020-10-13:
Можно ли прибор использовать с импульсным счетчиком электроэнергии?

GregoryGost, 2020-10-13:
> Можно ли прибор использовать с импульсным счетчиком электроэнергии?

Для начала нет. т.к. в электросчетчиках, как правило, используется выход типа "Открытый коллектор" и частота следования импульсов очень большая. Соответственно для учета всех импульсов необходимо использовать прерывания, а это ведет к тому, что постоянно включен ADC и батарейка/аккумулятор будут высаживаться гораздо, гораздо быстрее, чем это заложено в waterius

Далее вам нужно понимать, что это будет счетчик, к которому у вас есть полноценный доступ, а не опломбированный электроснабжающей организацией.
Даже если вам удастся "договориться", чтобы вывели два провода из электросчетчика - это не доступное применение рядовым пользователем и скорее является исключением, чем правилом.

Резюмируя все вышесказанное, это должно быть устройство работающее от сети 220, с аккумулятором на случай внезапного отключения электричества, учитывающее импульсы по прерываниям, подключенное к дополнительному счетчику электричества, который придется ставить следом, после основного(опломбированного)

Есть еще вариант, который я планирую испытать - это счет потребеления на базе светодиодного индикатора электросчетчика. Но пока это все опыты.

heX16, 2020-12-20:
Можно использовать опломбированный счетчик. Надо просто попросить компанию прислать представителя для повторной опломбировки (возможно они придут бесплатно). Если будут спрашивать зачем то можно сказать что меняете проводку (это будет проще чем объяснять про умный дом)). При повторной опломбировке никаких вопросов насчет лишних проводов не будет (им пофиг) - мне дважды пломбировали счетчик и ни разу не спросили про лишний провод.

GregoryGost, 2020-12-20:
> Можно использовать опломбированный счетчик. Надо просто попросить компанию прислать представителя для повторной опломбировки (возможно они придут бесплатно). Если будут спрашивать зачем то можно [...]

Вы имеете в виду, оборвать пломбу, подсоединить необходимые кабеля и потом вызвать специалиста на повторную пломбировку?

Получается что, вся надежда на то, что пришедший специалист просто придет, поставит новую пломбу и уйдет не задавая лишних вопросов. А если и спросит, то типа "меняли проводку... а это что за провода я хз".

Честно, такое себе предложение. Ваш случай полагаю не показательный. Тут необходима выборка из ситуаций, где это действительно будет работать и причем в различных городах и различных сбытовых компаниях. Тогда можно говорить о применимости для других.

Хорошо бы, чтобы кто-то, кто ещё, кто проделывал, что-то подобное тут отписался.
А так - это также не массовое решение задачи.

heX16, 2020-12-21:
> Вы имеете в виду, оборвать пломбу, подсоединить необходимые кабеля и потом вызвать специалиста на повторную пломбировку?
Нужно подождать когда придет представитель, при нем снять пломбу, установить импульсный ввод и дальше пусть пломбирует повторно.

Электросчетчик это ваша собственность, завод производитель оборудовал устройство штатным импульсным выходом, и вы имеете полное право его использовать. Если у представителя компании возникнут вопросы насчет провода, то ему достаточно показать руководство по эксплуатации элеронным счетчиком где написано про этот разьем.

dontsovcmc, 2020-12-21:
@heX16 в Мосэнергосбыте запрет на подключение не сертифицированного устройства к счетчикам мне объяснили тем, что, к примеру, у счетчиков Меркурий стандартный пароль администратора и изменить его нельзя. По паролю можно поменять коэффициенты, отвечающие за подсчет электроэнергии. 
Я не проверил достоверность этого.

GregoryGost, 2020-12-21:
> Нужно подождать когда придет представитель, при нем снять пломбу, установить импульсный ввод и дальше пусть пломбирует повторно.
А вот тут уже есть шанс того, что специалист может понять, что вы не просто "проводку меняете"
Я все о том же самом, о том, что это не массовое решение вопроса и создавать отдельное устройство или адаптировать текущее имеет смысл только в частных случаях. На сколько я понял Waterius нацелен именно на применение "обычными" людьми. Т.к. в большей массе у счетчиков воды провода доступны (если нет уже установленной системы) и никто н ограничивает им доступ пломбированием.
> 
> Электросчетчик это ваша собственность, завод производитель оборудовал устройство штатным импульсным выходом, и вы имеете полное право его использовать. Если у представителя компании возникнут вопросы насчет провода, то ему достаточно показать руководство по эксплуатации элеронным счетчиком где написано про этот разъем. Сбытовая компания не имеет права препятствовать использовать возможности вашего устройства (при условии что это не нарушает поверку устройства).
Кстати тут с 1 июля 2020 произошли изменения. Теперь пользователи не будут платить за установку электросчетчиков и их замену. Соответственно тема "ваша собственность" тут работать уже не будет.
Опять же электросчетчики в многоэтажках (и даже в пятиэтажках) располагаются в основном подъездах и не являются "вашей собственностью". Как быть с ними? Какой процент приборов без доступа пользователя в стране, а какой с доступом? Насколько велика разница в сравнении с приборами учета воды? Вопросы, вопросы, вопросы... У меня у самого счетчик стоит прямо в квартире, а вот у родителей уже все в подъезде.
Если интересно, то можете почитать статьи на тему **Постановления Правительства РФ от 18.04.2020 N 554 "О внесении изменений в некоторые акты Правительства Российской Федерации по вопросам совершенствования организации учета электрической энергии"** Там даже учли "умный" учет.

Поэтому еще раз повторюсь. Делать версию ватериуса под электричество или вообще какой-нибудь "электриус" :) дело Евгения или кого-нибудь из сообщества. Опять же посмотрите на рынок - это в основном промышленные устройства не для рядовых пользователей и за совершенно другие суммы.

> @heX16 в Мосэнергосбыте запрет на подключение не сертифицированного устройства к счетчикам мне объяснили тем, что, к примеру, у счетчиков Меркурий стандартный пароль администратора и изменить его [...]
Пароли в Меркуриях например меняются. Мосэнергосбыт, как и другие компании, тупо не хотят тратить на это время т.к. они и так все счетчики пломбируют, а конструкция электросчетчиков подразумевает отсутствие доступа рядового пользователя к разъемам с RS-485 (а заодно и импульсным, потому что, как правило на одной колодке)
Соответственно "политика партии" такая: "А нафига нам менять пароли если и так ни у кого не будет доступа..."
Отсюда имеем запрет на якобы "подключение не сертифицированного устройства к счетчикам".
Хотя они могли бы спокойно закрывать Админ доступ и оставлять User у которого права только на чтение.

Ну и мы уходим от изначального вопроса данного ишуса :)
Текущий ватериус надо дорабатывать для возможности подключения к импульсным выходам электросчетчика.

heX16, 2020-12-22:
> @heX16 в Мосэнергосбыте запрет на подключение не сертифицированного устройства к счетчикам мне объяснили тем, что, к примеру, у счетчиков Меркурий стандартный пароль администратора и изменить его [...]
Они говорили про подключение к RS485, и они наврали - менять пароль можно, просто у них нет специалистов для этого (вообще с трудом представляю классическую тетку которая вдруг достает ноут, цепляется через переходник и выставляет права доступа и пароли))).

А обсуждаемый здесь импульсный вывод это другая технология, и ее использовать можно.

dontsovcmc, 2020-12-22:
> А обсуждаемый здесь импульсный вывод это другая технология, и ее использовать можно.

Соглашусь, что импульсный вход "безопасен", но много ли сейчас электросчетчиков с 1м тарифом? выгоднее же 2-3 тарифа.

heX16, 2020-12-22:
> Для начала нет. т.к. в электросчетчиках, как правило, используется выход типа "Открытый коллектор" и частота следования импульсов очень большая. Соответственно для учета всех импульсов необходимо [...]

"частота следования импульсов очень большая" - это какая? Насколько я знаю частота идентична миганию светодиода, то есть в рамках 0,1-10 Герц. То есть наоборот - частота очень низкая.

"необходимо использовать прерывания, а это ведет к тому, что постоянно включен ADC" - Analog Input никак не связан с прерываниями. Если использовать прерывания (и пробуждение по прерыванию) то ADC включать не нужно, потребление будет минимально и никак не отразится на батарейке. Как я понял из статьи на хабре WiFi передача дает основной расход энергии.

heX16, 2020-12-22:
> Соглашусь, что импульсный вход "безопасен", но много ли сейчас электросчетчиков с 1м тарифом? выгоднее же 2-3 тарифа.

Да, есть небольшая сложность - нужно будет синхронизировать время и настраивать диапазоны времени тарифов (чтобы гаджет знал когда какой тариф считать).

GregoryGost, 2020-12-22:
Давно не работал по этой теме. Посмотрим поближе.

> "частота следования импульсов очень большая" - это какая? Насколько я знаю частота идентична миганию светодиода, то есть в рамках 0,1-10 Герц. То есть наоборот - частота очень низкая.

0,1 Герц = 10 сек.(период)
10 Герц = 100 мс.(период)

Как пример:
https://www.incotexcom.ru/files/em/docs/merkuriy-230-avlg-411152-021-re-izm-3-2018-04-01.pdf
Страница 6, таблица 2

Вот такие однотарифные ставит Мосэнергосбыт
https://www.meters.taipit.ru/upload/pasporta/ПС_рев.2_НЕВА_10_[6132517]_19х34_2019.pdf
Количество импульсов, соответствующих одному киловатт-часу, постоянная счетчика, указано рядом со светодиодным индикатором. У меня написано 3 200 imp/kw*h
Минимальная длительность импульса, формируемого на испытательном выходе, — **не менее 30 мс**
Т.е. в любом случае, чтобы не пропускать необходимо ставить не менее 30 мс.

30 мс(период) = ~33.33 Hz

Счетчики делаются очень по разному. В каких-то на 1 kw*h приходится 1 000 импульсов, в каких-то 3 200, в каких-то все 6 400 и т.д. в зависимости от производителя, модели, серии и прочее, прочее, прочее.

В 2020 году социальная норма потребления электроэнергии 300 кВт-ч в месяц на точку подключения. На практике не более 225.
Возьмем ту же Неву для расчета
225 на 30 дней = 7,5 кВт-ч в сутки = 3 200 * 7.5 = 24 000 импульсов в сутки при социальной норме потребления.
Предположим, что сферическая квартира в вакууме потребляет ровным графиком без скачков и просадок.
Отсюда 24 000 / 24 = 1 000 имп/час = ~16-17 имп/мин. Т.е. получаем где-то каждые 3,75 секунды по импульсу.

А есть счетчики с 6 400 imp/kw*h
http://permelectric.ru/attachments/Image/shkala-schyotchika_1.jpg?template=generic
Что тут? 6 400 * 7,5 = 48 000 импульсов в сутки
48 000 / 24 = 2 000 имп/час = ~33 имп/мин = каждые 1,8 секунды.
Ну и так далее.

Далее длительность импульса не менее 30 мс. Мы ловим сигнал с прерывания и проваливаемся в обработчик. Будет ли достаточно для максимального потребления во время пиковой нагрузки (включили духовой шкаф, стиральную машину, микроволновку, чайник, пылесос, др.)?
Понятно, что приборы с высокой нагрузкой лучше одновременно не включать, но если проводка рассчитана, то ничто не мешает включать несколько таких устройств одновременно. Условия же могут быть совершенно различные.
Предположим включили что-то и суммарная нагрузка подскочила.
В такие моменты кол-во импульсов может резко возрасти, предположим в два или три раза:
16 * 2 = 32 = каждые 1,87 секунды / 33 * 2 = 66 = каждые 0,9 секунды
16 * 3 = 48 = каждые 1,25 секунды / 33 * 3 = 99 = каждые 0,6 секунды

Для худшего случая(а мы обязаны учесть самый худший вариант) получаем меандр со скважностью 20 (600 мс/30 мс)

Теперь смотрим комментарии к прошивке Waterius
// настраиваем период сна и кол-во просыпаний за 1 минуту
// Итак, пробуждаемся (проверяем входы) каждые 250 мс
// 1 минута примерно равна 240 пробуждениям
И еще есть
//32 ms  Минута будет в 8 раз чаще
Но какой смысл в режиме сна если необходимо просыпаться каждые 30 мс или лучше даже каждые 28 мс?

Вообще можно было бы остановиться на длительности в 30 мс и все, но я хотел освежить знания.

> "необходимо использовать прерывания, а это ведет к тому, что постоянно включен ADC" - Analog Input никак не связан с прерываниями. Если использовать прерывания (и пробуждение по прерыванию) то ADC [...]

Верно. ADC тут уже не при делах. Но! Микроконтроллер должен быть постоянно включен т.к. у Attiny нет таймеров работающих в режиме сна. **Соответственно определение "потребление будет минимально" не подходит и "никак не отразится на батарейке" прям вообще не про этот режим.**
Посчитайте, за какое время разрядится гипотетический Ватериус со всеми озвученными выше данными.

По итогу получаем МК в Run режиме ждущий импульс на входе(входах) и периодически включающийся ESP(Wi-Fi).
Батарея по моему мнению тут будет нужна только, как резервный источник питания и тот можно заменить внешним БП на DIN рейку.

Кстати, как думаете, текущую схемотехнику входов Waterius нужно менять при подключении к выходу "ОК" электросчетчика?

heX16, 2020-12-25:
@GregoryGost , вау! Спасибо за столь подробный разбор про импульсный выход - это реально очень полезная информация. Я признаться не изучал документацию по этому вопросу - в проектах АСУТП мы просто читаем всю информацию из Меркурий-200 по RS485 и никогда не заморачивались с импульсными входами.

PS Насчет режимов сна в ATTiny85 я чуть позже отвечу более развернуто.

heX16, 2020-12-25:
Я много работал и экспериментировал с линейкой ATTiny.
Далее используются ссылки на Datasheet ["ATtiny25/45/85 - Complete Datasheet"](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-2586-AVR-8-bit-Microcontroller-ATtiny25-ATtiny45-ATtiny85_Datasheet.pdf) (ревизия Rev.258Q-AVR-08/2013).

ATTiny имеет "Watchdog Timer with Separate On-chip Oscillato" может делать пробуждение (именно пробуждение, а не перезагрузку) из режима "Power-down" (из самого глубокого сна). См пункт 7.1.3 и пункт 8.5.2 (таблица 8-2). Низкочастотный осциллятор Watchdog-га продолжает работать даже в глубоком сне (в отличие от всех остальных осцилляторов).

Также ATTiny может пробуждаться из "Power-down" по внешнему прерыванию INT0. См пункт 7.1.3.

Время пробуждения из "Power-down" может быть достаточно быстрым - от нескольких микросекунд (это зависит от выбранной конфигурации SUT Fuses). См пункт 6.2.1 таблица 6-3.

Также ATTiny имеет интересный режим - аппаратный подсчет импульсов без участия CPU. Я экспериментировал с этим режимом и добивался стабильного подсчета импульсов на частоте 4 МГц (на 8 МГц начинаются пропуски). В этом режиме CPU не используется и может находится в спячке, к сожалению этот режим не доступен в "Power-down" (см пункт 9.2.2 рисунок 9-1).
Но этот режим доступен в обычном сне (режим "Idle") - в этом режиме счетчик будет считать импульсы не пробуждая CPU.
При этом в случае переполнения счетчика будет вызвано прерывание TIMER0_OVF которое разбудит CPU - см пункт 7.1.1.

Осциллятор ADC (CLK_ADC) можно выключить отдельно чтобы он не потреблял энергию - см пункт 7.5.2, бит PRADC. И еще отключить ACD - см пункт 16.2.2, бит ACD.

Таким образом для энерго-эффективного подсчета дискретных импульсов есть 2 варианта:

1 Вариант
------------
Подходит для редких импульсов.
Режим "Power-down", пробуждение от каждого импульса. Плюс периодическое пробуждение от Watchdog для сервисных функций (можно настроить раз в 64 секунды).

2 Вариант
-------------
Подходит для частых импульсов.
Режим "Idle" с аппаратным подсчетом импульсов, пробуждение от переполнения счетчика. Плюс периодическое пробуждение от Watchdog (идентично предыдущему варианту).

Варианты можно выбирать и переключать на лету для достижения максимальной эффективности.

Значительных аппаратных доработок не требуется, может изменить схему подтяжки дискретного сигнала (не изучал текущую схему).

dontsovcmc, 2021-01-20:
Для считывания импульсов светодиода мне кажется правильнее использовать хардварный счетчик импульсов. Но нужно нарисовать схему энергоэффективную.

svpcom, 2021-08-31:
Для личного учета можно поставить https://shelly.cloud/products/shelly-3em-smart-home-automation-energy-meter/ после основного счетчика. Места в щите занимает как один автома (без учета токовых трансформаторов). Внутри тот-же esp8266 и при желании можно для него сделать открытую прошивку. Но и в базовой версии он отлично по mqtt отдает данные

AstashovAndrey, 2021-08-31:
> Для личного учета можно поставить https://shelly.cloud/products/shelly-3em-smart-home-automation-energy-meter/ после основного счетчика. Места в щите занимает как один автомат (без учета токовых трансформаторов). Внутри тот-же esp8266 и при желании можно для него сделать открытую прошивку. Но и в базовой версии он отлично по mqtt отдает данные

дороговато. есп + пзем вроде неплохо работает. Да, он на 2din места. Но стоит как шапка сухарей :)

svpcom, 2021-08-31:
> дороговато. есп + пзем вроде неплохо работает. Да, он на 2din места. Но стоит как шапка сухарей :)

У них есть однофазная версия https://shelly.cloud/products/shelly-em-smart-home-automation-device/ Без без спец микросхемы интегрировать силу тока и напряжение на esp не получится (нужен точный и быстрый АЦП). У делал токовое реле на stm32f103

AstashovAndrey, 2021-08-31:
> У них есть однофазная версия https://shelly.cloud/products/shelly-em-smart-home-automation-device/ Без спец микросхемы интегрировать силу тока и напряжение на esp не получится (нужен точный и быстрый АЦП). Я делал токовое реле на stm32f103. Ну и делать высоковольтную схему (для счетчика придется делать делитель напряжения или ставить развязывающий трансформатор) самому довольно сложно

Я взял за пример вот эту реализацию
https://mysku.ru/blog/aliexpress/79653.html
правда софт переработал на 95% практически :)
Работает вроде довольно удовлетворительно.

dontsovcmc, 2022-06-16:
https://github.com/vad7/PowerMeter-IrDA  главный DIY проект сбора данных. что б не потерять.

dontsovcmc, 2023-09-15:
https://github.com/dontsovcmc/waterius/discussions/290

closed 2023-09-15 by dontsovcmc