dontsovcmc/waterius #150: Безбожно завышает по горячему каналу

Kabron287, 2021-02-05:
В 4 раза.
Дребезг?

dontsovcmc, 2021-02-07:
Хм... может плохой контакт (провод в пыли или не закрепили в разъеме) ?
От дребезга надежная софтовая защита...

Kabron287, 2021-02-07:
Да ладно! Барабашка гораздо вероятнее.
Там вроде в настройках коэффициент можно подкрутить?
Ибо, холодная тоже не идеальна.
[image]

OloloevReal, 2021-02-07:
@Kabron287 а вы с разводкой не напутали? У валтека видел счетчики с комбинированным выходом - и сухой контакт и namur.

Kabron287, 2021-02-07:
Т.е. я типа два провода местами перепутал?

dontsovcmc, 2021-02-16:
@Kabron287 Очень странно себя ведет. С Valtek Ватериус работает, но обычно у них 4 проводка, а у вас 2. Пришлите свою почту на contact@waterius.ru, посмотрим графики.

dontsovcmc, 2021-02-18:
@Kabron287 "Там вроде в настройках коэффициент можно подкрутить?" - счетчики либо 1л/имп, либо 10л/имп. Если ваш на 1л и вы не вылили 3л холодной воды во время настройки, то потребление должно ровно в 10 раз быть выше.
Я думаю будет правильно вам проверить импульсный выход тестером - измерьте сопротивление при открытой воде.

closed 2021-02-18 by dontsovcmc

Kabron287, 2021-02-18:
Как раз пришел OWON BT41+.
Повешу в режиме логгера на выходных.