dontsovcmc/waterius #156: Предложение: немного подкорректировать метод "принудительного сброса показаний"

AstashovAndrey, 2021-02-26:
То что можно будет сбросить показания нажав 1сек кнопку - это конечно хорошо, но предлагаю добавить на плату контакты для подключения внешнего тригера. Можно наверное даже на корпус не выводить, кому нужен будет функционал, может подключиться внутри и вывести 2 провода из корпуса наружу.

Лично у меня, два варианта использования:
1) в место где планируется установка прибора(да, я только смотрю на прибор и желание купить есть) у меня и так уже есть комплект входов-выходов от контроллера. Т.е. по желанию контроллера можно попросить передать актуальные данные
2) подключение к системе защиты от протечек. Протечка может быть и не отключаемая, но прибор её зафиксирует. И следовательно он "моргнет" реле и скажет ватериусу что у меня беда и на всякий случай скинь данные, а там разберемся.

dontsovcmc, 2021-02-26:
Это будет оповещение только если протечка у вас после счетчиков возникла. А если сверху/снизу/полотенцесушитель? Есть ли смысл только от одного источника протечки защищаться? проще датчик протечки прилепить к Ватериусу. Он есть в версии для 4х счетчиков

AstashovAndrey, 2021-02-26:
> Это будет оповещение только если протечка у вас после счетчиков возникла. А если сверху/снизу/полотенцесушитель? Есть ли смысл только от одного источника протечки защищаться? проще датчик протечки прилепить к Ватериусу. Он есть в версии для 4х счетчиков

Если есть аквасторож, то зачем использовать ватериус не по назначению?  У меня на аквастороже 6 датчиков по квартире лежит. Пускай каждый своим делом занимается. А тут больше не для оповещения, этим будет заниматься хотя бы тот же аквасторож через симкарту, или скажет об этом на другому участнику, и тот уже и смс отправит, или позвонит мне, или .... да хоть сразу такси закажет мне на адрес :) 

Я же писал про тригер для принудительного сброса показаний. На всякий случай. А вдруг пригодится.
В прошлом году был случай, наш офис в жилом доме на первом этаже. Дом не бедный. Счетчики на регулярном контроле у управляющей компании. Чуть ли не в реальном времени. Ночью у одного жильца, в тамбуре потек бойлер и пол ночи заливал все и вся. С утра, председатель УК сказал что бойлер на 8 кубов "накапал" :) Суммарно за месяц там конечно совсем другая цифры, но вот хозяину этого бойлера очень понравилось, что есть полный контроль над данными. Вроде и мелочь, вроде и не важно(все равно платить), но зато интерес удовлетворен, и человек доволен :)

dontsovcmc, 2021-03-14:
Я думаю это реально в модификации платы для 4х входов добавить, хотя и там пинов почти не осталось =(

closed 2021-03-14 by dontsovcmc