dontsovcmc/waterius #157: Предложение: Вход от датчика "Обнаружение магнитного поля"

AstashovAndrey, 2021-02-26:
У меня на счетчиках Бетар 4 провода. Два провода импульсные на подсчет расхода, и два провода на магнитное поле. Да, внутри стоит банальный геркон, который сработает до того, как крыльчатке станет труднее крутиться. И неплохо было бы об этом знать через интерфейс, что был факт использования магнита. В интерфейсе показывается тревога и дата тревоги. Пока руками не снимешь, не пропадает. Если в период было несколько фактов срабатывания тригера, то в интерфейсе отображать все факты.
Лично мне, в свою квартиру этот функционал не нужен, но я сейчас говорю как человек, который так же сдает квартиру. И вот там, такой функционал очень даже пригодится.

OloloevReal, 2021-02-26:
В каком интерфейсе показывается? В Blynk? Web? В интерфейсе настройки?

dontsovcmc, 2021-02-26:
Прикольный функционал. Вопрос: а так ли вам нужно спасать коммунальщиков от магнитов? Вы ведь не платите за ту воду, которую ваши жильцы вылили, используя магнит.

heX16, 2021-02-26:
> Лично мне, в свою квартиру этот функционал не нужен, но я сейчас говорю как человек, который так же сдает квартиру. И вот там, такой функционал очень даже пригодится.

Просто купите антимагнитную [пломбу](http://migakon.ru/plombirovochnoe-oborudovanie/plomba-antimagnit/) за 30 рублей. (А электронная защита не получится - арендаторы потом просто вынут батарейки и вы вообще ничего не узнаете)).

AstashovAndrey, 2021-02-26:
> В каком интерфейсе показывается? В Blynk? Web? В интерфейсе настройки?

Для начала хотя бы в веб. А аппетит иногда приходит во время еды. Оповестить что есть инцидент, и дать владельцу счетчика в интерфейсе на него отреагировать(прочитать).

AstashovAndrey, 2021-02-26:
> Просто купите антимагнитную [пломбу](http://migakon.ru/plombirovochnoe-oborudovanie/plomba-antimagnit/) за 30 рублей. (А электронная защита не получится - арендаторы в следующий раз вынут батарейки и вы вообще ничего не узнаете)).

Нет, не просто. В сети много мест, где продаются анти-анти магнитные пломбы. Сосед по квартире, во время ремонта повредил пломбу на счётчике. Нашел в сети ребят которые делают пломбы дубликаты. От оригинала практически не отличишь. Особенно когда пылью посыпать и водичкой полить. Или грязной рукой прикоснуться, якобы сантехник с грязной рукой, после сантехнической пасты залез на счетчик. :) 
Сколько проверок и сверок у него уже было, ни у кого даже подозрения "новая" пломба не вызвала. А чего, картинка совпадает, номер совпадает, антимагнитные метки на месте.... А по факту на счетчике постоянно висит магнит :)
А при замене счетчика, лично у нас бригада с микроскопами не выезжает. Звоню в водоканал и говорю что надо сорвать пломбу для таких то работ, разрешаете? Меня записывают, дают два-три дня и сразу запись на опломбировку делают. Все. 

А по поводу отключения прибора - это другая история. По договору аренды - не имеют права без согласования что либо делать с инженерными системами. Т.е. если хотят поменять кран, звонят и спрашивают. Уточняю почему и зачем, и говорю какой именно им надо купить кран, если это по их вине. Иначе, оставляю за собой право на аннулирование гарантийного платежа, который возвращается если все хорошо по закрытию договора. И по итогу, стоит коробка в нише, на коробке или персональная пломба самоклейка, или пломба из любого DIY магазина, продаются разного размера и цветов. Которые при отклеивании расслаиваются, и обратно так же заколебаешься незаметно приклеивать.

Сразу отвечу по поводу интернета - в квартире ростелеком с модемом от ростелекома на оптике. менять нельзя :) Что бы не долбить и сверлить :) Модемы ростелекома перенастроить проблематично(по крайней мере у нас), они откатываются на настройки от РТ на раз. Они не охотно их отвязывают от своей системы автоконфигурирования. Мол если надо, ставьте дальше свой роутер и на нем рулите, а тут dmz и в добрый путь. А без автоконфигурирования роутера и регулярной регистрации роутера РТ в сети, будут проблемы с доступами и резервированием портов. Что есть правда :) У меня в местном РТ есть технари знакомые, и я как "юрик" просил убрать эту мерзость :) Сказали что лучше не надо, проблем будет больше :)

AstashovAndrey, 2021-02-26:
> Прикольный функционал. Вопрос: а так ли вам нужно спасать коммунальщиков от магнитов? Вы ведь не платите за ту воду, которую ваши жильцы вылили, используя магнит.

Тут больше не коммунальщиков, а свой тыл. То что вырастет ОДН, это не важно и не значительно. А вот если придут, и квартиранты хором скажу "Это не мы, это хозяин квартиры так сделал, мы то нет, мы то никогда". Вот тогда уж я, как собственник квартиры буду везде бегать и доказывать. И то не получится, потому что всем пофиг. И только через суд... И то докажи еще. Короче, полезный функционал :) 

А раз выше heX16 обмолвился об отключении устройства, то неплохо было бы, если показывалось предупреждение, что "коллектор", назову устройство так :), был перезагружен, после предыдущего отчетного периода. Просто добавить в интерфейс, в правый верхний угол "колокольчик", например как во вконтакте, и там циферками помечать что есть новые события. Ты бы глянул сюда. А так и отключение, и отсутствие данных больше ХХХ времени для конкретного устройства(счетчика), и попытки "скрысить" воду.

AstashovAndrey, 2021-02-26:
> А раз выше heX16 обмолвился об отключении устройства, то неплохо было бы, если показывалось предупреждение, что "коллектор", назову устройство так :), был перезагружен, после предыдущего отчетного периода.

Причем формально это совсем не сложно реализовать. Я так думаю. При старте фиксировать время в памяти. А дальше, или специальным пакетом отправлять время старта на сервер, или отправлять время начала роботы устройства при каждой передаче статы по счетчикам. Хотя думаю что это лишнее, но так как количество передач данных не велико, то это не значительное отклонение от морально этического поведения разработчика :)

closed 2021-03-14 by dontsovcmc