dontsovcmc/waterius #158: Прошитая attiny85-20SUR есть 250мкА

dominionish, 2021-03-04:
Всем привет!
Понравилась задумка, поставил счетчики с проводами и решил сваять сам вей девайс.
Плату сделал посредством ЛУТ, все вроде по технологии и по вашей инструкции
Прошил контроллер и проверив ток, оказалось что он кушает 250мкА. Что в 10 раз больше заявленного в вашей инструкции.
Плату отмывал от флюса с помощью средства flux off, затем промывал и вымчивал в спирте, чтобы вода вытеснилась
Не помогло.
Может ли быть дело еще в чем то? Я так понимаю ЛУТ - это самая первая версия платы. ИЛи не важна версия, ток будет 7-10 мкА в любом случае?
Вобщем буду раз любым советам.
Вот кстати фото платы, сделанной моими не очень прямыми руками. Но все правильно, проверял 10 раз по схеме:
[image]
[image]

dominionish, 2021-03-04:
Вобщем пока я это писал, плата сушилась уже от второй промывки средством Flux Off. После того как средство испарилось мультиметр показал потребление тока в 8-14 мкА. Что в принципе соответствует заявленному. Поэтому issue можно закрывать. 
И спасибо за такой крутой проект и за подробное описание с инструкцией!

dontsovcmc, 2021-03-04:
Отличная вышла плата!
Да, после прошивки должно быть около 10мкА (прыгает, т.к. измерения идут). Часто из-за флюса повышенное потребление. Проверьте еще после суток работы!
Спасибо за отзыв!

closed 2021-03-04 by dontsovcmc