dontsovcmc/waterius #163: Доработка под электромагнитные индукционные счетчики

maijye, 2021-03-16:
Здравствуйте! Возможно ли доработать эту плату (сделать 5-й выход) под счетчики/расходомеры, с 3 проводами.
Красный - питание+ (12в)
Черный - питание-
Желтый - электрические импульсы 5v по 10-15мА 

[image]

[image]

dontsovcmc, 2021-03-16:
Это хорошая задачка.
Программно: 
1. требуется добавить подсчет импульсов по прерыванию.
2. PULLUP входов всегда включен.

maijye, 2021-03-16:
> Это хорошая задачка.
> 
> Программно: 
> 
> 1. требуется добавить подсчет импульсов по прерыванию.
> 
> 2. PULLUP входов всегда включен.
> 
> 
> 
> 

Как Вы думаете, намного ли этот 5-й выход может увеличить потребление энергии?
Есть разные счетчики, примерно от 20 до 10000 импульсов/литр.

Есть счетчики для таких приборов с экраном. Работают от батареек.
[image]

[image]

dontsovcmc, 2021-03-16:
Не увеличит потребление никак. 

3. Переделать буфер, сохраняющий каждый импульс в eeprom в то, чтобы он сохранял, скажем каждый литр.
4. Добавить вес импульса в ESP.

maijye, 2021-03-17:
> Не увеличит потребление никак. 
> 
> 
> 
> 3. Переделать буфер, сохраняющий каждый импульс в eeprom в то, чтобы он сохранял, скажем каждый литр.
> 
> 4. Добавить вес импульса в ESP.

По-хорошему, нужно сначала через счетчик прокачивать литр воды, а в метрике указывать, что - вот, это был ровно литр. А в метрике будет фиксироваться, что в 1 литре тестовом 577 импульсов например.

Мало того, что все эти счетчики сильно отличаются по количеству импульсов/литр, так еще и каждый счетчик одной модели может иметь индивидуальные отклонения. Например, по тех. характеристикам должно быть 30 литров/литр, а реально эти счетчики дают по-разному 27-35 импульсов/литр.

dontsovcmc, 2021-03-22:
@maijye вау жесть =(. вообще интерфейс ватериуса позволяет реализовать градуировку легко - веб морда получает информацию от attiny о кол-ве импульсов. 
Но к сожалению задача специфическая, поэтому закрою тему. Хотя многие спрашивают про расходомеры, но я не подозревал про необходимость градуировки =(. Теперь буду знать.

closed 2021-03-22 by dontsovcmc