dontsovcmc/waterius #165: Обновить сертификат lets encrypt

dontsovcmc, 2021-03-22:
Ввиду отсутствия обновления по воздуху, целесообразно добавить не Let’s Encrypt R3 сертификат, который до 2025г, а повыше. Нужно проверить, будет ли с ним работать BearSSL: сделать запрос на https://waterius.ru и убедиться что сайт вернет 405 Method Not Allowed.

https://letsencrypt.org/certificates/

neitri, 2022-03-02:
Добавил сертификат. изменил адрес с cloud.waterius.ru на waterius.ru
Фрагмент кода добавления сертификатов
`    // Set wc client
    WiFiClient *wc;
    if (url.substring(0, 5) == "https") {
        wc = &wifiTlsClient;
        wifiTlsClient.setDebugOuput(true);
        //certs.append(lets_encrypt_x3_ca);
        //certs.append(lets_encrypt_x4_ca);
        //certs.append(cloud_waterius_ru_ca);
        certs.append(isrg_root_x1);
        wifiTlsClient.setTrustAnchors(&certs);`
Лог подключения:
`SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
00:00:087  INFO  : Booted
00:00:087  INFO  : Saved SSID: iot
00:00:090  INFO  : Saved password: home-iot
00:00:097  INFO  : mode: 2
00:00:116  INFO  : version: 21
00:00:116  INFO  : service: 1
00:00:116  INFO  : setup_started_counter: 0
00:00:116  INFO  : resets: 125
00:00:117  INFO  : MODEL: 0
00:00:119  INFO  : state0: 0
00:00:122  INFO  : state1: 0
00:00:124  INFO  : impulses0: 10
00:00:127  INFO  : impulses1: 0
00:00:130  INFO  : adc0: 0
00:00:133  INFO  : adc1: 0
00:00:135  INFO  : CRC ok
00:00:138  INFO  : Configuration CRC ok
00:00:141  INFO  : --- Waterius.ru ----
00:00:145  INFO  : email=neitri@gmail.com
00:00:148  INFO  : host=https://waterius.ru key=2B3B86E1BF71D71D1824A81A7B88800C
00:00:155  INFO  : wakeup min=1440
00:00:159  INFO  : --- Blynk.cc ----
00:00:162  INFO  : host=blynk-cloud.com key=
00:00:166  INFO  : email=
00:00:168  INFO  : --- MQTT ----
00:00:171  INFO  : host= port=1883
00:00:174  INFO  : login= pass=
00:00:177  INFO  : topic=waterius/4835673/
00:00:181  INFO  : --- Network ----
00:00:184  INFO  : DHCP turn off
00:00:187  INFO  : static_ip=192.168.0.3
00:00:191  INFO  : gateway=192.168.0.1
00:00:194  INFO  : mask=255.255.255.0
00:00:198  INFO  : --- Counters ----
00:00:201  INFO  : channel0 start=0.07, impulses=10, factor=10
00:00:207  INFO  : channel1 start=0.00, impulses=0, factor=10
00:00:212  INFO  : new impulses=10 0
00:00:216  INFO  : new value0=0.07 value1=0.00
00:00:220  INFO  : delta0=0 delta1=0
00:00:223  INFO  : Starting Wi-fi
fpm close 1
            mode : sta(c4:5b:be:49:c9:59)
                                         add if0
                                                00:00:231  INFO  : Static IP OK
00:00:234  INFO  : hostname Waterius-49c959
00:00:238  INFO  : bssid set 0
00:00:240  INFO  : Slow connect
00:00:244  INFO  : Status: 7
00:00:546  INFO  : Status: 7
00:00:847  INFO  : Status: 7
00:01:148  INFO  : Status: 7
00:01:448  INFO  : Status: 7
00:01:749  INFO  : Status: 7
00:02:050  INFO  : Status: 7
00:02:350  INFO  : Status: 7
00:02:651  INFO  : Status: 7
[... 167 lines cut ...]
'[HTTP-Client][handleHeaderResponse] RX: 'Date: Wed, 02 Mar 2022 16:38:46 GMT
'[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html; charset=utf-8
'[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 166
'[HTTP-Client][handleHeaderResponse] RX: 'Connection: close
'[HTTP-Client][handleHeaderResponse] RX: '
 [HTTP-Client][handleHeaderResponse] code: 405
                                              [HTTP-Client][handleHeaderResponse] size: 166
                                                                                           00:06:840  INFO  : Response code:    405
[HTTP-Client][end] tcp is closed
                                00:06:846  INFO  : Response body:       <html>
<head><title>405 Not Allowed</title></head>
<body bgcolor="white">
<center><h1>405 Not Allowed</h1></center>
<hr><center>nginx</center>
</body>
</html>

[HTTP-Client][end] tcp is closed
                                00:06:866  INFO  : -- END --
00:06:869  INFO  : Send HTTP code:      405
00:06:872  INFO  : -- END --
00:06:877  INFO  : Wakeup period, min:1440
00:06:879  INFO  : Wakeup period, tick:1440
00:06:929  INFO  : Config stored OK
00:06:930  INFO  : Going to sleep`

vzagorovskiy, 2023-02-15:
Fixed #3f91e8ff0e2c23623bc8058c599e4bbab1dca61c

closed 2023-04-29 by dontsovcmc