dontsovcmc/waterius #168: А не RF_DISABLED ли нужен при выключении?

dontsovcmc, 2021-03-30:
Сейчас в setup(): WiFi.mode(WIFI_OFF), а в конце loop  ESP.deepSleepInstant(0, RF_DEFAULT); и это "как то не определенно":

neitri, 2021-04-04:
Я немного читал про deepSleep, и если указать RF_DISABLED, то при следующем включении радио тракт будет отключен. И проблема будет в том, что нельзя будет его просто включить. Необходимо будет отправить в сон с параметром RF_DEFAULT. И при следующем просыпании все будет работать.

dontsovcmc, 2021-04-04:
@neitri А WiFi.begin(); не включит радиотракт? Я RF_DISABLED вижу у просыпающихся раз в 5 мин проектов.

neitri, 2021-04-04:
Посмотрел сейчас sdk api там в описании функции syste_deep_sleep говорится, что Disable RF после пробуждения приём и получение данных будет недоступно. 
Если esp сам себя будет через перемычку между выводами xpd_dcdc и ext_rstb то можно просыпаться без радиотракта, выполнять какие либо операции и кратковременно уходить в сон с включением радиотракта.
В режиме сна rtc продолжает работать. Но ватериус уходит в shutdown. Он возможно в любом случае запустится с радиотрактом.

dontsovcmc, 2021-07-13:
Мы пока не перешли на 3.0.1, но я вижу там 
BREAKING: Disable WiFi at boot by default (#7902, #7990, #7993)
https://github.com/esp8266/Arduino/releases/tag/3.0.1

Если бы период отправки был небольшой - 15-30 минут, то с ESP экономичнее работать не снимая HIGH с EN, а просто ресетя её. Тогда подключение к точки доступа будет не 5 сек, а 0.5 - выгода больше, чем увеличение потребления в режиме сна. (Но для этого нужно в hardware делать изменения.)

closed 2021-07-13 by dontsovcmc