dontsovcmc/waterius #175: Подключение к счётчикам Бетар СГВ-15 + настройки роутера

ivakorin, 2021-04-11:
У бетар есть счётчики с четырмя проводами:
- красный
- жёлтый
- чёрный
- синий
Подключать нужно чёрный и синий. 
*** 
Дополнительно. 
Если роутер стоит в режиме Wireless Network Mode: N-only, то есть вероятность, что Ватериус не сможет подключится к сети, если уровень сигнала не очень высок. Как решение стоит перевести роутер в Wireless Network Mode: Mixed или перенести роутер поближе.

dontsovcmc, 2021-04-11:
Спасибо за информацию!! Бетар - это новый "зверь". А какой у вас роутер?

ivakorin, 2021-04-11:
Ротуер к которому подключается Ватериус TP-Link ARCHER-C7 v5 
Но он с DD-WRT на борту, не очень репрезентативно будет, полагаю.

AstashovAndrey, 2021-04-11:
> Спасибо за информацию!! Бетар - это новый "зверь". А какой у вас роутер?

Какой же он новый. Он давным давно уже продается. У нас(Омск), я других счетчиков и не видал. Бетар раньше чуть ли не в одно рыло был, как обязали устанавливать счетчики. Пока DIY сетевеки не стали привозить все подряд. 
И я говорил, что на этом счетчике 4 провода, два под геркон на крыльчатку, два под геркон у корпуса, что бы факт намагничивания ловить.

dontsovcmc, 2021-04-12:
Добавил https://github.com/dontsovcmc/waterius/issues/65 спасибо
Про роутер понял, была у кого-то такая проблема.

closed 2021-04-12 by dontsovcmc