dontsovcmc/waterius #176: Отодвинуть контакты ESP подальше от Attiny

bublikOff, 2021-04-13:
Отодвинуть контакты ESP подальше от Attiny. Дабы была возможность подкинуть прищепку для программирования Attiny ... в текущей реализацию прищепка не влазит

dontsovcmc, 2021-04-13:
А есть фото прищепки?

bublikOff, 2021-04-13:
https://www.google.com/search?q=%D0%BF%D1%80%D0%B8%D1%89%D0%B5%D0%BF%D0%BA%D0%B0+SO-8&tbm=isch&sa=X
В принципе достаточно распространены черные варианты ... как самые ходовые от китайцев. Часто идут с программаторами. На вскидку прямая ссылка на фото https://media2.24aul.ru/imgs/58734f3023bbeb1ce824f976/

dontsovcmc, 2021-04-13:
А можете сделать фото прищепки и ватериуса?
А пин ресет не мешает?
Если пин ресет и C3 перенести к ESP, а attiny поднять на 2мм, то влезет ли прищепка между ESP и кнопкой?

bublikOff, 2021-04-13:
https://i.ibb.co/MZb1ZQ1/viber-image-2021-04-14-02-16-05.jpg ... вот так прищепка встает относительно разъема ESP

dontsovcmc, 2021-04-13:
А влезает между кнопкой и ESP? А пин не будет мешать?

bublikOff, 2021-04-14:
Влазит ... Но, получается что аттини надо сдвинуть до границы контакта rst
https://i.ibb.co/k4xVGXH/IMG-20210414-095457.jpg

obrianw, 2021-05-04:
> Влазит ... Но, получается что аттини надо сдвинуть до границы контакта rst
> https://i.ibb.co/k4xVGXH/IMG-20210414-095457.jpg
Прошу прощения, что вмешиваюсь, а может проще допилить у прищепки внешние края глубже? Ведь там нет ничего.
Плата реально плотная и уже вписана под распространенный корпус для батарейного отсека. :)

"Напильник наше всё!" (c) народная мудрость. :)

dontsovcmc, 2022-12-19:
done версия [waterius-factory-3.2.brd](https://github.com/dontsovcmc/waterius/blob/master/Board/waterius-factory-3.2.brd)

closed 2022-12-19 by dontsovcmc