dontsovcmc/waterius #177: Переименовать Ватериус в роутерах

dontsovcmc, 2021-04-14:
Кажется Ватериус определяется роутером как «ESPxxxx». Нужно сделать чтобы роутер его определял как «Waterius-xxxxx», где хххх - chip id.

neitri, 2021-04-15:
Нужно копать в сторону wifi.hostname("имя сети")

dontsovcmc, 2021-04-15:
Огонь, спасибо!
WiFi.hostname("relay");
WiFi.begin(ssid, password);

dontsovcmc, 2021-07-05:
@neitri вы добавил установку hostname уже после подключения к роутеру, думаю нужно ее ставить перед: WiFi.begin();
(у вас может работать так, т.к. достаточно 1 раз установить и esp8266 будет использовать это имя впредь)

neitri, 2021-07-06:
Пробовал и в setup и вначале loop. Неработало. Только после соединения заработало. До соединения даже имя не возвращает, пустая строка приходит.

dontsovcmc, 2022-02-21:
done

closed 2022-02-21 by dontsovcmc